openwrt/packages #9776: nginx: build fails with OpenWrt git master

dbermond, 2019-08-19:
**Maintainer:** @heil @Ansuel 

**Environment:**

- **Architecture:** arm_cortex-a15_neon-vfpv4
- **Model:** R7800
- **OpenWrt and packages:** git master
- **Host system:** Arch Linux x86_64

**Description:**
- nginx/nginx-ssl fails to build when using OpenWrt git master.

- [Build error log](https://paste.ee/p/Sjuht) ([alternative link](https://hastebin.com/nozonitoda.coffeescript)).

- Steps used to build:

```
$ curl -O https://downloads.openwrt.org/snapshots/targets/ipq806x/generic/openwrt-sdk-ipq806x_gcc-7.4.0_musl_eabi.Linux-x86_64.tar.xz
$ tar xf openwrt-sdk-ipq806x_gcc-7.4.0_musl_eabi.Linux-x86_64.tar.xz
$ cd openwrt-sdk-ipq806x_gcc-7.4.0_musl_eabi.Linux-x86_64
$ ./scripts/feeds update -a
$ ./scripts/feeds install nginx
$ make menuconfig
  -> unselect everything in 'Global build settings'
  -> Network -> Web Servers/Proxies
  -> select nginx-ssl (or nginx)
  -> Configuration
  -> select all nginx-ssl (or nginx) modules
  -> exit and save .config
$ make V='s' -j1
```

- nginx/nginx-ssl also fails to build when using the OpenWrt build system (git master) for building the full image from scratch (the error is different, but similar).

- When using the build system for generating the full image, my last shot was r10366-c2e819537f and nginx-ssl was built fine in this revision (nginx-ssl version 1.16.0-1).

- It currently builds fine when using the SDK from stable release 18.06.4: [openwrt-sdk-18.06.4-ipq806x_gcc-7.3.0_musl_eabi.Linux-x86_64.tar.xz](https://downloads.openwrt.org/releases/18.06.4/targets/ipq806x/generic/openwrt-sdk-18.06.4-ipq806x_gcc-7.3.0_musl_eabi.Linux-x86_64.tar.xz)

Ansuel, 2019-08-19:
I think it's related to something else as I currently have compilation
error with a luci-lib package... Last week both compiled well so..

dbermond, 2019-08-20:
I found the cause: this error is caused by the latest brotli module update from commit https://github.com/openwrt/packages/commit/b2f0cb973ecaf8a878f2d7b97fab96ce4e0bc302.

This commit brakes the nginx build for me.

Reverting commit https://github.com/openwrt/packages/commit/b2f0cb973ecaf8a878f2d7b97fab96ce4e0bc302 or disabling the brotli module makes nginx build fine again.

dengqf6, 2019-08-20:
I cannot reproduce when it's built from source.

The first error in your log:
```
/storage/linux/src/openwrt-sdk-ipq806x_gcc-7.4.0_musl_eabi.Linux-x86_64/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.4.0_musl_eabi/include/fortify/unistd.h:43:29: error: redefinition of 'confstr'
 _FORTIFY_FN(confstr) size_t confstr(int __n, char *__s, size_t __l)
                             ^~~~~~~
```
I don't think it's related to brotli.
Buildbot does not have nginx faillogs either.

> ```
> $ ./scripts/feeds update -a
> $ ./scripts/feeds install nginx
> ```

By doing this you will miss a lot of dependencies..

diizzyy, 2019-08-22:
Indeed, at least for now we don't support partially importing the package tree.

InBetweenNames, 2019-08-31:
I was able to reproduce this.

InBetweenNames, 2019-09-20:
Investigating further: the wrong headers are being pulled in.  They're being sourced from `/usr/include` and not the buildroot.

neheb, 2020-01-17:
Is this still an issue?

InBetweenNames, 2020-01-17:
Yes it is, just tried with latest git.

InBetweenNames, 2020-01-17:
Patch with known working version of Brotli: https://gist.github.com/InBetweenNames/4ed32fddc257203930fc9af47a221020

neheb, 2020-01-17:
Is this the same as https://github.com/openwrt/packages/issues/6905 ?

InBetweenNames, 2020-01-17:
Nope, this happens during compilation, it has nothing to do with runtime behaviour.

neheb, 2020-01-17:
Can't reproduce locally.

The issue in the beginning of the thread has to do with fortify-headers (which I personally want to remove).

Maybe some combination of variables in include/hardening will fix this.

Also note that the error is with GCC7. Mainline has moved to GCC8.

InBetweenNames, 2020-01-17:
I'm on GCC 9 here.  Can you post your diffconfig?

neheb, 2020-01-17:
https://gist.github.com/neheb/3037333560c138095ecef33fc5585260

InBetweenNames, 2020-01-17:
Weird.  I've done several completely clean rebuilds since I first started encountering this and I still haven't been able to shake it.  I've been just using the patch I posted to use an earlier brotli version until it gets sorted out.

InBetweenNames, 2020-01-17:
My diffconfig: https://gist.github.com/InBetweenNames/2bed87839ac065a4e07ed4dbb14ec316

dbermond, 2020-01-17:
I can also confirm that this issue is still present for me.

I can only compile nginx with brotli module by reverting commit b2f0cb973ecaf8a878f2d7b97fab96ce4e0bc302.

neheb, 2020-01-20:
Strange. @dengqf6 any idea?

dengqf6, 2020-01-20:
Try using tar.gz instead of tar.xz?

Ansuel, 2020-01-21:
I really can't reproduce this... I followed all the steps and i have the same target... 
Can someone check if the buildbot fail to build the nginx all variant?

neheb, 2020-01-22:
Neither can I.

What confuses the issue is that the log in the first post deals with fortify-headers. The config provided by @InBetweenNames uses glibc, which does not use fortify-headers.

InBetweenNames, 2020-01-22:
I'll spend some time today and see if I can nail down why my system's `/usr/include` headers are being pulled into the build system (to my knowledge `/usr/include` shouldn't even be touched).

InBetweenNames, 2020-01-23:
I believe I have identified the issue.  The `ngx-brotli` module is being sourced from this repo:

https://github.com/eustas/ngx_brotli

which is now archived and has been merged to https://github.com/google/ngx_brotli.

Modifying the Makefile for nginx to source from that repo, and use the latest commit `e505dce68` has completely resolved the issue for me.

InBetweenNames, 2020-01-23:
I will prepare a PR shortly and link this issue.

InBetweenNames, 2020-01-23:
To be clear, the issue was on `ngx_brotli`'s side, not OpenWRT -- something changed in their build system that started pulling in system headers instead of the buildroot ones.

closed 2020-01-23 by neheb