openwrt/packages #9817: Upgrade Problem

jkle112, 2019-08-24:
Simple-Adblock that installled in OpenWrt version 18.06.4 within Linksys WRT1900AC

When add more sources urls in the Simple-Adblock,but it is full not download more new sources to update cache file.
[image]

DazzyWalkman, 2019-08-24:
Maybe you can look at package adblock. 
Some of your expections may not go 'simple'.

jkle112, 2019-08-25:
Adblock :could not to upgrade by myself,because when it is starting,download config file from github.com with author ,rewrite my config file.If used it for big data,could not to deal with that without enough memory.Sometimes ,that's could not to processing very well,and downloading error,some AD that could not to filter.
Simple-Adblock:only one problems:could not support big data.Downloading error ,that need support curl /wget with SSL to fix .
So if use more better hardware like X64,need Simple-Adblock to upgrade.I will use it more powerful and become a loacal big cache DNS sever.
Thank you !

jkle112, 2019-09-02:
Simple-Adblock that installled in openwrt-koolshare-mod-v2.32-r11575

[image]

When add more sources urls in the Simple-Adblock,but it is full not download more new sources to update cache file.
[image]

Need upgrade Simple-Adblock to support Big Data for powerful and add more functions to speed up ,dns fast look up low memory
[image]

[image]
Hope  Simple-Adblock add a button:manual update for every url,that update more quickly.
Thank you!

neheb, 2019-11-04:
ping @stangri

stangri, 2019-11-04:
There is a newer version of Luci app which has force redownload button. If your request was about something else, please clarify.

stangri, 2019-11-09:
How long do we keep this open if there's no comment from OP?

neheb, 2019-11-10:
I’ll close. @jkle112 please open a new issue if you still have problems.

closed 2019-11-10 by neheb