openwrt/packages #9819: keepalived: depends on "file" package due to libmagic

petzah, 2019-08-24:
Maintainer: Florian Eckert <fe@dev.tdt.de>
Environment: keepalived - 2.0.10-1 , turris omnia 

Description:
keepalived is trying to open /usr/share/misc/magic.mgc , if not found then sigabrt
The magic.mgc is part of "file" package

```
[pid 25966] chdir("/etc/keepalived")    = 0
[pid 25966] readv(9, [{"", 0}, {"global_defs {\n   notification_em"..., 1024}], 2) = 1024
[pid 25966] clock_gettime(CLOCK_MONOTONIC, {4840, 441718101}) = 0
[pid 25966] clock_gettime(CLOCK_MONOTONIC, {4840, 441867872}) = 0
[pid 25966] clock_gettime(CLOCK_MONOTONIC, {4840, 442000241}) = 0
[pid 25966] clock_gettime(CLOCK_MONOTONIC, {4840, 442127370}) = 0
[pid 25966] readv(9, [{"", 0}, {" priority 95\n    advert_int 1\n  "..., 1024}], 2) = 408
[pid 25966] clock_gettime(CLOCK_MONOTONIC, {4840, 442554520}) = 0
[pid 25966] readv(9, [{"", 0}, {" priority 95\n    advert_int 1\n  "..., 1024}], 2) = 0
[pid 25966] close(9)                    = 0
[pid 25966] fchdir(10)                  = 0
[pid 25966] close(10)                   = 0
[pid 25966] clock_gettime(CLOCK_MONOTONIC, {4840, 443064437}) = 0
[pid 25966] stat64("/root/.magic.mgc", 0xbeef45a0) = -1 ENOENT (No such file or directory)
[pid 25966] stat64("/root/.magic", 0xbeef45a0) = -1 ENOENT (No such file or directory)
[pid 25966] open("/usr/share/misc/magic.mgc", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
[pid 25966] rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1 RT_2], [HUP INT USR1 USR2 TERM], 8) = 0
[pid 25966] gettid()                    = 25966
[pid 25966] tkill(25966, SIGABRT)       = 0
[pid 25966] rt_sigprocmask(SIG_SETMASK, [HUP INT USR1 USR2 TERM], NULL, 8) = 0
[pid 25966] --- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=25966, si_uid=0} ---
[pid 25966] rt_sigprocmask(SIG_BLOCK, ~[], NULL, 8) = 0
[pid 25966] --- SIGILL {si_signo=SIGILL, si_code=ILL_ILLOPC, si_addr=0xb6f76094} ---
[pid 25966] +++ killed by SIGILL +++
<... epoll_pwait resumed> {{EPOLLIN, {u32=3067453264, u64=3067453264}}}, 2, -1, NULL, 8) = 1
clock_gettime(CLOCK_MONOTONIC, {4840, 444547622}) = 0
read(6, "\21\0\0\0\0\0\0\0\2\0\0\0ne\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 128) = 128
wait4(-1, [{WIFSIGNALED(s) && WTERMSIG(s) == SIGILL}], WNOHANG, NULL) = 25966
clock_gettime(CLOCK_REALTIME, {1566671684, 786075500}) = 0
getpid()                                = 25964
sendto(3, "<30>Aug 24 18:34:44 Keepalived[2"..., 78, 0, NULL, 0) = 78
wait4(-1, 0xbeef469c, WNOHANG, NULL)    = -1 ECHILD (No child process)
read(6, 0xbeef46d0, 128)                = -1 EAGAIN (Resource temporarily unavailable)
clock_gettime(CLOCK_REALTIME, {1566671684, 786579376}) = 0
getpid()                                = 25964
sendto(3, "<25>Aug 24 18:34:44 Keepalived[2"..., 79, 0, NULL, 0) = 79
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1 RT_2], [HUP INT USR1 USR2 TERM CHLD], 8) = 0
gettid()                                = 25964
tkill(25964, SIGTERM)                   = 0
rt_sigprocmask(SIG_SETMASK, [HUP INT USR1 USR2 TERM CHLD], NULL, 8) = 0
timerfd_settime(5, 0, {it_interval={0, 0}, it_value={2147483647, 0}}, NULL) = 0

```

feckert, 2019-08-27:
On my build i have no file `/usr/share/misc/magic.mgc` in file package

> root@APU3:~# cat /usr/lib/opkg/info/file.list 
> /usr/share/misc/magic
> /usr/bin/file

On openwrt 18.06 the keepalived version is on 1.4.4
On openwrt 19.07 the keepalived version is on 2.0.16
On openwrt master the keepalived version is on 2.0.18

So the version 2.0.10 is not supported by openwrt

petzah, 2019-08-27:
Closing, v2.0.10 is in TurrisOS 3.11.6 , must be their build

closed 2019-08-27 by petzah