openwrt/packages #9829: grep Context Line Control bug

souravndp, 2019-08-26:
Maintainer: @Zokormazo  
OpenWrt 18.06.4, Package architecture: mips_24kc, Grep Version: 2.26-1

Description:
1. I am using the following command: `grep -A5 -m1 -e "2" tmp.txt`
2. tmp.txt contains:
1
2
3
4
2
5
6
7
8
3. I should get the following output:
2
3
4
2
5
6
4. But I am getting this output:
2
3
4
5. As soon `grep` finds another match, it discards the rest of the lines. I have attached a screenshot which shows this behaviour. The left window is OpenWRT and the right one is Ubuntu (WSL)

[image]

dibdot, 2019-08-29:
Does the same happen with the grep version in current master?

souravndp, 2019-08-29:
As per your suggestion, I installed grep version 3.3. It doesn't have the bug which I have reported for version 2.26. Unfortunately, version 3.3 is not available in [OpenWRT 18.06.4 repository](http://downloads.openwrt.org/releases/18.06.4/packages/mips_24kc/packages/).  Also [this ](https://github.com/openwrt/packages/tree/openwrt-18.06/utils/grep )and [this](https://openwrt.org/packages/pkgdata/grep). So, I downloaded the version 3.3 compiled package from [here](http://downloads.openwrt.org/releases/packages-19.07/mips_24kc/packages/).

Thank you for your help. I don't know much about how these things work but do you have any idea why they don't include the latest updated packages in the repository? OpenWRT 18.06.4 is the latest release and the grep has been updated months ago.

[image]

closed 2019-08-29 by souravndp