openwrt/packages #9842: strongswan uci interface is outdated

dront78, 2019-08-27:
Maintainer: @stintel
Environment: master
Package strongswan-full

Description:

logread
```
00[LIB] plugin 'uci' failed to load: Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: uci_lookup: symbol not found
```

ldd /usr/lib/ipsec/plugins/libstrongswan-uci.so
```
        ldd (0x77ea2000)
        libuci.so => /lib/libuci.so (0x77e76000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77e52000)
        libc.so => ldd (0x77ea2000)
        libubox.so => /lib/libubox.so (0x77e39000)
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: nop: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: traffic_selector_create_dynamic: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: linked_list_create_from_enumerator: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: return_false: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: chunk_create_clone: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: builtin_vfprintf: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: callback_job_create_with_prio: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: enumerator_create_token: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: auth_cfg_create: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: uci_lookup: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: proposal_create_from_string: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: strerror_safe: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: traffic_selector_create_from_cidr: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: controller_cb_empty: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: return_null: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: peer_cfg_create: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: thread_cancelability: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: ike_cfg_create: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: shared_key_create: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: charon: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: child_cfg_create: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: lib: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: enumerator_enumerate_default: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: builtin_fprintf: symbol not found
Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: identification_create_from_string: symbol not found
```

stintel, 2019-09-03:
To be honest I have never used the UCI interface. Feel free to send a PR to update it.

dront78, 2019-09-05:
I have no idea, how to do this. Could you please clarify?

feckert, 2019-09-05:
I think you have to update/patch the libstrongswan-uci plugin so that it works again.

dront78, 2019-09-08:
for me it's easier to remove libstrongswan-uci.so from the package

neheb, 2020-02-09:
Is this still an issue? There was an update 23 days ago.

dront78, 2020-03-17:
@neheb please attach commit number if it was fixed. Personally, I'm not tracking a git branches )

Linaro1985, 2020-07-08:
The UCI API was changed in 2008 and the uci_lookup function no longer exists. Now this plugin doesn't work. I think strongswan-mod-uci should be droped or reimplemented with a new API.

CodeFetch, 2021-06-12:
@pprindeville Is this outdated with your recent changes?

pprindeville, 2021-06-17:
> @pprindeville Is this outdated with your recent changes?

It's not required by `swanctl`, which is what everything is migrating to.

ptpt52, 2023-03-23:
@pprindeville 
but it is still needed by strongswan-isakmp, and issue still for latest version in master
```
daemon.info : 00[LIB] plugin 'uci' failed to load: Error relocating /usr/lib/ipsec/plugins/libstrongswan-uci.so: uci_lookup: symbol not found
```

RobsonRojas, 2023-05-24:
Is there any patch to fix this uci_lookup missing symbol?

pprindeville, 2024-04-09:
Any progress on this?  Is anyone still interested?  If not, I can close this out.

trippleflux, 2024-05-10:
Bump, It seems with gcc 14 toolchain managed to pickup inexistence symbol : 
```
OpenWrt-libtool: compile:  ccache x86_64-openwrt-linux-gnu-gcc -DHAVE_CONFIG_H -I. -I../../../.. -I../../../../src/libstrongswan -I../../../../src/libcharon -I/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/usr/include -I/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/include -I/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/libiconv-full/include -I/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/libintl-full/include -rdynamic -Wno-format -Wno-format-security -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-pointer-sign -Wno-sign-compare -Wno-type-limits -Wno-unused-parameter -march=tigerlake -mtune=tigerlake -O3 -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fno-caller-saves -fno-plt -march=tigerlake -mtune=tigerlake -O3 -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fhonour-copts -fmacro-prefix-map=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/strongswan-5.9.14=strongswan-5.9.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/libiconv-full/include -I/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/libintl-full/include -include /home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/strongswan-5.9.14/config.h -c uci_control.c  -fPIC -DPIC -o .libs/uci_control.o
uci_control.c: In function 'status':
uci_control.c:107:49: warning: unknown conversion type character ' ' in format [-Wformat=]
  107 |                         fprintf(out, "%-8s %-20D %-16H ", ike_sa->get_name(ike_sa),
      |                                                 ^
uci_control.c:107:55: warning: unknown conversion type character ' ' in format [-Wformat=]
  107 |                         fprintf(out, "%-8s %-20D %-16H ", ike_sa->get_name(ike_sa),
      |                                                       ^
uci_control.c:107:38: warning: too many arguments for format [-Wformat-extra-args]
  107 |                         fprintf(out, "%-8s %-20D %-16H ", ike_sa->get_name(ike_sa),
      |                                      ^~~~~~~~~~~~~~~~~~~
uci_control.c:115:49: warning: unknown conversion type character 'R' in format [-Wformat=]
  115 |                                 fprintf(out, "%#R", list);
      |                                                 ^
uci_control.c:115:46: warning: too many arguments for format [-Wformat-extra-args]
  115 |                                 fprintf(out, "%#R", list);
      |                                              ^~~~~
uci_parser.c: In function 'section_enumerator_enumerate':
uci_parser.c:76:21: error: implicit declaration of function 'uci_lookup'; did you mean 'uci_lookup_ptr'? [-Wimplicit-function-declaration]
   76 |                 if (uci_lookup(this->ctx, &element, this->package,
      |                     ^~~~~~~~~~
      |                     uci_lookup_ptr
make[8]: *** [Makefile:603: uci_parser.lo] Error 1
make[8]: Leaving directory '/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/strongswan-5.9.14/src/libcharon/plugins/uci'
```

vortexilation, 2024-06-10:
My newbie attempt, based on here old [uci_lookup](https://git.openwrt.org/?p=project/uci.git;a=blob_plain;f=uci.h;hb=9e56d37528e25c0759185e4c263c52a1eefaa9dd) api definition: 
```
/**
 * uci_lookup: Look up an uci element
 *
 * @ctx: uci context
 * @res: where to store the result
 * @package: config package
 * @section: config section (optional)
 * @option: option to search for (optional)
 *
 * If section is omitted, then a pointer to the config package is returned
 * If option is omitted, then a pointer to the config section is returned
 */
extern int uci_lookup(struct uci_context *ctx, struct uci_element **res, char *package, char *section, char *option);
```

And possibly new [uci_lookup_next](https://github.com/openwrt/uci/blob/master/uci.h#L325) definition is : 
```
/**
 * uci_lookup_next: lookup a child element
 * @ctx: uci context
 * @e: target element pointer
 * @list: list of elements
 * @name: name of the child element
 *
 * if parent is NULL, the function looks up the package with the given name
 */
int uci_lookup_next(struct uci_context *ctx, struct uci_element **e, struct uci_list *list, const char *name);
```

or [uci_lookup_ptr](https://github.com/openwrt/uci/blob/master/uci.h#L166) : 
```
/**
 * uci_lookup_ptr: Split an uci tuple string and look up an element tree
 * @ctx: uci context
 * @ptr: lookup result struct
 * @str: uci tuple string to look up
 * @extended: allow extended syntax lookup
 *
 * if extended is set to true, uci_lookup_ptr supports the following
 * extended syntax:
 *
 * Examples:
 *   network.@interface[0].ifname ('ifname' option of the first interface section)
 *   network.@interface[-1]       (last interface section)
 * Note: uci_lookup_ptr will automatically load a config package if necessary
 * @str must not be constant, as it will be modified and used for the strings inside @ptr,
 * thus it must also be available as long as @ptr is in use.
 *
 * This function returns UCI_ERR_NOTFOUND if the package specified in the tuple
 * string cannot be found.  Otherwise it will return UCI_OK.
 *
 * Note that failures in looking up other parts, if they are also specfied,
 * including section and option, will also have a return value UCI_OK but with
 * ptr->flags * UCI_LOOKUP_COMPLETE not set.
 */
extern int uci_lookup_ptr(struct uci_context *ctx, struct uci_ptr *ptr, char *str, bool extended);
```

old code : 
```
METHOD(enumerator_t, section_enumerator_enumerate, bool,
	section_enumerator_t *this, va_list args)
{
	struct uci_element *element;
	char **value;
	int i;

	if (&this->current->list == this->list)
	{
		return FALSE;
	}

	value = va_arg(args, char**);
	if (value)
	{
		if (uci_lookup(this->ctx, &element, this->package,
					   this->current->name, "name") == UCI_OK)
		{	/* use "name" attribute as config name if available ... */
			*value = uci_to_option(element)->v.string;
		}
		else
		{	/* ... or the section name becomes config name */
			*value = uci_to_section(this->current)->type;
		}
	}

	/* followed by keyword parameters */
	for (i = 0; this->keywords[i]; i++)
	{
		value = va_arg(args, char**);
		if (value && uci_lookup(this->ctx, &element, this->package,
						  this->current->name, this->keywords[i]) == UCI_OK)
		{
			*value = uci_to_option(element)->v.string;
		}
	}

	this->current = list_to_element(this->current->list.next);
	return TRUE;
}
```

New code (untested!) : 
```
METHOD(enumerator_t, section_enumerator_enumerate, bool,
	section_enumerator_t *this, va_list args)
{
	struct uci_element *element;
	char **value;
	int i;

	if (&this->current->list == this->list)
	{
		return FALSE;
	}

	value = va_arg(args, char**);
	if (value)
	{
		if (uci_lookup_next(this->ctx, &element, &this->current->list,
					   this->current->name) == UCI_OK)
		{	/* use "name" attribute as config name if available ... */
			*value = uci_to_option(element)->v.string;
		}
		else
		{	/* ... or the section name becomes config name */
			*value = uci_to_section(this->current)->type;
		}
	}

	/* followed by keyword parameters */
	for (i = 0; this->keywords[i]; i++)
	{
		value = va_arg(args, char**);
		if (value && uci_lookup_next(this->ctx, &element, &this->current->list,
						  this->current->name) == UCI_OK)
		{
			*value = uci_to_option(element)->v.string;
		}
	}

	this->current = list_to_element(this->current->list.next);
	return TRUE;
}
```

patch (untested!) 0905-new-uci-api.patch : 
```
diff -Naur a/src/libcharon/plugins/uci/uci_parser.c b/src/libcharon/plugins/uci/uci_parser.c
--- a/src/libcharon/plugins/uci/uci_parser.c
+++ b/src/libcharon/plugins/uci/uci_parser.c
@@ -73,8 +73,8 @@
 	value = va_arg(args, char**);
 	if (value)
 	{
-		if (uci_lookup(this->ctx, &element, this->package,
-					   this->current->name, "name") == UCI_OK)
+		if (uci_lookup_next(this->ctx, &element, &this->current->list,
+					   this->current->name) == UCI_OK)
 		{	/* use "name" attribute as config name if available ... */
 			*value = uci_to_option(element)->v.string;
 		}
@@ -88,8 +88,8 @@
 	for (i = 0; this->keywords[i]; i++)
 	{
 		value = va_arg(args, char**);
-		if (value && uci_lookup(this->ctx, &element, this->package,
-						  this->current->name, this->keywords[i]) == UCI_OK)
+		if (value && uci_lookup_next(this->ctx, &element, &this->current->list,
+						  this->current->name) == UCI_OK)
 		{
 			*value = uci_to_option(element)->v.string;
 		}

```

Ping @neheb @nbd168

mkrle, 2025-01-14:
> My newbie attempt, based on here old [uci_lookup](https://git.openwrt.org/?p=project/uci.git;a=blob_plain;f=uci.h;hb=9e56d37528e25c0759185e4c263c52a1eefaa9dd) api definition:

As far as I can tell this was never reviewed/accepted?