openwrt/packages #9898: No golang for mips?

ardevd, 2019-09-01:
I can't seem to find the golang package for mips on OpenWrt. The golang package doesn't show up at all.

```
opkg install golang
Unknown package 'golang'.
Collected errors:
 * opkg_install_cmd: Cannot install package golang.
```

What am I missing here?

jefferyto, 2019-09-02:
Are you using 18.06?

ardevd, 2019-09-03:
Yes, I'm on 18.06.

jefferyto, 2019-09-03:
The only mips architecture that doesn't have Go packages is mips64_octeon. This is because support for mips64/mips64el was added in Go 1.11, but 18.06 has Go 1.10.

You can try taking the files from [master](https://github.com/openwrt/packages/tree/master/lang/golang) and compiling your own Go packages for mips64_octeon, though I don't know if there will be any incompatibilities with 18.06.

ardevd, 2019-09-03:
I'm on a GL iNet 750S which I don't think is mips64_octeon. I also have a 750 which also has no go package available. Does it need to be built into the image or should I be able to install it with opkg?

jefferyto, 2019-09-04:
Are you sure you are using the [default OpenWrt packages feed](https://github.com/openwrt/openwrt/blob/openwrt-18.06/feeds.conf.default#L1)?

ardevd, 2019-09-04:
I assumed so. Any way I can confirm and/or add the default package feed?

jefferyto, 2019-09-04:
On stock OpenWrt, the feeds are listed in `/etc/opkg/distfeeds.conf` and `/etc/opkg/customfeeds.conf`.

The default distfeeds.conf looks something like this:
```
src/gz openwrt_core http://downloads.openwrt.org/releases/18.06.2/targets/ar71xx/generic/packages
src/gz openwrt_base http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/base
src/gz openwrt_luci http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/luci
src/gz openwrt_packages http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/packages
src/gz openwrt_routing http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/routing
src/gz openwrt_telephony http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/telephony
```

The default customfeeds.conf only has comments.

I'm not sure if GL.iNet's build even uses these files but you can check.

feckert, 2019-09-07:
I think we could close this ticket because on openwrt-18.06 openwrt-19.07 and master we support golang.

closed 2019-09-07 by feckert