openwrt/packages #9905: mwan3: doesn't support VTI tunnel interfaces

ashway83, 2019-09-02:
Maintainer: @feckert , @luizluca 
Environment: OpenWrt 18.06.4 r7808-ef686b7292 / LuCI openwrt-18.06 branch (git-19.170.32094-4d6d8bc)

Description:
Mwan3 doesn't support VTI tunnel interfaces. The firewall rules direct ESP payload (in case of IPsec, but in theory it could be different transport payload) via VTI interface, when it should be routed based on the routing to reach the endpoint (e.g. via the wan interface).

Apparently this happens due to ESP payload already marked with iface_id of the VTI interface, as packets traveled via VTI interface before encryption.

Expected behavior:
There must be a possibility to specify interfaces for which iface_id mark would be ignored, so a new correct iface_id mark would be applied. More sophisticated approach would be to properly handle interface marks based on the interface type.

```
root@OpenWRT:~# tcpdump -i vti-nordvpn esp
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on vti-nordvpn, link-type RAW (Raw IP), capture size 262144 bytes
11:51:48.427747 IP xxx.net > 91.132.137.245: ESP(spi=0xc66b7237,seq=0x12a), length 136
11:51:49.428960 IP xxx.net > 91.132.137.245: ESP(spi=0xc66b7237,seq=0x12b), length 136
11:51:50.289907 IP xxx.net > 91.132.137.245: ESP(spi=0xc66b7237,seq=0x12c), length 136

feckert, 2019-09-25:
Thanks for your feedback. I'm not so familiar with ipsec.
So is the mark 0x500 always the same Or does this mark could change.

luizluca, 2019-09-26:
I played some year ago with VTI, but not both (mwan3 and ipsec) in the same device.

I'm guessing the issue is with outgoing traffic, not incoming. Is it correct?
Is VTI interface configured as a wan interface in mwan3?
Is there an iface_id for an interface not configured as mwan3 interface?

mwan3 might not be ready to deal with an intermediate interface (vti) that receives the traffic and redirects it (inside kernel) to one of the wan interfaces. It expects to be mwan3 duty to decide which outgoing interface to choose. OpenVPN, for example, might not affected because the tun interface returns to userland and openvpn program uses an unrelated connection to send the encrypted traffic. With ipsec, it seems that conntrack (mark) is maintained even after traffic is encrypted. It might affect any L3 tunnel interface (ip-in-ip? ipsec? wireguard?) that maintain conntrack info.

The workaround iptables seems to simply avoid the use of wman3 for ESP that originated from VTI interface. Is that correct? So, it will use only the default route (which might be down). It should, ideally, use only a working interface.

ipsec can deal with different source/dest pair for incoming and outgoing traffic, although it is normally better to have both together. However, it might get confused if the source address begin to swap (load balancing) for each connection. Is it possible to workaround this issue by simply forcing to use a fallback-only mwan3 policy? Or once iface_id is defined, mwan3 does nothing else? Anyway, mwan3 should be sticky to any ipsec traffic by default for a given source/dest pair, similar to what happens with https.

OpenWrt firewall already has some IPsec rules by default. I would say mwan3 should deal with IPsec whenever possible by default.

feckert, 2019-09-27:
> I'm guessing the issue is with outgoing traffic, not incoming. Is it correct?

Yes 

> Is VTI interface configured as a wan interface in mwan3?

From my point of view I think only physical interfaces should be configured in mwan3 as a tracking interface.

> mwan3 might not be ready to deal with an intermediate interface (vti) that receives the traffic and redirects it (inside kernel) to one of the wan interfaces. It expects to be mwan3 duty to decide [...]

Yes this is correct we also have problems with wireguard this does not work as expected with mwan3

> The workaround iptables seems to simply avoid the use of wman3 for ESP that originated from VTI interface. Is that correct? So, it will use only the default route (which might be down). It should, ideally, use only a working interface.

On my system I have for this reason the following script which changes the default route based on a default metric and so called online_metric. I am note sure if this is a elegant solution but it works for me. 
So ipsec does use always the current default route specified by the metric in the default routing table. Policy based routing does not make sense for ipsec in my scenario.

[13-mwan3.txt](https://github.com/openwrt/packages/files/3661992/13-mwan3.txt)

feckert, 2020-05-29:
@ashway83 We haven't heard from you in a while. Did my suggestion solved your issue? If not please reopen.

closed 2020-05-29 by feckert

sagarjainta, 2020-06-28:
how to create VTI interface for ipsec  in openwrt ?