openwrt/packages #9912: compile error with nfs-kernel-server

nouman8, 2019-09-02:
`SHELL= flock openwrt/tmp/.root-copy.flock -c 'cp -fpR openwrt/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
ERROR: module 'openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.69/net/sunrpc/auth_gss/rpcsec_gss_krb5.ko' is missing.
make[3]: *** [modules/fs.mk:407: openwrt/bin/targets/x86/64/packages/kmod-fs-nfs-common-rpcsec_4.19.69-1_x86_64.ipk] Error 1
make[3]: Leaving directory 'openwrt/package/kernel/linux'
time: package/kernel/linux/compile#5.30#0.45#5.76
make[2]: *** [package/Makefile:113: package/kernel/linux/compile] Error 2
make[2]: Leaving directory 'openwrt'
make[1]: *** [package/Makefile:107: openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory 'openwrt'
make: *** [openwrt/include/toplevel.mk:218: world] Error 2`

zhouruixi, 2019-10-05:
With my test, this error occurred with commit c3a78955f34a61d402044f357f54f21c75a19ff9 (kernel: move crypto-arc4 into a module).
So, I just add depend kmod-crypto-arc4 for fs-nfs-common-rpcsec, then I can compile kmod-fs-nfs-common-rpcsec withou this error.

neheb, 2020-01-17:
It seems this was fixed.

closed 2020-01-17 by neheb