openwrt/packages #9914: Tinc does not start using init.d

Henner, 2019-09-03:
I have a running version of tinc in its default foulders which I can start with tincd -D -n netname 

But it does not start as a service.

I have of course enabled the config file of tinc with but only with netname and hostname.
As I read the docs I understood i takes all it does not have from the original tinc config files

neheb, 2020-01-14:
ping @zioproto 

AFAIK, tinc does not use procd.

zioproto, 2020-01-19:
Did you enable the service as described here ?
https://openwrt.org/docs/guide-user/base-system/managing_services

```
service tinc enable
```

This should enable the service to start after reboot

neheb, 2020-02-28:
@zioproto Can tinc be transitioned to procd or does it require daemons + multiple processes?

zioproto, 2020-02-29:
@neheb I confirm tinc can be transitioned to procd because it runs a single process, and it can work on foreground.
@Henner could you please confirm the version of OpenWrt and the version of the tinc package you are using ?

Henner, 2020-03-02:
I have: 
OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06 branch (git-18.196.56128-9112198)

zioproto, 2020-10-02:
For whoever has time to implement this, I will be happy to review.

 * Here is the procd documentation: https://openwrt.org/docs/guide-developer/procd-init-scripts
 * Here is an example PR that does the same thing for another package: https://github.com/openwrt/packages/pull/11475/files

neheb, 2020-10-02:
I looked into doing this before. Got stuck on the relationship between tinc-host, tinc-net, and how there are multiple possible instances.

zioproto, 2020-10-02:
Oh no! :(  I looked at the code again, I was wrong in https://github.com/openwrt/packages/issues/9914#issuecomment-592919654

This is the file with the current config syntax: [tinc.config](https://github.com/openwrt/packages/blob/c8bec163ad00a30f8be5dde7989e445b9a37aee5/net/tinc/files/tinc.config)

It makes possible to define multiple `NETNAME`. This is needed if you connect to multiple VPNs at once.

Having multiple `tinc-net` blocks will result in 2 tinc processes to be run.

The intent of the author of the init script was to support the `tincd` parameter `-n` : 

```
     -n, --net=NETNAME
             Connect to net NETNAME.  This will let tinc read all configuration files from /etc/tinc/ NETNAME.  Specifying .
             for NETNAME is the same as not specifying any NETNAME.
````

tinc always supported having multiple configurations in different folders. ☝️ 

I am not sure this approach of a single init script is compatible with procd, because we need to run multiple tincd processes. Also to my understanding the current init script does not permit to start or stop a single tinc net, either you start/stop all nets or nothing.

Either we drop the support for multiple `NETNAME`, but this would be a huge regression and change in default behaviour, for a package that was like this for more than 6 years, or we need multiple init scripts per `NETNAME`.

ErwanMAS, 2021-06-09:
what is the benefit of procd ?

neheb, 2021-06-09:
Yes