openwrt/packages #9921: ddns-scripts: TLS encryption for update requests

TwizzyDizzy, 2019-09-04:
Maintainer: @dibdot
Environment: **arch**: Atheros AR9344 rev 2, **model**: TP-Link TL-WDR4300 v1, **OpenWrt version**: OpenWrt 18.06.4 r7808-ef686b7292

Description:

As of now, https://github.com/openwrt/packages/blob/openwrt-18.06/net/ddns-scripts/files/services has only ONE service that calls the respective DDNS service via TLS ("https").

Is there a specific reason for that? For example, I just checked my DDNS provider `afraid.org-basicauth` that also supports TLS.

If there is no specific reason, I'd send a PR for at least this provider.

Cheers
Thomas

dibdot, 2019-09-06:
Sorry, I'm not maintainer of that package - I'm not even using it yet. Probably the consideration of space consumption for SSL libraries plays a role here, but this is just a rough guess. Maybe it's better to discuss this in the forum with other users, cause this package is currently unmaintained.

TwizzyDizzy, 2019-09-14:
I see! I will try to take it to the forums then and see what comes up there. Will close when things are clear.

Cheers & Thanks for getting back despite not being the maintainer!
Thomas

rezad1393, 2019-09-22:
I think they all support https and there is an option in /etc/config/ddns that is :
        option use_https '1'

neheb, 2020-01-28:
Fixed in: https://github.com/openwrt/packages/commit/925068d4f8366240d2aeb2d69b3df12382320ec3

closed 2020-01-28 by neheb