openwrt/packages #9927: glib2: build error libelf.h: No such file or directory

feckert, 2019-09-06:
Maintainer: @tripolar @luizluca
Environment: lantiq_xrx200 x86_64 x86_geode, APU3, master openwrt-19.06

Description:

I have update my build yesterday (master / openwrt-19.06 ) and did a fresh build. I do not know what the problem is, but recently I get the following **error** on my own builds. 

> gresource-tool.c:32:20: fatal error: libelf.h: No such file or directory
>  #include <libelf.h>

I have seen in the logs before update the following messages

> /home/bbworker/bbworker/owrt_dev_x86_64/System6/build/openwrt/logs/target/linux/install.txt:1:warning: Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel

maybe this is related.

As a workaround I have installed  on debian the package libelf-dev with the command
```aptitude install libelf-dev```

I think it is a bad idea to depend on packages from the hostbuild package system. in this case **libelf-dev**.

I think it would be better that the Openwrt-buildsystem build his own libelf-dev for the host. So we do not have a build depend on the buildsystem (Distribution).

Would it make sense to build libelf-dev for the build host as well https://github.com/openwrt/openwrt/tree/master/package/libs/elfutils So we do not have do install libelf-dev from the build distribuiton?

prowong, 2019-09-06:
Same thing happened to me.

luizluca, 2019-09-07:
I didn't expect to have libelf.h from elfutils in staging/host as it does not include host-build.mk. However it was there. I discovered that its source might be tools/libelf/Makefile.

@jow- touched it some days ago (f3ab336d7ceda929dc8dc4a8b09a9552dc6be0b1). It might be what caused this issue. [libelf1](https://github.com/WolfgangSt/libelf) is a different beast from [elfutils](https://sourceware.org/elfutils/) and it seems to be a dead project.

I don't know why libelf is used and if libelf from elfutils could replace it. Make elfutils host build wouldn't be that hard.

I'm elfutils maintainer by accident in a failed attempt to port another program. I'm not using or depending on it directly. However, I'm happy to help however I can.

closed 2019-09-09 by tripolar

feckert, 2019-09-09:
@micmac1 and @tripolar thanks for your work

guidosarducci, 2021-01-24:
@feckert @luizluca  I've run into this issue with `libelf` again, while adding [OpenWRT support for BTF](https://github.com/guidosarducci/openwrt/commits/master-add-btf-kernel) (BPF Type Format) and needing to add the host tool `pahole`, which requires the modern `libelf` from `elfutils`. This seems to be the problem that keeps on giving, and worse than when you first encountered it because there are now **multiple** host `libelf` sources:
1. Host build in `tools/libelf` is based on code last updated in 2009.
2. OpenWRT [build prerequisites](https://openwrt.org/docs/guide-developer/build-system/install-buildsystem#prerequisites) include a `libelf` development package for your Linux distro.
3.  @neheb added a host build to the target package `elfutils` in f4da28c301890115e5c6b632aa6550fbc59f8e24 ("elfutils: Add host build")

> I think it is a bad idea to depend on packages from the hostbuild package system. in this case **libelf-dev**.
>
> I think it would be better that the Openwrt-buildsystem build his own libelf-dev for [...]

> I didn't expect to have libelf.h from elfutils in staging/host as it does not include host-build.mk. However it was there. I discovered that its source might be tools/libelf/Makefile.
> ...
> I don't know why libelf is used and if libelf from elfutils could replace it. Make elfutils host build wouldn't be that hard.

I wholeheartedly agree with your previous comments, and would to like to replace all the above `libelf` sources with a `tools/elfutils` host build. Does this change seem reasonable, and would you support it?

Also, I can find no record of why the old BSD-derived `libelf` was used or if it's still needed. Can you help clarify?  

Also pinging @ffainelli @nbd168 and @jow- who have previously done maintenance on the old `libelf` in case they can shed light on the history.

Thanks everyone!

neheb, 2021-01-24:
I've had a patch removing tools/libelf that I carried locally for a while. I have no idea what its purpose is.

micmac1, 2021-01-24:
Hello Tony (@guidosarducci),

To me it looks like anything under "tools/" can't have deps on anything in "package/". So if you want to build against package/libs/elfutils you'd need to move dwarves to package/.

tools/ -> staging_dir/host
package/ -> staging_dir/hostpkg

If dwarves needs to stay in tools/ then libdw-dev would need to be added as an OpenWrt prerequisite. But if you depend on the host system to provide both libdw-dev and libelf-dev then you run into the problem that the headers will be used from these but the lib used will be staging_dir/host/lib/libelf.a. So there would need to be a workaround for that. Other distros rename this lib to /usr/lib/libelf-freebsd.a to get around this.

Replacing tools/libelf with tools/elfutils may not be feasible, I don't know.

Regards,
Seb

guidosarducci, 2021-01-26:
Thanks for the comments folks.

@micmac1 Hi Seb, I should clarify that BTF support is part of the kernel, so "dwarves" (pahole) is a kernel build dependency, and thus needs to be in "tools/". Making it a system prerequisite is not reliable because of the wide variety of old versions packaged for distros.

It's true "libelf-dev" is already a system dependency and we could simply add "libdw-dev" as yet another one. Would that work in practice however? I assume managed systems like the buildbots follow the prerequisites, but it seems hit or miss with users. Wouldn't it be better to build "elfutils" as part of OpenWRT as @feckert suggests?

@neheb Hi Rosen, when you added the host build for "package/elfutils" in f4da28c301890115e5c6b632aa6550fbc59f8e24 what was the reason the "glib2" host build couldn't find the prerequisite "libelf-dev" on the system? And if we added a "tools/elfutils" then I assume we could also drop the "package/elfutils" host build?

And to be clear, were you saying previously that you've completely removed the old "tools/libelf" and it had no impact on your builds?

Overall, replacing the old "tools/libelf" with "tools/elfutils" *should* be fine for any Linux system, but I'm suspicious of e.g. Mac OSX. I recall something about "tools/libelf" being used for building stack validation in the kernel, but I've seen this is disabled on non-Linux system. The only OSX builder I know is.... (drumroll)

@ldir-EDB0 Hi Kevin, any idea if or how "tools/libelf" is needed on OSX?

neheb, 2021-01-26:
Yeah removing tools/libelf completely works for at least fedora. No idea about other distributions.

glib2 has a check for libelf that can't be disabled as far as I'm aware so it's patched out: https://github.com/openwrt/packages/blob/master/libs/glib2/patches/001-no-libelf.patch

IIRC, I added the host build and then @micmac1 fixed glib2 to not depend on the host build anymore. That was back with glib2 2.58.

guidosarducci, 2021-01-28:
Thanks for clarifying, and good to know we could also simplify the the "elfutils" target package and drop the host build. I tested this in my  [BTF branch](https://github.com/guidosarducci/openwrt/commits/master-add-btf-kernel) and confirmed "glib2" still builds OK. I also saw that dropping "tools/libelf" seem to cause no build/run issues on Linux.

I still need to understand whether "tools/libelf" can be dropped on non-Linux i.e. OSX and hoping @ldir-EDB0 can comment.

commit b6212c87 (2019-09-09): glib2: don't use libelf during host build

Since commit d3f86c9cc3e465fbca51aaadbb274856831ba56c tools/libelf
installs a pkgconfig file, which in turn is picked up during the glib2
host build. On the build bots the same occurs, although it so happens
that (sometimes?) libelf.h is not found and the host build fails. This
may be a timing issue specific to the build bots, as the header is in
fact installed by tools/libelf and I can't reproduce this here on my own
computer.

In any case, libelf (if detected) is only used by gresource. gresource
can be used without it. gresource is not used by Open [... cut]
PR #9936 (2019-09-09): glib2: don't use libelf during host build
Since commit d3f86c9cc3e465fbca51aaadbb274856831ba56c tools/libelf
installs a pkgconfig file, which in turn is picked up during the glib2
host build. On the build bots the same occurs, although it so happens
that (sometimes?) libelf.h is not found and the host build fails. This
may be a timing issue specific to the build bots, as the header is in
fact installed by tools/libelf and I can't reproduce this here on my own
computer.

In any case, libelf (if detected) is only used by gresource. gresource
can be used without it. gresource is not used by OpenW [... cut]