openwrt/packages #9928: strongswan: route table 220 is empty after successfully negotiation

liudf0716, 2019-09-06:
Maintainer: @stintel
Environment: x86/64, OpenWrt SNAPSHOT r10960-641f6b6, no-all glibc busybox
Package: strongswan-full

Description:

**responder:**
CentOS Linux release 7.6.1810 (Core)+ strongswan U5.7.2/K3.10.0-957.el7.x86_64 
ipsec.conf
```
config setup
        charondebug="cfg 4, dmn 4, ike 4, net 4"
        uniqueids=yes
        strictcrlpolicy=no

conn %default
        ikelifetime=60m
        keylife=20m
        rekeymargin=3m
        keyingtries=1
        keyexchange=ikev2
        authby=psk

conn nat-t
        left=10.152.240.153
        leftsubnet=0.0.0.0/0
        leftfirewall=yes
        right=%any
        rightsubnet=0.0.0.0/0
        auto=add
```

**initiator**
ipsec.conf
```
config setup
        charondebug="all"
        uniqueids=yes
        strictcrlpolicy=no

conn %default
        ikelifetime=60m
        keylife=20m
        rekeymargin=3m
        keyingtries=1
        keyexchange=ikev2
        authby=psk
        type=tunnel
        mobike=no

conn nat-t
        left=%defaultroute
        leftsubnet=10.10.100.0/24
        right=10.152.240.153
        rightsubnet=192.168.1.0/24
        auto=add
```
openwrt is initiator, execute the following command:

ipsec up nat-t

the result is the following
```
initiating IKE_SA nat-t[1] to 10.152.240.153
generating IKE_SA_INIT request 0 [ SA KE No N(NATD_S_IP) N(NATD_D_IP) N(FRAG_SUP) N(HASH_ALG) N(REDIR_SUP) ]
sending packet: from 192.168.100.144[500] to 10.152.240.153[500] (1080 bytes)
received packet: from 10.152.240.153[500] to 192.168.100.144[500] (272 bytes)
parsed IKE_SA_INIT response 0 [ SA KE No N(NATD_S_IP) N(NATD_D_IP) N(FRAG_SUP) N(HASH_ALG) N(MULT_AUTH) ]
selected proposal: IKE:AES_CBC_128/HMAC_SHA2_256_128/PRF_AES128_XCBC/ECP_256
local host is behind NAT, sending keep alives
no IDi configured, fall back on IP address
authentication of '192.168.100.144' (myself) with pre-shared key
establishing CHILD_SA nat-t{1}
generating IKE_AUTH request 1 [ IDi N(INIT_CONTACT) IDr AUTH SA TSi TSr N(MULT_AUTH) N(EAP_ONLY) N(MSG_ID_SYN_SUP) ]
sending packet: from 192.168.100.144[4500] to 10.152.240.153[4500] (304 bytes)
received packet: from 10.152.240.153[4500] to 192.168.100.144[4500] (208 bytes)
parsed IKE_AUTH response 1 [ IDr AUTH SA TSi TSr N(AUTH_LFT) ]
authentication of '10.152.240.153' with pre-shared key successful
IKE_SA nat-t[1] established between 192.168.100.144[192.168.100.144]...10.152.240.153[10.152.240.153]
scheduling reauthentication in 3405s
maximum IKE_SA lifetime 3585s
selected proposal: ESP:AES_CBC_128/HMAC_SHA2_256_128/NO_EXT_SEQ
CHILD_SA nat-t{1} established with SPIs c8dcaa24_i c25e6729_o and TS 10.10.100.0/24 === 192.168.1.0/24
connection 'nat-t' established successfully
```
it seems all ok, checking log find no error info. however, when checking route table 220 both in responder and openwrt side, it's empty.
ip rule show
```
0:      from all lookup local
220:    from all lookup 220
32766:  from all lookup main
32767:  from all lookup default
```

ip route show table 220

 empty

stintel, 2020-01-16:
Closing due to >3 months no response and this most likely being a configuration issue.

rimmeraj, 2021-03-30:
I seem to be having the same issue.

In 
/etc/strongswan/strongswan.d/charon.conf
sharon.conf:    install_routes = yes

jiargei, 2021-08-25:
Im' having the same issue:

root@linux:~# grep -r install_routes /etc/strongswan.*
/etc/strongswan.d/charon.conf:    install_routes = yes

stintel, 2021-08-25:
> Ubuntu 18.04 LTS

This is an OpenWrt specific repository.

stintel, 2021-08-25:
> I seem to be having the same issue.

Just to clarify, this is on OpenWrt right? If so, please specify OpenWrt target + version, strongSwan version.

closed 2025-11-14 by GeorgeSapkin