openwrt/packages #9932: mjpg-streamer: fails to build if protobuf-c is installed on the host

patrakov, 2019-09-06:
Maintainer: @thess 
Environment: Tried to build the 19.07 branch for ar71xx on Arch Linux

Description: Build of mjpg-streamer failed as follows:

```
...
-- Found protobuf-c: /usr/lib/libprotobuf-c.so  <-- this is on the host, so always wrong!
-- The following features have been enabled:

 * PLUGIN_INPUT_FILE, File input plugin
 * PLUGIN_INPUT_HTTP, HTTP input proxy plugin
 * PLUGIN_INPUT_UVC, Video 4 Linux input plugin
 * PLUGIN_OUTPUT_FILE, File output plugin
 * PLUGIN_OUTPUT_HTTP, HTTP server output plugin
 * PLUGIN_OUTPUT_RTSP, RTSP output plugin
 * PLUGIN_OUTPUT_UDP, UDP output stream plugin
 * PLUGIN_OUTPUT_ZMQSERVER, ZMQ Server output plugin  <-- this is the thingy that needs protobuf-c
...
[ 92%] Building C object plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/package.pb-c.c.o
cd /home/aep/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/_build/plugins/output_zmqserver && /home/aep/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/bin/mips-openwrt-linux-musl-gcc -Doutput_zmqserver_EXPORTS -I/home/aep/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/_build/plugins/output_zmqserver  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/aep/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25:mjpg-streamer-2018-10-25 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/aep/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/aep/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/aep/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/include -I/home/aep/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/include/fortify -I/home/aep/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/include -O3 -DNDEBUG -fPIC   -o CMakeFiles/output_zmqserver.dir/package.pb-c.c.o   -c /home/aep/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/_build/plugins/output_zmqserver/package.pb-c.c
In file included from /home/aep/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/_build/plugins/output_zmqserver/package.pb-c.c:9:0:
/home/aep/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/_build/plugins/output_zmqserver/package.pb-c.h:7:10: fatal error: protobuf-c/protobuf-c.h: No such file or directory
 #include <protobuf-c/protobuf-c.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

The full log of this package is attached.
[mjpg-streamer.log](https://github.com/openwrt/packages/files/3585592/mjpg-streamer.log)

neheb, 2019-09-21:
This should be fixed with @dangowrt 's rework (although there's now a different issue).

jaro3, 2019-10-13:
A new related issue, mjpg-streamer cannot be compiled, no help disabling the protobuf/libprotobuf nor all of the mjpg-streamer-addons....  I'm clueless, host (Arch Linux) protobuf has no effect on this issue.

make[6]: Leaving directory '/home-c/myname/devel/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
[ 85%] Built target output_udp
make -f plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/build.make plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/depend
make[6]: Entering directory '/home-c/myname/devel/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
[ 88%] Running C protocol buffer compiler on package.proto
cd /home-c/myname/devel/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/plugins/output_zmqserver && /home-c/myname/devel/openwrt/staging_dir/hostpkg/bin/protoc-c --c_out /home-c/myname/devel/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/plugins/output_zmqserver -I /home-c/myname/devel/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/plugins/output_zmqserver /home-c/myname/devel/openwrt/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/plugins/output_zmqserver/package.proto
/home-c/myname/devel/openwrt/staging_dir/hostpkg/bin/protoc-c: error while loading shared libraries: libprotoc.so.3.8.0.0: cannot open shared object file: No such file or directory
make[6]: *** [plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/build.make:62: plugins/output_zmqserver/package.pb-c.c] Error 127
...

neheb, 2019-10-13:
That...can't be right. I'm sure I fixed that issue already.

jaro3, 2019-10-13:
you mean re-download the feeds?
I pulled fresh yesterday though....

neheb, 2019-10-13:
just rm -rf build_dir/hostpkg and rm -rf staging_dir/hostpkg

jaro3, 2019-10-13:
then run "make V=sc" again?  this will be no different from what I have already done....
Isn't there any way to tell it to use LD_LIBRARY_PATH=./build_dir/target-mips_24kc_musl/protobuf-3.8.0

neheb, 2019-10-13:
I understand but this is already fixed: https://github.com/openwrt/packages/commit/7a2b6bfdb8495479142825bfc27edb12e0db8d89#diff-d75f65c006a5505188a1bf8d68200bf6

I manually verified by copying the host binaries to another location and running ldd on all of them.

jaro3, 2019-10-13:
The patch (package/feeds/packages/protobuf-c/Makefile) is there yet it fails as I wrote previously....

libprotobuf-c_1.3.2-3_mips_24kc.ipk
protobuf_3.8.0-4_mips_24kc.ipk
protobuf-lite_3.8.0-4_mips_24kc.ipk

are all built, what fails is the mjpg-streamer and subsequently luci-app-mjpg-streamer with the protoc-c: error

jaro3, 2019-10-13:
did make dirclean which removed build_dir/hostpkg and staging_dir/hostpkg...
new compilation only luci-app-mjpg-streamer with dependencies selected by menuconfig - same error:

make[6]: Leaving directory '/home-c/username/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
[ 85%] Built target output_udp
make -f plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/build.make plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/depend
make[6]: Entering directory '/home-c/username/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
[ 88%] Running C protocol buffer compiler on package.proto
cd /home-c/username/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/plugins/output_zmqserver && /home-c/username/devel/openwrt.r11192/staging_dir/hostpkg/bin/protoc-c --c_out /home-c/username/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/plugins/output_zmqserver -I /home-c/username/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/plugins/output_zmqserver /home-c/username/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/plugins/output_zmqserver/package.proto
/home-c/username/devel/openwrt.r11192/staging_dir/hostpkg/bin/protoc-c: error while loading shared libraries: libprotoc.so.3.8.0.0: cannot open shared object file: No such file or directory
make[6]: *** [plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/build.make:62: plugins/output_zmqserver/package.pb-c.c] Error 127
make[6]: Leaving directory '/home-c/username/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
make[5]: *** [CMakeFiles/Makefile2:458: plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/all] Error 2
make[5]: Leaving directory '/home-c/username/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
make[4]: *** [Makefile:130: all] Error 2
make[4]: Leaving directory '/home-c/username/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
make[3]: *** [Makefile:283: /home-c/username/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/.built] Error 2
make[3]: Leaving directory '/home-c/username/devel/openwrt.r11192/feeds/packages/multimedia/mjpg-streamer'
time: package/feeds/packages/mjpg-streamer/compile#0.09#0.04#0.15
make[2]: *** [package/Makefile:113: package/feeds/packages/mjpg-streamer/compile] Error 2
make[2]: Leaving directory '/home-c/username/devel/openwrt.r11192'
make[1]: *** [package/Makefile:107: /home-c/username/devel/openwrt.r11192/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home-c/username/devel/openwrt.r11192'
make: *** [/home-c/username/devel/openwrt.r11192/include/toplevel.mk:227: world] Error 2

neheb, 2019-10-13:
I'm confident it's something on your end. mjpg-streamer does not fail on the buildbots.

I will test again to make sure.

Can you try adding ```-DCMAKE_BUILD_WITH_INSTALL_RPATH=ON``` to CMAKE_OPTIONS? I had to do this for a different package to fix this very issue. Yet I'm confident I tested this already.

jaro3, 2019-10-14:
Which (Make)file do I add this option in please?  I am still not familiar with this openwrt build system.  I added:

CMAKE_COMMAND = /home-c/username/devel/openwrt.r11192/staging_dir/host/bin/cmake -DCMAKE_BUILD_WITH_INSTALL_RPATH=ON
 
to build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/Makefile but still the same:

make package/mjpg-streamer/compile V=s

make[5]: Entering directory '/home-c/username/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
[ 88%] Running C protocol buffer compiler on package.proto
/home-c/username/devel/openwrt.r11192/staging_dir/hostpkg/bin/protoc-c: error while loading shared libraries: libprotoc.so.3.8.0.0: cannot open shared object file: No such file or directory
make[5]: *** [plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/build.make:62: plugins/output_zmqserver/package.pb-c.c] Error 127

jaro3, 2019-10-14:
Speaking of buildbots I did not install all the feeds by "scripts/feeds install -a" though, just doing something like:

scripts/feeds install -p packages -d y lua-bit32 fwknop fwknopd libfko wget bind-host bind-client ddns-scripts_nsupdate ssmtp rpcbind fping bind-host hostip tinc openssh-sftp-server \
libcap libidn2 idn2 rpcbind

scripts/feeds install -a -p luci
...

neheb, 2019-10-14:
to https://github.com/openwrt/packages/commit/7a2b6bfdb8495479142825bfc27edb12e0db8d89

be mindful of the \

jaro3, 2019-10-14:
You mean to recompile the protobuf package on the Arch build host system??

neheb, 2019-10-14:
I cleaned both build_dir and staging_dir and cannot reproduce your failure. Something is out of date on your end.

jaro3, 2019-10-14:
How do you download the sources, which version are you using?

git clone https://www.github.com/openwrt/openwrt
scripts/feeds update -a

I have this supposedly latest version & Arch openwrt-devel 0.1-1 packages installed (updated a few days ago)

jaro3, 2019-10-14:
Perhaps it's the config, can you send your config or config.diff ?  I would try to compare compilation with yours versus mine

dangowrt, 2019-10-14:
I have build this successfully on Archlinux myself before modifying the package build. Possibly protobuf-c/host build had already completed by being dragged-in by another package before. Thank you @neheb for cleaning this up, and sorry for the inconveniences caused.

neheb, 2019-10-14:
@jaro3 wipe your build_dir and staging_dir directories , update the openwrt directory (git pull) as well as feeds (./scripts/feeds/update -a && ./script/feeds/install -a), and recompile. This will fix it.

@dangowrt no need to thank me. I broke it in the first place...

jaro3, 2019-10-14:
Yes, did exactly that, same error, it's mystery...

...
[ 85%] Linking C shared library output_udp.so
...
[ 88%] Running C protocol buffer compiler on package.proto
....
openwrt.r11192/staging_dir/hostpkg/bin/protoc-c: error while loading shared libraries: libprotoc.so.3.8.0.0: cannot open shared object file: No such file or directory
make[6]: *** [plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/build.make:62: plugins/output_zmqserver/package.pb-c.c] Error 127

jaro3, 2019-10-17:
Well I compiled all again flawlessly without  mjpg-streamer....  Then added just the package mjpg-streamer and run  "make package/feeds/packages/mjpg-streamer/compile V=s"...  The error is still there, I cannot get the package mjpg-streamer compiled under any circumstances :-(

/home-c/myname/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/plugins/output_udp/output_udp.c:211:54: warning: '%s' directive output may be truncated writing up to 1023 bytes into a region of size 1022 [-Wformat-truncation=]
             snprintf(buffer1, sizeof(buffer1), "%s \"%s\"", command, udpbuffer);
                                                      ^~              ~~~~~~~~~
/home-c/myname/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/plugins/output_udp/output_udp.c:211:13: note: 'snprintf' output 4 or more bytes (assuming 1027) into a destination of size 1024
             snprintf(buffer1, sizeof(buffer1), "%s \"%s\"", command, udpbuffer);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 85%] Linking C shared library output_udp.so
make[5]: Leaving directory '/home-c/myname/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
[ 85%] Built target output_udp
make[5]: Entering directory '/home-c/myname/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
[ 88%] Running C protocol buffer compiler on package.proto
/home-c/myname/devel/openwrt.r11192/staging_dir/hostpkg/bin/protoc-c: error while loading shared libraries: libprotoc.so.3.8.0.0: cannot open shared object file: No such file or directory
make[5]: *** [plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/build.make:62: plugins/output_zmqserver/package.pb-c.c] Error 127
make[5]: Leaving directory '/home-c/myname/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
make[4]: *** [CMakeFiles/Makefile2:458: plugins/output_zmqserver/CMakeFiles/output_zmqserver.dir/all] Error 2
make[4]: Leaving directory '/home-c/myname/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
make[3]: *** [Makefile:130: all] Error 2
make[3]: Leaving directory '/home-c/myname/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25'
make[2]: *** [Makefile:283: /home-c/myname/devel/openwrt.r11192/build_dir/target-mips_24kc_musl/mjpg-streamer-2018-10-25/.built] Error 2
make[2]: Leaving directory '/home-c/myname/devel/openwrt.r11192/feeds/packages/multimedia/mjpg-streamer'
time: package/feeds/packages/mjpg-streamer/compile#1.51#0.26#1.80
make[1]: *** [package/Makefile:113: package/feeds/packages/mjpg-streamer/compile] Error 2
make[1]: Leaving directory '/home-c/myname/devel/openwrt.r11192'
make: *** [/home-c/myname/devel/openwrt.r11192/include/toplevel.mk:227: package/feeds/packages/mjpg-streamer/compile] Error 2

neheb, 2019-10-17:
I can only suggest this patch:

```
--- a/libs/protobuf-c/Makefile
+++ b/libs/protobuf-c/Makefile
@@ -49,7 +49,8 @@ endef
 CMAKE_HOST_OPTIONS += \
 	-DBUILD_SHARED_LIBS=ON \
 	-DCMAKE_SKIP_RPATH=OFF \
-	-DCMAKE_INSTALL_RPATH="${STAGING_DIR_HOSTPKG}/lib"
+	-DCMAKE_INSTALL_RPATH="${STAGING_DIR_HOSTPKG}/lib" \
+	-DCMAKE_BUILD_WITH_INSTALL_RPATH=ON
 
 CMAKE_OPTIONS += \
 	-DBUILD_SHARED_LIBS=ON

```

I had to do this for a different package.

jaro3, 2019-10-18:
Thanks but that did not work either :-(  Same error when compiling on single cpu as root:

openwrt.r11192/staging_dir/hostpkg/bin/protoc-c: error while loading shared libraries: libprotoc.so.3.8.0.0: cannot open shared object file: No such file or directory

are any of the host packages/libraries involved in the build???  That could be the culprit......

neheb, 2019-10-18:
Your CMake version could be outdated.

Again, you're the only one with this problem. Me and the buildbots cannot replicate this.

jaro3, 2019-10-18:
cmake 3.15.4-1

where can I find the buildbots .config to test ?  I'm running into the same error on 2 different hardware systems but both the latest Arch updates, perhaps it's too new for the build??

jaro3, 2019-10-20:
have you tried to re-compile the latest sources ?  I did yesterday and the error is still there as well as the patches you mentioned.  The problem seems to be that the goddam "staging_dir/hostpkg/bin/protoc-c" cannot see the f*cking "staging_dir/hostpkg/lib64/libprotoc.so.3.8.0.0" now can some code "guru" please enlighten me how to tell the sucker to look at its neighbourhood for the libprotoc.so ???  I'm giving up on the mjpg-streamer package and will rather play with motion...

[mjpg-streamer.txt](https://github.com/openwrt/packages/files/3748024/mjpg-streamer.txt)

neheb, 2019-10-20:
I have.

Again, you are the only one with this problem. Something is wrong on your end.

Try this patch:

```
--- a/libs/protobuf-c/Makefile
+++ b/libs/protobuf-c/Makefile
@@ -49,11 +49,13 @@ endef
 CMAKE_HOST_OPTIONS += \
 	-DBUILD_SHARED_LIBS=ON \
 	-DCMAKE_SKIP_RPATH=OFF \
-	-DCMAKE_INSTALL_RPATH="${STAGING_DIR_HOSTPKG}/lib"
+	-DCMAKE_INSTALL_RPATH=${STAGING_DIR_HOSTPKG}/lib
 
 CMAKE_OPTIONS += \
 	-DBUILD_SHARED_LIBS=ON
 
+HOST_LDFLAGS += -Wl,-rpath-link=${STAGING_DIR_HOSTPKG}/lib
+
 define Package/libprotobuf-c/install
 	$(INSTALL_DIR) $(1)/usr/lib
 	$(CP) $(PKG_INSTALL_DIR)/usr/lib/libprotobuf-c.so* $(1)/usr/lib/
--- a/multimedia/mjpg-streamer/Makefile
+++ b/multimedia/mjpg-streamer/Makefile
@@ -209,7 +209,7 @@ define Build/Configure
 $(call Build/Configure/Default)
 endef
 
-TARGET_LDFLAGS+= -ljpeg
+TARGET_LDFLAGS+= -ljpeg -Wl,-rpath-link=${STAGING_DIR_HOSTPKG}/lib
 
 ifeq ($(CONFIG_MJPG_STREAMER_V4L2),y)
     TARGET_CFLAGS+= -DUSE_LIBV4L2

```

This will definitely work. I put a bunch of stuff in it as I do not know which will work (rpath for protobuf-c, rpath for mjpg-streamer, or quote removal). Note that those quotes could be an issue if you are not using bash and are using some other special shell.

patrakov, 2019-10-20:
@jaro3 please contact me privately (e.g. via email, patrakov@gmail.com) regarding your build environment, I will try to set up a VM and reproduce.

jaro3, 2019-10-21:
> I have.
>
> Again, you are the only one with this problem. Something is wrong on your end.
>
> Try this patch:
>
> ```
> --- a/libs/protobuf-c/Makefile
> +++ b/libs/protobuf-c/Makefile
> @@ -49,11 +49,13 @@ endef [...]

did not work either....

openwrt_r11254/staging_dir/hostpkg/bin/protoc-c: error while loading shared libraries: libprotoc.so.3.8.0.0: cannot open shared object file: No such file or directory

neheb, 2019-11-17:
@jaro3 I managed to reproduce it.

Please try https://github.com/neheb/openwrt/commit/28bdc50540d74ffca47ae547c60d3fe3f24fb1be

This issue happens only on some hosts. Which explains why I could not replicate (I switched hosts) as well as the buildbots.

neheb, 2020-01-12:
This was merged.

closed 2020-01-12 by neheb