openwrt/packages #9935: The latest ‘Lighttpd’ update package have some problem "Undefined config variable: var.home_dir"

Mitchell-kw-Lee, 2019-09-07:
The latest ‘Lighttpd’ update package have some problem "Undefined config variable: var.home_dir"

Latest lighttpd package cannot be launched. with the error

"Undefined config variable: var.home_dir"

Need urgent fix or workaround.
This newly occurred after update.

dibdot, 2019-09-07:
Did you even notice the issue template?
Wrong subject line, no maintainer, no environment information ... most probably this ticket will rotten ...

languagegame, 2019-09-07:
Maintainer: @flyn-org
Packages: lighttpd and lighttpd-mod-webdav
Environment: openwrt-19.07 and master

There var.home_dir is used in 30-webdav.conf in this line:
`webdav.sqlite-db-name = home_dir + "/webdav.db"`

However, it is not defined elsewhere in the openwrt configuration files included with the package.  It would be nice if it was configured by default, but I was able to workaround this issue by manually defining the variable in the lighttpd.conf file, such as:
`var.home_dir          = "/var/log/lighttpd"`

Mitchell-kw-Lee, 2019-09-07:
@linaguagegame.
Thanks for the information. I also can find the workaroud as below.
But the point what I am disappointed is the problem happened that exactly same issue as 7 year ago on the another project at https://drobocommunity.m-ize.com/t/droboapp-announcement-lighttpd-1-4-30/6866.
I have to say the OpenWRT 1.8.4 is worst stability ever. OpenWRT must consider change the version number to lower on the website from 'Stable version'.

@dibdot, 
Sorry for the inconvenience, but I'm the guest of outside of community and trying to let you know the problem asap for all your reputation(I don't even understand the term and words!). And please understand that this is not a feature request but issue report that fully contain all information and not a my personal needs.

Mitchell-kw-Lee, 2019-09-07:
I think I found the part of root cause of the problem.
The lighttpd package overwrite all the user customized .conf file in /etc/lighttpd/conf.d to default conf file when call 'Opkg Updated'.
This is serious problem.

flyn-org, 2019-09-07:
I think this might be fixed in master by https://github.com/openwrt/packages/pull/8791. @Mitchell-kw-Lee, an you confirm this is what you are talking about in the previous comment?

Mitchell-kw-Lee, 2019-09-08:
> I think this might be fixed in master by #8791. @Mitchell-kw-Lee, an you confirm this is what you are talking about in the previous comment?

Erm... have no idea whether same or not. So I cannot confirm. One thing I can tell you is that there was no problem before 3th September. It happened on the new lighttpd version that between 31th August and 7th September.

BKPepe, 2019-09-10:
I would like to get sorted out this issue so it won't happen in the future because it was actually me, who updated lighttpd in `OpenWrt 18.06` due to [CVE-2018-19052](https://nvd.nist.gov/vuln/detail/CVE-2018-19052) which was there. While I was at it, I updated lighttpd to the latest version in `master` and cherry-picked it to `openwrt-19.07` branch. 

May I ask you about your lighttpd version and as well with your OpenWrt version? These details are important to us and that's why they are in the template, which you didn't fill.

You can find it in LuCI, but if you prefer to use CLI then I would appreciate if you can share the output of these two commands:

```
opkg list-installed | grep lighttpd
cat /etc/openwrt_release
```

As it was said by the maintainer of lighttpd in OpenWrt, it should be fixed in master. We most probably can include it in OpenWrt 18.06 as well, but most likely the issue will appear one more time.

Anyway, before upgrading is recommended to have a backup.

Mitchell-kw-Lee, 2019-09-10:
Hi. Thanks for the reply.

Here you go the information!

> BusyBox v1.28.4 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 18.06.4, r7808-ef686b7292
 -----------------------------------------------------
root@BarrierSystemForMitchell:~# opkg list-installed | grep lighttpd
lighttpd - 1.4.48-4
lighttpd-mod-access - 1.4.48-4
lighttpd-mod-accesslog - 1.4.48-4
lighttpd-mod-alias - 1.4.48-4
lighttpd-mod-auth - 1.4.48-4
lighttpd-mod-authn_file - 1.4.48-4
lighttpd-mod-cgi - 1.4.48-4
lighttpd-mod-evasive - 1.4.48-4
lighttpd-mod-expire - 1.4.48-4
lighttpd-mod-proxy - 1.4.48-4
lighttpd-mod-redirect - 1.4.48-4
lighttpd-mod-rewrite - 1.4.48-4
lighttpd-mod-webdav - 1.4.48-4

Mitchell-kw-Lee, 2019-09-11:
Oops, I have missed more information for you. 

Here you go again.

`# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='18.06.4'
DISTRIB_REVISION='r7808-ef686b7292'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3'
DISTRIB_DESCRIPTION='OpenWrt 18.06.4 r7808-ef686b7292'
DISTRIB_TAINTS=''
`

BKPepe, 2019-09-18:
Should be fixed in stable branch. For more details look at referenced issue. Closing.

closed 2019-09-18 by BKPepe