openwrt/packages #9954: banip: not working error in line 1

finasfinas, 2019-09-10:
```
Tue Sep 10 01:48:20 2019 user.info banIP-[0.2.0]: 15 IPSets with overall -15 IPs/Prefixes loaded successfully (Netgear Nighthawk X4S R7800, OpenWrt SNAPSHOT r10975-681acdcc54)
Tue Sep 10 01:48:36 2019 user.info banIP-[0.2.0]: start banIP processing (reload)
Tue Sep 10 01:48:36 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:38 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:38 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:38 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:38 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:39 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:39 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:39 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:40 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:41 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:41 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:41 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:43 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:44 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:44 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:44 2019 daemon.err banip.sh[30395]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:48:45 2019 user.info banIP-[0.2.0]: 16 IPSets with overall -16 IPs/Prefixes loaded successfully (Netgear Nighthawk X4S R7800, OpenWrt SNAPSHOT r10975-681acdcc54)
Tue Sep 10 01:49:16 2019 user.debug banIP-[0.2.0]: f_jsnup ::: status: running, setcnt: 0, cnt: 0
Tue Sep 10 01:49:16 2019 user.info banIP-[0.2.0]: start banIP processing (reload)
Tue Sep 10 01:49:16 2019 user.debug banIP-[0.2.0]: f_main  ::: fetch_util: /usr/bin/wget (built-in), fetch_parm: --no-cache --no-cookies --max-redirect=0 --timeout=20 --no-check-certificate -O, ssh_daemon: dropbear, interface(s): WAN, device(s):  eth0.12, all_devices: eth0 eth1 ifb0 ifb1 gre0 gretap0 erspan0 teql0 eth0.12 CasaChavesWGVPN METRIC4_WG wlan1 wlan0 ifb4eth0.12 pptp-iP, backup_dir: /tmp, mem_total: 475, mem_free: 349, max_queue: 32
Tue Sep 10 01:49:16 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: -, mode: initial, chain: banIP, ruleset: input_wan_rule forwarding_wan_rule input_lan_rule forwarding_lan_rule, ruleset_6: input_wan_rule forwarding_wan_rule input_lan_rule forwarding_lan_rule
Tue Sep 10 01:49:16 2019 user.debug banIP-[0.2.0]: f_main  ::: name: whitelist, src_on: 1
Tue Sep 10 01:49:16 2019 user.debug banIP-[0.2.0]: f_main  ::: name: whitelist_6, src_on: 0
Tue Sep 10 01:49:16 2019 daemon.err banip.sh[32645]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:49:16 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: whitelist_6, mode: flush
Tue Sep 10 01:49:16 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: whitelist_6, mode: remove, out_rc: 4
Tue Sep 10 01:49:16 2019 user.debug banIP-[0.2.0]: f_main  ::: name: blacklist, src_on: 1
Tue Sep 10 01:49:16 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: whitelist, mode: create, settype: net, setipv: inet, ruletype: src+dst, count(sum/ip/cidr): -1/-1/0, time: 0, out_rc: 1
Tue Sep 10 01:49:16 2019 user.debug banIP-[0.2.0]: f_main  ::: name: blacklist_6, src_on: 0
Tue Sep 10 01:49:16 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: blacklist, mode: create, settype: net, setipv: inet, ruletype: src+dst, count(sum/ip/cidr): 0/0/0, time: 0, out_rc: 0
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: blacklist_6, mode: flush
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: blacklist_6, mode: remove, out_rc: 4
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_main  ::: name: bogon, src_on: 1
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_main  ::: name: bogon_6, src_on: 0
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: bogon_6, mode: flush
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: bogon_6, mode: remove, out_rc: 4
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_main  ::: name: tor, src_on: 1
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_main  ::: name: threat, src_on: 1
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_main  ::: name: debl, src_on: 1
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_main  ::: name: debl_6, src_on: 0
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: debl_6, mode: flush
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: debl_6, mode: remove, out_rc: 4
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_main  ::: name: myip, src_on: 1
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_main  ::: name: myip_6, src_on: 0
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: myip_6, mode: flush
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: myip_6, mode: remove, out_rc: 4
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_main  ::: name: yoyo, src_on: 1
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_main  ::: name: zeus, src_on: 1
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_main  ::: name: sslbl, src_on: 1
Tue Sep 10 01:49:17 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: tor, mode: backup, out_rc: 0
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_main  ::: name: ransomware, src_on: 1
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_main  ::: name: feodo, src_on: 1
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: threat, mode: backup, out_rc: 0
Tue Sep 10 01:49:18 2019 daemon.err banip.sh[32645]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_main  ::: name: dshield, src_on: 1
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: zeus, mode: restore, out_rc: 0
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_main  ::: name: proxy, src_on: 1
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: bogon, mode: backup, out_rc: 0
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: zeus, mode: create, settype: ip, setipv: inet, ruletype: src, count(sum/ip/cidr): 0/0/0, time: 1, out_rc: 0
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: sslbl, mode: backup, out_rc: 0
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_main  ::: name: iblocklist, src_on: 1
Tue Sep 10 01:49:18 2019 daemon.err banip.sh[32645]: ipset v7.3: Error in line 1: The set with the given name does not exist
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: sslbl, mode: create, settype: ip, setipv: inet, ruletype: src, count(sum/ip/cidr): 0/0/0, time: 1, out_rc: 0
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_main  ::: name: drop, src_on: 1
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: ransomware, mode: backup, out_rc: 0
Tue Sep 10 01:49:18 2019 user.debug banIP-[0.2.0]: f_main  ::: name: drop_6, src_on: 0
Tue Sep 10 01:49:18 2019 daemon.err banip.sh[32645]: ipset v7.3: Error in line 1: The set with the given name does not exist
[... 153 lines cut ...]
Tue Sep 10 01:49:53 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: proxy, mode: create, settype: ip, setipv: inet, ruletype: src, count(sum/ip/cidr): -1/-1/0, time: 3, out_rc: 1
Tue Sep 10 01:49:53 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: country_6, mode: flush
Tue Sep 10 01:49:53 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: firehol2, mode: backup, out_rc: 0
[... cut ...]
Tue Sep 10 01:49:56 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: firehol2, mode: create, settype: net, setipv: inet, ruletype: src, count(sum/ip/cidr): -1/-1/0, time: 5, out_rc: 1
Tue Sep 10 01:49:56 2019 user.debug banIP-[0.2.0]: f_ipset ::: name: debl, mode: create, settype: ip, setipv: inet, ruletype: src, count(sum/ip/cidr): -1/-1/0, time: 7, out_rc: 1
Tue Sep 10 01:49:56 2019 user.info banIP-[0.2.0]: 16 IPSets with overall -16 IPs/Prefixes loaded successfully (Netgear Nighthawk X4S R7800, OpenWrt SNAPSHOT r10975-681acdcc54)
Tue Sep 10 01:49:56 2019 user.debug banIP-[0.2.0]: f_jsnup ::: status: enabled, setcnt: 16, cnt: -16

```

finasfinas, 2019-09-10:
I am running latest master, wget with openssl but also tried ufetch and curl. same error.

dibdot, 2019-09-10:
Ups, sorry for the mess - will fix it later today.

closed 2019-09-10 by dibdot