openwrt/packages #9955: ddns-scripts_cloudflare.com-v4 error of invalid format for X-Auth-Key

brunoamaral, 2019-09-10:
I am not 100% this is the place to file this. So if it's not, please point me in the right direction.

**Maintainer**: @dibdot 

**Environment**: 

Model	TP-Link Archer C7 v2
Architecture	Qualcomm Atheros QCA9558 ver 1 rev 0
Firmware Version	OpenWrt 18.06.4 r7808-ef686b7292 / LuCI openwrt-18.06 branch (git-19.170.32094-4d6d8bc)
ddns-scripts_cloudflare.com-v4 2.7.8-1	

**Description:**

After installing the script and applying the basic configuration, I get the following error:

```json 
{"success":false,"errors":
  [{"code":6003,"message":"Invalid request headers",
     "error_chain":
         [{"code":6103,"message":"Invalid format for X-Auth-Key header"}]}],
"messages":[],"result":null}
```

dibdot, 2019-09-10:
I'm not the maintainer. Please re-test with latest master.

brunoamaral, 2019-09-10:
Sorry about the mishap. I installed using the interface and thought that I was running the latest version, but just noticed that the files on my router don't match the github version.

I'll see how I can update it to your version. Thanks for pointing in the right way.

dibdot, 2019-09-10:
You'll find all snapshot related packages here, e.g. https://downloads.openwrt.org/snapshots/packages/x86_64/ (ddns-scripts is arch-independent)

brunoamaral, 2019-09-10:
Thanks. I did install 2.7.8-11 and did some more research. 

Here's what I found:

1. The password should be the API Key.
2. Upon re-submiting the form, it changes between plaintext and the url encode version of mail%40domain.com 

I'm going to be looking for alternatives but leaving this here in case someone finds it useful.

brunoamaral, 2019-09-13:
just to close the loop on this story, I did get the DDNS working using the script. The problem is that in fact the form sometimes converts %40 back to @ when you edit the settings a second time.

Another problem was that, at first, I was using my password instead of the cloudflare API Key.

neheb, 2020-01-28:
I'll close this as this doesn't seem to be an issue with ddns-scripts as much as it is with the LuCI interface.

I also pushed a fix for the cloudflare URL.

closed 2020-01-28 by neheb