openwrt/packages #9970: postfix: 100-fsstat.patch unnecessarily restricts queues on non-root mounts

flyn-org, 2019-09-12:
Maintainer: @Shulyaka 
Environment: x86_64 master 6484e631

Description:
The Postfix package applies a patch named 100-fsstat.patch. This patch adjusts Postfix so that it determines the remaining capacity of the filesystem hosting "/overlay" when checking whether the queue filesystem has enough room to process a message. The replaced default was to check the capacity of ".".

This patch makes sense for many OpenWrt installations, but it breaks many others. For example, Postfix will wrongly check the capacity of /overlay even when the system's configuration sets its queue directory to a larger, mounted disk. In other cases, OpenWrt does not use an overlay filesystem at all.

I think the patch should instead do something like this:

  1. Obtain the full path of ".".
  2. Walk /proc/self/mounts to find the closest-match mount point (m) (i.e.,, the filesystem containing ".".
  3. Determine whether m corresponds to an overlay.
  4. Pass "/overlay" to fsspace() if m corresponds to an overlay. Otherwise pass "." to fsspace().

I started a related discussion on the Postfix users mailing list:

https://marc.info/?l=postfix-users&m=156814845220945

jow-, 2019-09-12:
Is this patch even relevant anymore? Iiirc it was only very old overlayfs implementations (and `mini_fo` ones prior to that) which reported improper free space, nowadays it should be more or less transparent. At least `statvfs()` (and I assume `statfs()` too) appear to return proper free space values:

```
root@mir3g:~# lua -lnixio.fs -e 's = nixio.fs.statvfs("/etc"); print(s.bfree, s.blocks)'
23811	25259
```

Shulyaka, 2019-09-12:
Hi,

The patch was a workaround for a mysterious bug when fsstat call caused postfix to crash.
This might be no longer the case. Will you be able to remove the patch, recompile, and test if it works on your system? I will do the same on my side.

flyn-org, 2019-10-05:
@Shulyaka, I tested the package after building it without the patch. It worked. This seems to fix the original problem.

flyn-org, 2019-11-29:
@Shulyaka: ping.

Shulyaka, 2019-11-29:
Sorry, I missed your reply for some reason. Thank you for the PR!

PR #10673 (2019-11-29): postfix: remove 100-fsstat.patch
Resolves #9970.

Maintainer: @Shulyaka
commit d87f61a1 (2019-11-29): postfix: remove 100-fsstat.patch

Resolves #9970.
closed 2019-11-29 by neheb