openwrt/packages #10003: nlbwmon: Sometimes is stuck at 100% CPU

cshoredaniel, 2019-09-15:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @jow- 
Environment: ath79, CR3000, snapshot, packages master from a month to two months ago

Description:

This is more to be a placeholder until someone has more meaty information, as I noticed but haven't had time to dig into it.

Anyway I've observed that on some of my devices that nlbwmon gets stuck at 100% CPU, and this bogs down the network speeds on the router (but it'll stick at 100% even with minimal network activity).  Stopping and disabling nlbwmon gets the router back to normal.

The 100% CPU happens for long periods of time, not just a brief spike.

diizzyy, 2019-09-16:
I've noticed that issue too a few months back, never had time to track it down.

danielfdickinson, 2021-08-01:
No longer using OpenWrt; not sure if still happens or not.

closed 2021-08-01 by danielfdickinson