openwrt/packages #10004: openconnect: unknown reason 'attempt-reconnect'. Maybe vpnc-script is out of date

dengqf6, 2019-09-15:
Maintainer: @nmav 
Environment: mvebu, OpenWrt master r11006

Description:
Got an error log message when OpenConnect tried to connect after router finished startup:
`openconnect: unknown reason 'attempt-reconnect'. Maybe vpnc-script is out of date`
Then I kept getting `DTLS handshake failed: 2` until I restart the vpn interface

vpnc-scripts version is 20151220 (last update was almost 4 years ago a0391c4f758b6f3743993a3039041cd12fb862be)

nmav, 2019-09-23:
Would you like to create a PR to update it?

dengqf6, 2019-09-23:
@nmav But you're the maintainer?
And you uploaded the script directly instead of using git source in Makefile. I'm not sure if you have made any changes to the script

nmav, 2019-09-25:
Most likely all you need to add is:
```
	attempt-reconnect)
		# Invoked before each attempt to re-establish the session.
		# If the underlying physical connection changed, we might
		# be left with a route to the VPN server through the VPN
		# itself, which would need to be fixed.
		run_hooks attempt-reconnect
		;;
```

KevinWhalen, 2019-12-16:
@nmav, I am trying that hook now with `openconnect`. I do not have a great way to reproduce the issue, but have generally been getting disconnected within a day or two, since upgrading to Fedora 31, with the following message:

```
DTLS Dead Peer Detection detected dead peer!
DTLS handshake failed: Resource temporarily unavailable, try again.
CSTP Dead Peer Detection detected dead peer!
unknown reason 'attempt-reconnect'. Maybe vpnc-script is out of date
Script '/etc/vpnc/vpnc-script' returned error 1
```

~~The RPM lists <http://www.infradead.org/openconnect/> for `openconnect`. The `vpnc-script` says it should be called by `vnpc`. The `openconnect` site lists <https://www.unix-ag.uni-kl.de/~massar/vpnc/> for `vnpc`, but that site has not been updated since 2008. They also list <https://gitlab.com/openconnect/openconnect/> and <http://git.infradead.org/users/dwmw2/openconnect.git>.~~ Found, what may not be the official version, but seems to be, the matching version for `openconnect`: <http://www.infradead.org/openconnect/vpnc-script.html> links to **<http://git.infradead.org/users/dwmw2/vpnc-scripts.git/blob_plain/HEAD:/vpnc-script>**. There are quite a few difference between the Fedora 31 packaged version the one hosted by infradead.org. I do not know if only adding the hook name to the case's reason statement will be enough to fix my problem, but since this issue was the first search result, maybe linking to that newer source code will help some people out.

nmav, 2019-12-18:
The commit 80ab3fdc49f965782dcf667e727a7111942a9560 should address the issue.

closed 2019-12-18 by nmav