openwrt/packages #10024: lua-rs232 needs to support more baud rates

ec1oud, 2019-09-18:
Maintainer: @\srdgame
Environment: ramips-rt305x-rt5350f-olinuxino-evb 18.06.4

Description:
I would like https://github.com/srdgame/librs232/pull/6 to get included in the lua-rs232 package because I need to use 600 baud to communicate with a CS5490 chip.  With such a peripheral available, the olinuxino board with 2 relays becomes a nice power controller that can also monitor the power usage of whatever device it's switching.  See https://github.com/ec1oud/CS5490_Energy_Monitor/blob/master/CS5490-client/lua/test-cs5490-5.1.lua

karlp, 2019-09-23:
well, it's been merged upstream at least, now just encourage them to release it, and then just say "please update package blah" 

as it stands, this is quite out there as far as being an openwrt issue :)

neheb, 2019-10-11:
Please check https://github.com/openwrt/packages/pull/10199

neheb, 2020-01-14:
This was merged.

closed 2020-01-14 by neheb