openwrt/packages #10036: protobuf-c compile error

butaford, 2019-09-20:
Maintainer: Ken Keys @kenkeys  
Environment: (mips, netgear r6350, OpenWrt master d6bd3fd5c4)

Description:
firmware compilation:
```
command_line_interface.cc:(.text._ZN6google8protobuf8compiler20CommandLineInterface12ErrorPrinter17AddErrorOrWarningERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEiiSB_SB_RSo[_ZN6google8protobuf8compiler20CommandLineInterface12ErrorPrinter17AddErrorOrWarningERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEiiSB_SB_RSo]+0x69): undefined reference to `google::protobuf::compiler::DiskSourceTree::VirtualFileToDiskFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'/home/user/build/openwrt/staging_dir/hostpkg/lib/libprotoc.a(command_line_interface.cc.o):(.data.rel.ro._ZTIN6google8protobuf8compiler20CommandLineInterface12ErrorPrinterE[_ZTIN6google8protobuf8compiler20CommandLineInterface12ErrorPrinterE]+0x18): undefined reference to `typeinfo for google::protobuf::compiler::MultiFileErrorCollector'
collect2: error: ld returned 1 exit status
Makefile:1182: recipe for target 'protoc-c/protoc-gen-c' failed
make[4]: *** [protoc-c/protoc-gen-c] Error 1
make[4]: Leaving directory '/home/user/build/openwrt/build_dir/hostpkg/protobuf-c-1.3.2'
Makefile:67: recipe for target '/home/user/build/openwrt/build_dir/hostpkg/protobuf-c-1.3.2/.built' failed
make[3]: *** [/home/user/build/openwrt/build_dir/hostpkg/protobuf-c-1.3.2/.built] Error 2
make[3]: Leaving directory '/home/user/build/openwrt/feeds/packages/libs/protobuf-c'
time: package/feeds/packages/protobuf-c/host-compile#0.52#0.15#4.15
package/Makefile:111: recipe for target 'package/feeds/packages/protobuf-c/host/compile' failed
make[2]: *** [package/feeds/packages/protobuf-c/host/compile] Error 2
make[2]: Leaving directory '/home/user/build/openwrt'
package/Makefile:107: recipe for target '/home/user/build/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/user/build/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/build/openwrt'/home/user/build/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

butaford, 2019-09-21:
revert commit for file  libs/protobuf/Makefile https://github.com/openwrt/packages/commit/0350a1571663a2ba61c167207c3643bab0ecae5e

neheb, 2019-09-21:
I will look into this

neheb, 2019-09-21:
I found a solution. Will do some more testing before I push a fix.

neheb, 2019-10-11:
Is this still an issue?

closed 2019-11-18 by neheb