openwrt/packages #10039: mosh-* fails to compile due to protoc mismatch

flipreverse, 2019-09-20:
Maintainer: @tymmej 
Environment: OpenWRT d6bd3fd5c, IPQ8064, armv7
Description:

The makefile for the packages mosh-* uses the host protobuf compiler. There is actually no dedicated host version of the protoc build.
This leads to a compile error complaining about a protobuf version mismatch since the protobuf version is too new.  It uses version 3.8.0. Debian and Ubuntu only offer 3.6.1.
From what I can tell, the protobuf package needs a host compile target. The mosh package should then depend on it.

tymmej, 2019-09-20:
I'm not sure I'm using host package.

1) In Makefile there's `PKG_BUILD_DEPENDS:=protobuf/host`
2) I'm building on host where protoc is not installed and it works:
```
which protoc
protoc not found
```
3) there's` ./build_dir/hostpkg/protobuf-3.8.0/protoc` binary

I had similar problem (mismatch between protobuf and mosh version, but error was during runtime) and reason was that in build directory were leftovers from older protobuf compilation. `make dirclean` resolved problem.

flipreverse, 2019-09-20:
> I'm not sure that I'm using distro protoc.
> 
>     1. In Makefile there's `PKG_BUILD_DEPENDS:=protobuf/host`
> 
>     2. I'm building on host where protoc is not installed and it works:
> 
> 
> ```
> which protoc
> protoc not found
> ```
> 
>     1. there's` ./build_dir/hostpkg/protobuf-3.8.0/protoc` binary
> 
> 
> I had similar problem (mismatch between protobuf and mosh version, but error was during runtime) and reason was that in build directory were leftovers from older protobuf compilation. `make dirclean` resolved problem.

Which Makefile are you talking about? The one from mosh?
There is no such line: https://github.com/openwrt/packages/blob/master/net/mosh/Makefile

tymmej, 2019-09-20:
Oh, I see now.
I have different version locally than commited to OpenWRT.
Can you check Makefile from https://github.com/tymmej/packages/blob/mosh/utils/mosh/Makefile ?
It it works for you I'll do pull request.

flipreverse, 2019-09-20:
Yes and no. :)
I simply added `PKG_BUILD_DEPENDS:=protobuf/host`. This solved my problem.

neheb, 2019-09-21:
Probably fails because of: https://github.com/openwrt/packages/commit/0350a1571663a2ba61c167207c3643bab0ecae5e

I'll add the dependency to mosh.

neheb, 2019-10-11:
Fix pushed.

closed 2019-10-11 by neheb