openwrt/packages #10056: adblock:

claell, 2019-09-23:
Maintainer: @dibdot 
Environment: Xiaomi Router 3G, OpenWrt 18.06.4

Description:
I am getting the common `blocklist with overall 0 domains loaded successfully` message after installing and enabling `adblock`. I found some suggestions to install `libustream-ssl-*` libraries and it is also mentioned somewhere in the README.

However, I don't like to simply try to install something until a problem is fixed, also there is no real error message in the log.

Output of debug logging:
```
Mon Sep 23 15:05:55 2019 user.info adblock-3.5.5-2[14929]: adblock instance started ::: action: reload, priority: 0, pid: 14929
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_jsnup::: status: running, mode: normal mode, cnt: 0, notify: 0, notify_cnt: 0, notify_pid: -
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_uci  ::: config: dhcp, change:
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_uci  ::: config: firewall, change:
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: dns: dnsmasq, fetch_util: /bin/uclient-fetch (-), backup: 0, backup_mode: 0, dns_jail: 0, force_srt: 0, force_dns: 0, mem_total: 253, mem_free: 211, max_queue: 4
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: adaway, enabled: 1
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: adguard, enabled: 0
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: adaway, url: https://adaway.org/hosts.txt, rc: 1, log: /bin/uclient-fetch: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: adguard, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: bitcoin, enabled: 0
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: bitcoin, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: blacklist, enabled: 1
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: disconnect, enabled: 1
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: dshield, enabled: 0
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: disconnect, url: https://s3.amazonaws.com/lists.disconnect.me/simple_malvertising.txt, rc: 1, log: /bin/uclient-fetch: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: dshield, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: feodo, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: feodo, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: hphosts, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: hphosts, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: malware, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: malware, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: malwarelist, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: malwarelist, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: openphish, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: openphish, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: ransomware, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: ransomware, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: reg_cn, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: reg_cn, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: reg_cz, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: reg_cz, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: reg_de, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: reg_de, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: reg_id, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: reg_id, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: reg_nl, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: reg_nl, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: reg_pl, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: reg_pl, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: reg_ro, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: reg_ro, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: reg_ru, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: reg_ru, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: shalla, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: shalla, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: spam404, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: spam404, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: sysctl, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: sysctl, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: urlhaus, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: urlhaus, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: ut_capitole, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: ut_capitole, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: whocares, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: whocares, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: winspy, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: winspy, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: winhelp, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: winhelp, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: youtube, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_list ::: name: youtube, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: yoyo, enabled: 1
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: zeus, enabled: 0
Mon Sep 23 15:05:56 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: yoyo, url: https://pgl.yoyo.org/adservers/serverlist.php?hostformat=nohtml&showintro=0&mimetype=plaintext, rc: 1, log: /bin/uclient-fetch: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificate [... cut ...]
Mon Sep 23 15:05:56 2019 user.info adblock-3.5.5-2[14929]: blocklist with overall 0 domains loaded successfully (Xiaomi Mi Router 3G, OpenWrt 18.06.4 r7808-ef686b7292)
```

So either this is a problem with my specific device or it is a general problem of `adblock`. I like it when applications run out of the box, so it might be good to add a dependency for those lib (that might cause problems with users not needing it and having a low disk space). At least some more documentation and log output would be really nice.

claell, 2019-09-23:
Ok, I apparently did not see this line:
`Mon Sep 23 15:05:55 2019 user.debug adblock-3.5.5-2[14929]: f_main ::: name: disconnect, url: https://s3.amazonaws.com/lists.disconnect.me/simple_malvertising.txt, rc: 1, log: /bin/uclient-fetch: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.`

However the points of offering a better documentation and maybe installing needed dependencies automatically are still there.

DazzyWalkman, 2019-09-24:
Please read the doc: https://github.com/openwrt/packages/blob/master/net/adblock/files/README.md

claell, 2019-09-24:
@DazzyWalkman 
> and it is also mentioned somewhere in the README.

I read the doc. Maybe I should have pointed this out more clearly.

I searched it now, and only found, what I already have seen when reading it:
> a download utility:
>    to support all blocklist sources a full version (with ssl support) of 'wget', 'uclient-fetch' with one of the 'libustream-*' ssl libraries, 'aria2c' or 'curl' is required

This was at least for me not documented in a way that made it easy for me to install adblock.
So most points I mentioned still stand.

dibdot, 2019-09-24:
There is no "one-size fits all" dependency, most users use uclient-fetch, others use wget or curl (and intentional not uclient-fetch!), others (on small devices with < 64 MB RAM) use http only without any SSL library. Currently the dependency is noted in the readme (Prerequisites!) and the logfile error message is good enough ...

closed 2019-09-24 by dibdot

claell, 2019-09-28:
Understood that adding a dependency will not work.

> Currently the dependency is noted in the readme

I found that, but it did not really help me.

> the logfile error message

I did not catch it the first time, since the Luci UI where I was viewing the log constantly refreshed every ~5 seconds leading to the log being scrolled down again. Also I had to scroll vertically a bit to reach the important part.

I guess it boils down to that the current documentation was not enough for me, so I will adjust it a bit to state more clearly that the default configuration needs SSL libraries installed in order to work. Also I'd like to add an example command prompt, since it was also not that convenient to lookup the package name of one of the libustream-* SSL libraries. Also I don't know whether it does not matter which is installed or whether there are better and worse ones. So could you name one package which is suggested? I will add that to the documentation then.

claell, 2019-09-28:
Created PR #10100