openwrt/packages #10060: /utils/bluez/Makefile - missing meshctl

ilfirin-ms, 2019-09-24:
From Bluez 5.47 ther should be **meshctl** utility to control new Bluetooth mesh networks, but cannot be seen it in Makefile files to install.

neheb, 2019-11-04:
ping @br101

br101, 2019-11-04:
bluez mesh support needs not only meshctl but also mesh daemon and configuration files. config option --enable-mesh can be used to activate this, but it fails to compile:

```
mesh/mesh.c:24: warning: "_GNU_SOURCE" redefined
 #define _GNU_SOURCE
 
<command-line>: note: this is the location of the previous definition
In file included from mesh/mesh.c:29:
./mesh/net.h:140:17: error: field 'sub_time' has incomplete type
  struct timeval sub_time;
```

I personally don't have the time to fix this.

JabinGP, 2021-02-19:
Can anyone make the "meshctl" work?