openwrt/packages #10069: Samba: Samba eats up all the memory

alexsmartens, 2019-09-25:
# Issue template (remove lines from top till here)

Maintainer: @nonicknamewasleftforme
Environment:  `Samba36 Version 3.6.2` at OpenWrt 18.06.4, r7808-ef686b7292 
Hardware: Linksys wrt1900acs

### Description:

After copying files by a client samba still keeps the ram. 
- BEFORE copying files:
```
free -m
             total       used       free     shared    buffers     cached
Mem:        511948     114740     397208      10148       2324      13760
-/+ buffers/cache:      98656     413292
Swap:            0          0          0
```
- Long AFTER the files are copied:
```
free -m
             total       used       free     shared    buffers     cached
Mem:        511948     482520      29428      10148       2788     375236
-/+ buffers/cache:     104496     407452
Swap:            0          0          0
```

After clearing the cash the memory frees up again:
`echo 3 > /proc/sys/vm/drop_caches`

One possibly releated issue that I found is [ FS#2237 - Samba4 and Samba 36 - Speed capped at 50Mbit/sec when used through Wi-Fi. Wired LAN works fine.](https://bugs.openwrt.org/index.php?do=details&task_id=2237&string=samba&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)

**This memory issue does not seem to be a healthy thing. What would you recommend?**

I could potentially try Samba4 but I am quite a newbie with OpenWRT. I followed the [OpenWRT Samba tutorial](https://openwrt.org/docs/guide-user/services/nas/samba_configuration) which is based on samba36 and cannot find any documentation about samba4 installation. I would appreciate any clarification/suggestion on this too

Thanks!

nonicknamewasleftforme, 2019-09-25:
Hi Alex,

I'm not sure, how you came to add my alias as maintainer? I'm not a maintainer of the samba package. 

Anyway, what you're describing is not an issue. It is normal that the kernel vm fills up caches when doing I/O. Just regard cache memory as free memory (i.e. read the second row in the "free" output), the memory can still be claimed by processes and the caches will be reduced automatically by the kernel vm as needed.

jow-, 2019-09-25:
I see nothing wrong. Your used memory increased from `98656` to `104496`.

closed 2019-09-25 by jow-