openwrt/packages #10070: transmission:Running error

shanpo, 2019-09-25:
Maintainer: @\rosenp
Environment: (x86_64 openwrt master e39d1cf34f)

transmission-daemon-openssl_2.94-6_x86_64 Unable to run as a normal user.

```
Wed Sep 25 09:19:24 2019 user.notice transmission: Starting with 4030912000 virt mem
Wed Sep 25 09:19:25 2019 daemon.err transmission-daemon[3771]: Couldn't create "//.config": Permission denied (file-posix.c:189)
Wed Sep 25 09:19:25 2019 daemon.err transmission-daemon[3771]: Couldn't create "//.config": Permission denied (file-posix.c:189)
Wed Sep 25 09:19:25 2019 daemon.err transmission-daemon[3771]: Couldn't create "//.config": Permission denied (file-posix.c:189)
Wed Sep 25 09:19:25 2019 daemon.info transmission-daemon[3771]: Transmission 2.94 (d8e60ee44f) started (session.c:740)
Wed Sep 25 09:19:25 2019 daemon.info transmission-daemon[3771]: RPC Server Adding address to whitelist: 127.0.0.1 (rpc-server.c:971)
Wed Sep 25 09:19:25 2019 daemon.info transmission-daemon[3771]: RPC Server Serving RPC and Web requests on port 127.0.0.1:9091/transmission/ (rpc-server.c:1213)
Wed Sep 25 09:19:25 2019 daemon.info transmission-daemon[3771]: RPC Server Whitelist enabled (rpc-server.c:1217)
Wed Sep 25 09:19:25 2019 daemon.info transmission-daemon[3771]: DHT Generating new id (tr-dht.c:311)
Wed Sep 25 09:19:25 2019 daemon.info transmission-daemon[3771]: Using settings from "//.config/transmission-daemon" (daemon.c:528)
Wed Sep 25 09:19:25 2019 daemon.err transmission-daemon[3771]: Couldn't save temporary file "//.config/transmission-daemon/settings.json.tmp.XXXXXX": No such file or directory (variant.c:1280)
Wed Sep 25 09:19:30 2019 daemon.err transmission-daemon[3778]: Couldn't create "//.config": Permission denied (file-posix.c:189)
Wed Sep 25 09:19:30 2019 daemon.err transmission-daemon[3778]: Couldn't create "//.config": Permission denied (file-posix.c:189)
Wed Sep 25 09:19:30 2019 daemon.err transmission-daemon[3778]: Couldn't create "//.config": Permission denied (file-posix.c:189)
Wed Sep 25 09:19:30 2019 daemon.info transmission-daemon[3778]: Transmission 2.94 (d8e60ee44f) started (session.c:740)
Wed Sep 25 09:19:30 2019 daemon.info transmission-daemon[3778]: RPC Server Adding address to whitelist: 127.0.0.1 (rpc-server.c:971)
Wed Sep 25 09:19:30 2019 daemon.info transmission-daemon[3778]: RPC Server Serving RPC and Web requests on port 127.0.0.1:9091/transmission/ (rpc-server.c:1213)
Wed Sep 25 09:19:30 2019 daemon.info transmission-daemon[3778]: RPC Server Whitelist enabled (rpc-server.c:1217)
Wed Sep 25 09:19:30 2019 daemon.err transmission-daemon[3778]: Couldn't bind port 51413 on 0.0.0.0: Address in use (Is another copy of Transmission already running?) (net.c:380)
Wed Sep 25 09:19:30 2019 daemon.err transmission-daemon[3778]: Couldn't bind port 51413 on ::: Address in use (Is another copy of Transmission already running?) (net.c:380)
Wed Sep 25 09:19:30 2019 daemon.err transmission-daemon[3778]: UDP Couldn't bind IPv4 socket (tr-udp.c:271)
Wed Sep 25 09:19:30 2019 daemon.err transmission-daemon[3778]: UDP Couldn't rebind IPv6 socket (tr-udp.c:189)
Wed Sep 25 09:19:30 2019 daemon.info transmission-daemon[3778]: DHT Generating new id (tr-dht.c:311)
Wed Sep 25 09:19:30 2019 daemon.info transmission-daemon[3778]: Using settings from "//.config/transmission-daemon" (daemon.c:528)
Wed Sep 25 09:19:30 2019 daemon.err transmission-daemon[3778]: Couldn't save temporary file "//.config/transmission-daemon/settings.json.tmp.XXXXXX": No such file or directory (variant.c:1280)
```

neheb, 2019-09-27:
Is the error still present? I pushed a fix recently.

shanpo, 2019-09-27:
The log no longer throws an error, but it still cannot read the existing configuration.

neheb, 2019-09-27:
I'll try to figure out why.

shanpo, 2019-09-28:
The new submission solves the problem.
transmission: fix command param issue and use default respawn settings

closed 2019-09-28 by neheb