openwrt/packages #10077: boost-python3: doesn't show up in menuconfig

tiagogaspar8, 2019-09-26:
PKG_MAINTAINER:=Carlos M. Ferreira <carlosmf.pt@gmail.com>
Environment: WRT3200ACM, on the latest trunk

Description:

The packages boost-python3 and boost-python don't show up ink menuconfig making for example the package kea not appearing

neheb, 2019-10-11:
ping @ClaymorePT

ClaymorePT, 2019-10-11:
@tiagogaspar8 Next time, use the handler :)
I will look into this.

Thanks @neheb

neheb, 2019-10-11:
Note that kea is fine now that the boost-python dependency is gone from kea.

tiagogaspar8, 2019-10-15:
@ClaymorePT I'm sorry I have no idea of what "the handler" is 😅😅😅
Ank @neheb is correct, kea doesn't need it as a dependency anymore, but I thought it'd be nice to report this issue

ClaymorePT, 2019-10-16:
@tiagogaspar8 I was speaking about the "@" notation :) Don't worry. The thing is, if I'm not mentioned, then I will not receive an e-mail. 

I have not forgotten about this issue. It's just that I'm currently busy with other issues and since this is no longer urgent, I have yet to check the issue.

ClaymorePT, 2019-10-21:
@neheb I cannot reproduce this issue with the latest trunk version of OpenWRT. 
I've selected the proper target system, sub target and target profile
```
Target System (Marvell EBU Armada)  --->
Subtarget (Marvell Armada 37x/38x/XP)  --->
Target Profile (Linksys WRT3200ACM)  --->
```

Then I selected the python2 and python3 language packages.

Boost.Python was available and selectable for Python2 and Python3 

What ever this was, it could have something to do with the way how the dependencies were defined in package kea.

@tiagogaspar8 Can you recheck this again with the latest version? Thank you.

ClaymorePT, 2019-10-23:
@tiagogaspar8  You also need to enable python 2 or python 3 language, since they are boost.python dependencies.
In order to see the packages available, press <Z> within Boost package selection menu. You will see that Boost.Python and Boost.Python3 are disabled due to missing dependency.

tiagogaspar8, 2019-10-24:
Ah nice trick! I see what you mean, but shouldn't it be the other way around? For example if I selected boost python it should automatically select the python package right? Sorry for my delay answering.

ClaymorePT, 2019-10-24:
@neheb if there is no other subject to discuss, I believe this issue can be closed.

tiagogaspar8, 2019-10-24:
OK I see, I'll close it then! Thanks for the help!

closed 2019-10-24 by tiagogaspar8