openwrt/packages #10082: perl: using host header while cross-compilation

ryzhovau, 2019-09-26:
Maintainer: @pprindeville, 
Environment: Debian 10.1 (stable)
Description: `Errno_pm.PL` implicitly uses host headers via `$sysroot/usr/include` because $sysroot [defined as empty](https://github.com/openwrt/packages/blob/master/lang/perl/files/base.config#L941). [This fix](https://github.com/Entware/entware-packages/commit/18697ca0900c63f178bae9c8e311af396577a3b4) may be useful.

```
make[4]: Entering directory '/home/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/perl/perl-5.28.1/ext/Errno'
"../../../../../../staging_dir/hostpkg/usr/bin/perl" "-I../../lib" Errno_pm.PL Errno.pm
/usr/include/errno.h:48:10: fatal error: bits/types/error_t.h: No such file or directory
```

neheb, 2019-11-04:
I suggest opening a pull request.

neheb, 2020-02-09:
Merged to master.

closed 2020-02-09 by neheb

pprindeville, 2020-05-25:
What was the PR and why doesn't it refer back to this issue?

neheb, 2020-05-25:
I manually merged. I did not create a PR.