openwrt/packages #10089: openssh-sftp-server: ChrootDirectory error

alazanjc, 2019-09-26:
MAINTAINER: @\Peter Wagner <tripolar@gmx.at>
Environment: openssh-sftp-server	ver. 7.7p1-2
                      OpenWrt ver. 18.06.2 r7676-cddd7b4c77

Description: Apparently the use of the ChrootDirectory parameter is invalid

```
Hi, I have two devices with openwrt. The first one has Chaos Calmer version 15.05 and the second one has version 18.06.2.

In both, an sftp server is installed, in the version corresponding to each openwrt.

It is intended set a root directory to limit access to certain subdirectories of the file system using the ChrootDirectory parameter.

In version 18.06.2 the use of this parameter gives an error that I cannot solve. I can only use the server without ChrootDirectory, but therefore exposing the entire file system.

I include various configuration settings and the results obtained in each case.

**The Client**: Ubuntu 16.04.6 LTS

**Server 1**:
	OpenWrt Chaos Calmer 15.05
	openssh-sftp-server ver 7.1p2-1

	passwd:
		jc:*:1003:55:Juan Carlos:/JC:/bin/false
	group:
		ftp:x:55:
	sshd_config:
		Subsystem sftp internal-sftp
		Match group ftp
                    ChrootDirectory /mnt/DiscoAV
                   ForceCommand internal-sftp

	file system /:
		drwxr-xr-x    1 root     root             0 Apr 25 02:13 mnt
	file system /mnt:
		drwxr-xr-x    7 root     root          4096 Aug 11 17:53 DiscoAV
	file system /mnt/DiscoAV
		drwxr--r--    3 jc       ftp           4096 Sep 17 21:23 JC

$ sftp jc@device1
Connected to device1.
sftp> pwd
Remote working directory: /JC
sftp> 

This configuration works fine!!

Server 2:
	OpenWrt ver. 18.06.2 r7676-cddd7b4c77
	openssh-sftp-server	ver. 7.7p1-2

	passwd:
		jc:*:1003:55:Juan Carlos:/JC:/bin/false
	group:
		ftp:x:55:
	sshd_config:
		Subsystem sftp internal-sftp
		Match group ftp
                     ChrootDirectory /mnt/DiscoAV
                     ForceCommand internal-sftp

	file system /:
		drwxr-xr-x    1 root     root             0 Apr 25 02:13 mnt
	file system /mnt:
		drwxr-xr-x    7 root     root          4096 Aug 11 17:53 DiscoAV
	file system /mnt/DiscoAV
		drwxr--r--    3 jc       ftp           4096 Sep 17 21:23 JC

$ sftp jc@device2
Connected to device2.
Couldn't canonicalize: No such file or directory
Need cwd

**Error!!!**

I change the root directory in the passwd file:

	passwd:
		jc:*:1003:55:Juan Carlos:/mnt/DiscoAV:/bin/false

New attempt:

$ sftp jc@device2
Connected to device2.
Couldn't canonicalize: No such file or directory
Need cwd

**Error!!!**

I remove the ChrootDirectory parameter:

	sshd_config:
		Subsystem sftp internal-sftp
		Match group ftp
    #    	ChrootDirectory /mnt/DiscoAV
	        ForceCommand internal-sftp

And I try again:

$ sftp jc@device1
Connected to device1.
sftp> pwd
Remote working directory: /mnt/DiscoAV
sftp> 

**The connection is good, but the entire file system is exposed!!**

If you need any other information, you just have to ask for it
```

neheb, 2020-01-14:
Is this still an isssue?

alazanjc, 2020-01-14:
Hi Rosen, I finally abandoned the sFTP option due to this problem and looked for other options.

The last time I was on the subject it was still not working properly.

In any case I see that in the repository of my openwrt version, the sFTP server version is the same.

neheb, 2020-01-14:
It should be 8.1 now. Anyway, I will close this. I noticed that Chaos Calmer is being used. That's way out of date.

closed 2020-01-14 by neheb

alazanjc, 2020-01-14:
The version that fails is **18.06**, the comments I make regarding version 15.05 are to indicate that in that, the behavior is correct.

In the repository of 18.06, the sFTP version that returns me is the one with the problem. I have verified that there is a new Openwrt version available (for less than a week). I have nothing to say about that version, it is very premature.

neheb, 2020-01-14:
Googling the error reveals that it's also a problem elsewhere. Unfortunately, I don't think anyone will work on fixing it for 18.06 as 19.07 is out.

orange47, 2021-01-15:
I have the exact same problem with latest version ( currently OpenWrt 19.07.5 ).
sad, very sad indeed.