openwrt/packages #10114: How can I set the ipt_args option, thank you

vbzc, 2019-09-30:
How can I set the ipt_args option

I want to use this function to set the port of the proxy. I tried a lot of writing methods and it is not successful. Please tell me how to set it in luci, thank you.
```
Formating code blocks by wrapping them with pairs of ```
```

neheb, 2020-01-28:
This question seems to have to do with shadowsocks-libev.

yousong, 2020-02-03:
> I want to use this function to set the port of the proxy. I tried a lot of writing methods and it is not successful. Please tell me how to set it in luci, thank you.

"port of the proxy".  Please be more specific which component's port, local port or remote servers' port?  Likely ipt_args is not the argument to achieve what you want.

yousong, 2020-02-21:
@vbzc , any updates?

vbzc, 2020-02-28:
@yousong I use the official original version of shadowsocks, which can be set to proxy the local port. openwrt version, I no longer use

yousong, 2020-03-03:
> @yousong I use the official original version of shadowsocks, which can be set to proxy the local port. openwrt version, I no longer use

Sorry, with the information provided thus far, I still do not know which part went wrong and how to help you.  There is a generic [README.md](https://github.com/openwrt/packages/blob/master/net/shadowsocks-libev/README.md#recipes) 
 with a recipe that should work.  Hope it helps

yousong, 2020-04-18:
For too long this issue receives no updates, I close it for now.  Feel free to comment here to provide further updates.

closed 2020-04-18 by yousong