openwrt/packages #10115: Is vlmcsd acceptable?

ysc3839, 2019-09-30:
Vlmcsd is a portable open-source KMS Emulator in C.
Released in https://forums.mydigitallife.net/threads/emulated-kms-servers-on-non-windows-platforms.50234/ (requires login).

But there's two issues I concern:
1. vlmcsd is a KMS emulator, which may have legal issue with Microsoft.
2. vlmcsd doesn't have an open-source license. (Which means all rights reserved?)

ysc3839, 2019-09-30:
I found that pr #2265 is about vlmcsd. So it means vlmcsd is acceptable?

ysc3839, 2019-10-02:
Ping @neheb

ryzhovau, 2019-10-03:
Not sure it's a good idea. This is a risk of ban openwrt/packages project here because of one "half-legal" package.

ysc3839, 2019-10-04:
No other reply? Well I consider it's not acceptable.

closed 2019-10-04 by ysc3839