openwrt/packages #10116: python3: LIBPL seems to be broken

jpc, 2019-09-30:
Maintainer: @commodo
Environment: mac OS host

Description:

I want to build a numpy host package to be able to build scipy (on 18.06 but IIUC the same applies to master). Unfortunately when I try to run `Build/Compile/HostPy3Mod` it fails because it cannot find `…/staging_dir/hostpkg/lib/python3.6/config-3.6-darwin/Makefile`. I have `…/staging_dir/hostpkg/lib/python3.6/config-3.6/Makefile` instead.

I traced the origins of `darwin` in this pathname and it was introduced upstream here: https://bugs.python.org/issue23968

OpenWRT has a patch that basically reverts this change:
https://github.com/openwrt/packages/blob/master/lang/python/python3/patches/016-adjust-config-paths.patch (same patch on 18.06 and master)
but it only changes the value of LIBPL in the Makefile. The same pathname is later reconstructed in the installed Python system (`get_makefile_filename` in `distutils/sysconfig.py`) using `sys.implementation._multiarch` which in turn is compiled into the binary from the `MULTIARCH` preprocessor define (which is kept the same as `PLATFORM_TRIPLET`).

It looks to me like this does not stand a chance to work correctly if `PLATFORM_TRIPLET` is not empty. OTOH if it is empty, then why force the conditional (see the linked patch)?

I'll try to drop the problematic patch and will see how the world burns... ;) I'll keep you updated.

jpc, 2019-09-30:
I dropped the LIBPL override from the patch (keeping the `_sysconfigdata` changes) and things started looking better.

Could someone help me understand the rationale behind changing `LIBPL`? I am not sure if it is a good idea to drop this change or should we modify distutils/sysconfig.py to remove the `PLATFORM_TRIPLET`?

commodo, 2019-10-01:
> Could someone help me understand the rationale behind changing `LIBPL`? I am not sure if it is a good idea to drop this change or should we modify distutils/sysconfig.py to remove the `PLATFORM_TRIPLET`?

That patch was introduced via https://github.com/openwrt/packages/commit/2244ad9f56d2ea8d2965e2a1f42ccf5937eb21a2
I remember it, but I don't remember the full-context.

General note [from my side] is that cross-building Python has usually been a voodoo-art, and never quite straightforward.
Nor was Python that great [out-of-the-box] to cross-build.
It could be that they fixed the issues with cross-building and that patch is no longer needed.

I'll try to build without it, and if it works, we can drop it.

commodo, 2023-05-27:
so, numpy has been around for a while now;

and i maintain it;

so closing this;

feel free to re-open if anything else is needed

closed 2023-05-27 by commodo