openwrt/packages #10125: qbittorrent

hong0980, 2019-10-01:
Hope to add qbittorrent

neheb, 2019-10-01:
The biggest issue as far as I can see is the libtorrent used being different than the one in rtorrent (already in here).

ryzhovau, 2019-10-03:
Tested on Entware feeds (x64/mipsel/arm), works fine on top devices. Required monstrous QT5 library to be compiled.

* [qbittorrent](https://github.com/Entware/rtndev/tree/master/qbittorrent),
* [qt5](https://github.com/Entware/rtndev/tree/master/qt5),
* [rblibtorrent](https://github.com/Entware/rtndev/tree/master/rblibtorrent).

Also `rblibtorrent` uses Boost libraries, so all this stuff may be too heavy for most of embedded devices. You need SWAP to use it on devices with 256MB RAM or less.

hong0980, 2019-10-08:
> Tested on Entware feeds (x64/mipsel/arm), works fine on top devices. It requires monstrous QT5 library to be compiled.
> 
> * [qbittorrent](https://github.com/Entware/rtndev/tree/master/qbittorrent),
> * [qt5](https://github.com/Entware/rtndev/tree/master/qt5),
> * [rblibtorrent](https://github.com/Entware/rtndev/tree/master/rblibtorrent).
> 
> Also `rblibtorrent` uses Boost libraries, so all this stuff may be too heavy for most of embedded devices. You need SWAP to use on devices with 256MB RAM or less.

Can be used as an alternate option, generally used on x64, aarch64.Thank you!

neheb, 2019-11-04:
Closing for now. qt5 is a no go for here. For now anyway.

closed 2019-11-04 by neheb