openwrt/packages #10133: USBIP does not find any USB device

jan233321, 2019-10-02:
Maintainer: Nuno Goncalves <nunojpg@gmail.com>
Environment: TP-Link TL-WDR3600 v1.5, OpenWrt 18.06.4
Description:

I've installed
`opkg install kmod-usb-ohci usbip-server usbip-client`

accordig to outdated wiki [https://openwrt.org/docs/guide-user/services/usb.iptunnel](url), it instaled automatically all needed modules.

But command
`usbip list -l`

give no devices at all, even no HUB, nothing.

Chaos Calmer found devices, and seemed to work, but I'd like to use up to date distro...

Any ideas? What debug data would be useful?
Thanks a lot,

Regards
Jan.

jan233321, 2019-10-02:
Hi all,
seems the bug is already known
[https://bugs.openwrt.org/index.php?do=details&task_id=2130](url)

with solution to manually replace one library:
[https://forum.openwrt.org/t/usb-over-ip-problems/7589/11](url)

and it works for me.

Have a nice day,
Jan.

nunojpg, 2019-10-07:
Is anyone willing to take usbip maintainership? I really don't use this for years!

morzexxx, 2020-07-17:
This id a libudev problem started in v18.
To solve copy libudev.so.1.6.3 from downloads.openwrt.org/releases/17.01.7/packages/(arc)/packages/libudev_3.2-1_(arc).ipk
and replace /usr/lib/libudev.so (or /overlay/upper/usr/lib/libudev.so for squashfs) with it.
For rpi4 (cortex a72) I used downloads.openwrt.org/releases/17.01.7/packages/aarch64_armv8-a/packages/libudev_3.2-1_aarch64_armv8-a.ipk with success.

illiliti, 2020-08-04:
#12844 will fix this bug

TaylorTWBrown, 2023-07-17:
PR is open to resolve issues with this.

https://github.com/openwrt/packages/pull/21586

closed 2023-10-08 by nunojpg