openwrt/packages #10137: openconnect:  Too many arguments on command line

rufik, 2019-10-03:
**Maintainer:** @nmav
**Environment:** (x86_64, VirtualBox 6.10, OpenWrt 19.07-SNAPSHOT  r10532-cf3b50377e)

**Description:**
It looks like openconnect client is having troubles with command line arguments when Auth Group has a space inside:
```
Thu Oct  3 15:42:54 2019 user.notice openconnect: executing 'openconnect vpn.xxxxxx.com -i vpn-ocvpn --non-inter --syslog -s /lib/netifd/vpnc-script -u mylogin --passwd-on-stdin --csd-wrapper=/etc/config/csd.sh --authgroup=GROUP WITH SPACE'
Thu Oct  3 15:42:54 2019 daemon.notice netifd: ocvpn (2855): Too many arguments on command line
Thu Oct  3 15:42:54 2019 daemon.notice netifd: ocvpn (2855): Usage:  openconnect [options] <server>
[...]
```

My config of Openconnect interface:
```

nmav, 2019-10-04:
Most likely yes. I wouldn't know how to help on that issue.

rmarkiewicz, 2019-10-04:
Any suggestions to whom it should be reported?

nmav, 2019-10-05:
For the main system there is https://bugs.openwrt.org/

For this particular problem are there other work-arounds we can use such as slashes?

rufik, 2019-10-08:
I've tried with backslashes but no go. Maybe some quotes and backslashes? Any idea to try?

jow-, 2019-10-08:
The `proto_run_command()` function is not to blame here, it is a variadic function so an invocation like `proto_run_command foo bar "baz qrx"` is properly turned into `("foo", "bar", "baz qrx")`.

The problem stems from first concatinating all arguments into a flat string, then passing this flat string unquoted to `proto_run_command` - this will cause the shell to split on `$IFS` (space, tab, newline) breaking argument values with spaces.

It is also prone to injctions, e.g. a value like `option auth_group '$(rm -rf /)'` would be happily interpolated by the shell before invoking `proto_run_command`.

jow-, 2019-10-08:
Can you give this patch a try? https://gist.github.com/jow-/f3d6687df398b5f3f472d5630356fff4

rufik, 2019-10-08:
Is this patch suitable to OpenWrt 19.07 revision r10532-cf3b50377e? Or could you just publish whole `openconnect.sh` file also, please?

jow-, 2019-10-08:
See here for the whole file: https://gist.github.com/jow-/c9b6bd14cc3832c9f6fd4974b714ea18

nmav, 2020-01-13:
@jow- would you like to send an MR? The patch doesn't have signoff-by for me to apply it.

nmav, 2020-02-02:
@jow- I'm attaching your patch in the git form. Could you add the Signoff-by line?
[0001-openconnect-allow-spaces-on-auth-groups.txt](https://github.com/openwrt/packages/files/4145410/0001-openconnect-allow-spaces-on-auth-groups.txt)

jow-, 2020-02-02:
@nmav - sorry I've only seen your inquiry now. I'm fine if you simply commit that change onder your name with me in `Suggested-by: ` if that's okay for you. Otherwise I'll open an MR tomorrow.

PR #11220 (2020-02-10): openconnect: preserve whitespace in arguments
Maintainer: @nmav 
Compile tested: -
Run tested: -

Description:

Properly quote arguments when assembling the command line and eval the
proto_run_command() invocation in order to prevent the shell from
improperly splitting the command arguments on $IFS.

Fixes: #10137
commit 900ac76b (2020-02-10): openconnect: preserve whitespace in arguments

Properly quote arguments when assembling the command line and eval the
proto_run_command() invocation in order to prevent the shell from
improperly splitting the command arguments on $IFS.

Fixes: #10137
closed 2020-02-10 by neheb