openwrt/packages #10184: MWAN3 does NOT work with an OpenVPN interface active

albertogonzalez, 2019-10-08:
Maintainer: @damadmai @feckert 

Environment:  
OpenWrt 18.06.4 r7808-ef686b7292 / LuCI openwrt-18.06 branch (git-19.170.32094-4d6d8bc), mwan3 2.6.18-1, openvpn-openssl 2.4.5-4.2

Description:  
I've spent more than a week trying to make MWAN3 work with an active OpenVPN interface with no results. 
As suggested in the guide 
<https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#openvpn> I added the following lines to the OpenVPN client config

```
route-nopull
route 0.0.0.0 0.0.0.0 vpn_gateway 20
```

I tried tens of combinations of OpenVPN/interface/firewall settings but none of them worked.

The OpenVPN interface works perfectly without MWAN3 and MWAN3 works well with no active OpenVPN interface, but as soon I start the OpenVPN interface I lose my Internet connection.

Honestly, I've run out of ideas, I don't know what else to try.

BTW, issue [#3486](https://github.com/openwrt/packages/issues/3486) looks quite similar to the problem I'm having.

hellion86, 2019-10-13:
подписываюсь под каждым словом. Хотел сделать резервный vpn канал через второго оператора. Столкнулся с

hoatienii, 2020-01-21:
same problem with me, i have mwan3 and openvpn

jamesmacwhite, 2020-05-01:
@albertogonzalez Have you tried the configuration example here:

https://www.leowkahman.com/2016/06/19/conditional-multiple-openvpn-routing-hostname-ip/

I'm running mwan3 and OpenVPN and it works OK. I have two tun interfaces with a VPN only policy that is load balanced.

```
route-nopull
route 0.0.0.0 0.0.0.0 vpn_gateway 20
```

Depending on your version of OpenVPN, you may want to update `route-nopull` to `pull-filter ignore redirect-gateway`, it won't really make a difference, but it's the OpenVPN 2.4 recommended approach, the route is actually not needed, despite what the wiki says. I found it actually can cause weird syslog errors related to recursive routing.

Give the guide above a try, it was the main resource I used when I first started looking at mwan3 and OpenVPN.

feckert, 2020-05-04:
@albertogonzalez  @hellion86 @hoatienii 
As described by @jamesmacwhite please use the newest stable release of openwrt for now this is 19.07
I know this issue on openvpn on 18.06 and before.

closed 2020-05-04 by feckert