openwrt/packages #10195: nginx: update nginx-dav-ext-module to 3.0

alex9434, 2019-10-10:
Maintainer: @Ansuel  @heil @dengqf6 @yousong @dirkfeytons @zhouruixi @Xotic750 @val-kulkov @sotux @lperkov @lucize @ja-pa @hnyman @paulfertser @donvipre @diizzyy @cotequeiroz 
    
Environment: All
Description: Nginx uses a very old version 0.1 of nginx-dav-ext-module, that does not support LOCK and UNLOCK. Please update the Makefile to the current version 3.0 that supports LOCK and UNLOCK. LOCK and UNLOCK are required by the Microsoft Windows WebDAV client.

Thank you!

yousong, 2019-10-10:
Hi @alex9434 , could be easier if it's you making the change and testing it

alex9434, 2019-10-11:
Hi @yousong,

I am very happy to try but I will need some help because I am really new to this. I have changed the Makefile in two areas:

1. Updated DEPENDS because release 3.0 has changed from libexpat to libxml2 (see readme):
`...
define Package/nginx/default
  SECTION:=net
  CATEGORY:=Network
  SUBMENU:=Web Servers/Proxies
  TITLE:=Nginx web server
  URL:=http://nginx.org/
  DEPENDS:=+NGINX_PCRE:libpcre +NGINX_SSL:libopenssl \
	+NGINX_HTTP_GZIP:zlib +NGINX_LUA:liblua +libpthread +NGINX_DAV:+libxml2 +libxslt
endef
...
`
2. Update module definition
`...
ifeq ($(CONFIG_NGINX_DAV),y)
  define Download/nginx-dav-ext-module
    VERSION:=f5e30888a256136d9c550bf1ada77d6ea78a48af
    SUBDIR:=nginx-dav-ext-module
    FILE:=nginx-dav-ext-module-$$(VERSION).tar.gz
    URL:=https://github.com/arut/nginx-dav-ext-module.git
    PROTO:=git
  endef
  $(eval $(call Download,nginx-dav-ext-module))

  define Prepare/nginx-dav-ext-module
	$(eval $(Download/nginx-dav-ext-module))
	gzip -dc $(DL_DIR)/$(FILE) | tar -C $(PKG_BUILD_DIR) $(TAR_OPTIONS)
  endef
endif
...
`
Unfortunately I get the following error when I try to compile:
`
adding module in /build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/nginx-headers-more
 + ngx_http_headers_more_filter_module was configured
adding module in /build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/nginx-brotli
 + ngx_brotli was configured
checking for PCRE library ... found
checking for PCRE JIT support ... found
checking for OpenSSL library ... found
checking for zlib library ... found
checking for libxslt ... not found
checking for libxslt in /usr/local/ ... not found
checking for libxslt in /usr/pkg/ ... not found
checking for libxslt in /opt/local/ ... not found

./configure: error: the HTTP XSLT module requires the libxml2/libxslt
libraries. You can either do not enable the module or install the libraries.
`
I don't know how to solve that error because I have already updated the DEPENDS. What am I missing?

yousong, 2019-10-11:
> 	+NGINX_HTTP_GZIP:zlib +NGINX_LUA:liblua +libpthread +NGINX_DAV:+libxml2 +libxslt

It should be `+NGINX_DAV:libxml2`.  The plus sign after colon character is not needed

alex9434, 2019-10-11:
OK, will try and report back.

alex9434, 2019-10-11:
Unfortunately I still have the same error (see below). How can I solve this?
```
adding module in /mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/nginx-naxsi/naxsi_src
 + ngx_http_naxsi_module was configured
adding module in /mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/nginx-dav-ext-module
 + ngx_http_dav_ext_module was configured
checking for PCRE library ... found
checking for PCRE JIT support ... found
checking for OpenSSL library ... found
checking for zlib library ... found
checking for libxslt ... not found
checking for libxslt in /usr/local/ ... not found
checking for libxslt in /usr/pkg/ ... not found
checking for libxslt in /opt/local/ ... not found

./configure: error: the HTTP XSLT module requires the libxml2/libxslt
libraries. You can either do not enable the module or install the libraries.

make[3]: *** [Makefile:542: /mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/.configured_3f169f2de35fbd269b47c614d3e54b1f] Error 1
make[3]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk/package/addon/nginx'
time: package/addon/nginx/ssl/compile#13.54#20.53#34.98
make[2]: *** [package/Makefile:109: package/addon/nginx/compile] Error 2
make[2]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk'
make[1]: *** [package/Makefile:103: /mnt/data/share/internal/software/openwrt-sdk/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk'
make: *** [/mnt/data/share/internal/software/openwrt-sdk/include/toplevel.mk:211: world] Error 2

```

yousong, 2019-10-11:
If I understand the nginx-dav-ext-module doc right, the libxslt is involved simply because the dav module want to reuse the search logic of `ngx_module_libs=LIBXSLT` in nginx.

https://github.com/arut/nginx-dav-ext-module/blob/f5e30888a256136d9c550bf1ada77d6ea78a48af/config#L6-L13

Adding `+NGINX_DAV:libxslt` should make the detection code happy, but it's unneeded bloat.

I would suggest patching that `config` to use `-lxml2` directly.  You can take lua-nginx for an example on how to do that.

alex9434, 2019-10-11:
Hi @yousong , do you mean to patch the makefile

https://github.com/arut/nginx-dav-ext-module/blob/f5e30888a256136d9c550bf1ada77d6ea78a48af/config#L6-L13

to comment out the line?
ngx_module_libs=LIBXSLT

Looking at the lua-nginx it seems to use patches to modify the makefile. Unfortunately I have never done that. Could you be so kind as to tell me a bit more specifically what I should do?

alex9434, 2019-10-11:
Hi @yousong , I have done an experiment. I manually edited the file:
/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/nginx-dav-ext-module
to the following:
`ngx_addon_name=ngx_http_dav_ext_module

ngx_module_type=HTTP
ngx_module_name=ngx_http_dav_ext_module

# nginx has robust builtin support for linking against
# libxml2+libxslt.  This is definitelty the right way to go if
# building nginx with the xslt module, in which case libxslt will
# be linked anyway.  In other cases libxslt is just redundant.
# If that's a big deal, libxml2 can be linked directly:
ngx_module_libs=-lxml2

#ngx_module_libs=LIBXSLT

ngx_module_srcs="$ngx_addon_dir/ngx_http_dav_ext_module.c"

. auto/module
`
Now I get the following error when I compile:
```
/mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/nginx-dav-ext-module/ngx_http_dav_ext_module.c:10:10: fatal error: libxml/parser.h: No such file or directory
 #include <libxml/parser.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [objs/Makefile:1304: objs/addon/nginx-dav-ext-module/ngx_http_dav_ext_module.o] Error 1
make[5]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1'
make[4]: *** [Makefile:8: build] Error 2
make[4]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1'
make[3]: *** [Makefile:541: /mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/.built] Error 2
make[3]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk/package/addon/nginx'
time: package/addon/nginx/ssl/compile#142.90#62.08#202.76
make[2]: *** [package/Makefile:109: package/addon/nginx/compile] Error 2
make[2]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk'
make[1]: *** [package/Makefile:103: /mnt/data/share/internal/software/openwrt-sdk/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk'
make: *** [/mnt/data/share/internal/software/openwrt-sdk/include/toplevel.mk:211: world] Error 2
```
Obviously I am doing it wrong. Can you help me to do it correctly?

cotequeiroz, 2019-10-11:
You need to add
```
ngx_module_incs=-I$(STAGING_DIR)/usr/include/libxml2
```
to the `config` file, so it will find those headers.

alex9434, 2019-10-11:
Hi @cotequeiroz 
I have changed the file ./build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/nginx-dav-ext-module/config:

```
ngx_addon_name=ngx_http_dav_ext_module
ngx_module_type=HTTP
ngx_module_name=ngx_http_dav_ext_module
ngx_module_libs=-lxml2
ngx_module_incs=$(STAGING_DIR)/usr/include/libxml2
#ngx_module_libs=LIBXSLT
ngx_module_srcs="$ngx_addon_dir/ngx_http_dav_ext_module.c"
. auto/module

```
Unfortunately I still got a similar error:

```
        -o objs/addon/nginx-dav-ext-module/ngx_http_dav_ext_module.o \
        /mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/nginx-dav-ext-module/ngx_http_dav_ext_module.c
cc1: note: someone does not honour COPTS correctly, passed 2 times
/mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/nginx-dav-ext-module/ngx_http_dav_ext_module.c:10:10: fatal error: libxml/parser.h: No such file or directory
 #include <libxml/parser.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [objs/Makefile:1304: objs/addon/nginx-dav-ext-module/ngx_http_dav_ext_module.o] Error 1
make[5]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1'
make[4]: *** [Makefile:8: build] Error 2
make[4]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1'
make[3]: *** [Makefile:541: /mnt/data/share/internal/software/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.16.1/.built] Error 2
make[3]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk/package/addon/nginx'
time: package/addon/nginx/ssl/compile#0.93#1.50#2.12
make[2]: *** [package/Makefile:109: package/addon/nginx/compile] Error 2
make[2]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk'
make[1]: *** [package/Makefile:103: /mnt/data/share/internal/software/openwrt-sdk/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/data/share/internal/software/openwrt-sdk'
make: *** [/mnt/data/share/internal/software/openwrt-sdk/include/toplevel.mk:211: world] Error 2
```

cotequeiroz, 2019-10-11:
My bad.  Instead of creating a patch, like you did, I took a shortcut, and used `sed` in `Prepare/nginx-dav-ext-module`.  Patching is more robust.  It's easier to just add `-I$(STAGING_DIR)/usr/include/libxml2` to `TARGET_CFLAGS`,  and not have to worry about passing it to `config`.

cotequeiroz, 2019-10-11:
Won't you think it is better to continue the discussion in a Pull Request, so that we can test your code directly?  It would be harder for me to give you bad advise again 😜.
Go ahead and open one.

paulfertser, 2019-10-11:
On Fri, Oct 11, 2019 at 06:19:54AM -0700, Eneas U de Queiroz wrote:
> My bad. Instead of creating a patch, like you did, I took a shortcut, and used
> sed in Prepare/nginx-dav-ext-module. Patching is more robust. It's easier to
> just add -I$(STAGING_DIR)/usr/include/libxml2 to TARGET_CFLAGS, and not have to
> worry about passing it to config.

I suggest to not patch the DAV module at all, and to make the process
less hackish (and more futureproof) by using this in
auto/lib/libxslt/conf:

    ngx_feature_path=$(pkg-config libxslt --cflags-only-I | sed 's/-I//g')
    ngx_feature_libs=$(pkg-config libxslt --libs)

Plus obvious:

diff --git a/net/nginx/Makefile b/net/nginx/Makefile
index 5063596..64bd4f6 100644
--- a/net/nginx/Makefile
+++ b/net/nginx/Makefile
@@ -83,7 +83,7 @@ define Package/nginx/default
   TITLE:=Nginx web server
   URL:=http://nginx.org/
   DEPENDS:=+NGINX_PCRE:libpcre +NGINX_SSL:libopenssl \
-       +NGINX_HTTP_GZIP:zlib +NGINX_LUA:liblua +libpthread +NGINX_DAV:libexpat
+       +NGINX_HTTP_GZIP:zlib +NGINX_LUA:liblua +libpthread +NGINX_DAV:libxslt
 endef
 
 define Package/nginx/description
@@ -520,11 +520,11 @@ endif
 
 ifeq ($(CONFIG_NGINX_DAV),y)
   define Download/nginx-dav-ext-module
-    VERSION:=430fd774fe838a04f1a5defbf1dd571d42300cf9
+    VERSION:=v3.0.0
     SUBDIR:=nginx-dav-ext-module
     FILE:=nginx-dav-ext-module-$$(VERSION).tar.gz
     URL:=https://github.com/arut/nginx-dav-ext-module.git
-    MIRROR_HASH:=0566053a8756423ecab455fd9d218cec1e017598fcbb3d6415a06f816851611e
+    MIRROR_HASH:=052dfc5a40b3b66ce27120e3ddab2a3dc9acbcb0184554e2b07552b2c490dc72
     PROTO:=git
   endef
   $(eval $(call Download,nginx-dav-ext-module))

made it buildable here.

-- 
Be free, use free (http://www.gnu.org/philosophy/free-sw.html) software!
mailto:fercerpav@gmail.com

cotequeiroz, 2019-10-11:
> 
> I suggest to not patch the DAV module at all, and to make the process less hackish (and more futureproof) by using this in auto/lib/libxslt/conf: ngx_feature_path=$(pkg-config libxslt --cflags-only-I | sed 's/-I//g') ngx_feature_libs=$(pkg-config libxslt --libs)

But then you're installing an unnecessary library, and `nginx` will fail to build if for some reason `libxslt` gets broken.  I agree with @yousong, that we just need libxml2.  Clearly, the author's motivation for depending on libxslt is that the xslt module would likely be enabled, which is not the case here.

paulfertser, 2019-10-11:
On Fri, Oct 11, 2019 at 06:45:02AM -0700, Eneas U de Queiroz wrote:
>   I suggest to not patch the DAV module at all, and to make the process less
>   hackish (and more futureproof) by using this in auto/lib/libxslt/conf:
>   ngx_feature_path=$(pkg-config libxslt --cflags-only-I | sed 's/-I//g')
>   ngx_feature_libs=$(pkg-config libxslt --libs)
> 
> But then you're installing an unnecessary library, and nginx will fail to build
> if for some reason libxslt gets broken. I agree with [1]@yousong, that we just
> need libxml2.

I would still say using pkg-config instead of hardcoding path would
make sense there; libxml2  ships a proper .pc file (as well as libxslt
does). And I'd say both cflags and ldflags should be honoured.

And this investigation uncovered a problem with libxslt, if some nginx
module is enabled that really needs it I think the configure stage
will fail; this should be fixed regardless.

-- 
Be free, use free (http://www.gnu.org/philosophy/free-sw.html) software!
mailto:fercerpav@gmail.com

alex9434, 2019-10-11:
Hi @cotequeiroz and @paulfertser , I must admit that this discussion is now a bit above my head. What is the right way to proceed now? I have never done a pull request but I am more than willing to learn! However, I am not clear now what is the best way forward and which files specifically needs to be modified?

@paulfenster, since you mentioned that you have a working config, could you help me with a few questions:
1. Where do I make changes to auto/lib/libxslt/conf? Is this creating an additional patch file in libxlst/patches? If so, could you share the patch-file?
2. Out of curiosity: is there a quick way to get the changed hash directly from github or do I have to download and manually create a hash? 

@cotequeiroz, I understand that you suggest a different approach, that has the advantage of not requiring libxlst (this seems to be an advantage esp. for users with limited ram). Is your suggestion to change the nginx Makefile 
from
`
TARGET_CFLAGS += -fvisibility=hidden -ffunction-sections -fdata-sections -DNGX_LUA_NO_BY_LUA_BLOCK
`
to

`TARGET_CFLAGS += -fvisibility=hidden -ffunction-sections -fdata-sections -DNGX_LUA_NO_BY_LUA_BLOCK -I$(STAGING_DIR)/usr/include/libxml2
`
and revert the changes that I have done to the config file?

yousong, 2019-10-11:
The following patch for the module should work.

```
--- a/config	2018-12-17 08:45:12.000000000 +0000
+++ b/config	2019-10-11 13:49:41.446919258 +0000
@@ -8,9 +8,10 @@ ngx_module_name=ngx_http_dav_ext_module
 # building nginx with the xslt module, in which case libxslt will
 # be linked anyway.  In other cases libxslt is just redundant.
 # If that's a big deal, libxml2 can be linked directly:
-# ngx_module_libs=-lxml2
+ngx_module_libs=-lxml2
+ngx_module_incs=$(pkg-config --cflags-only-I libxml-2.0 | sed 's/^-I//')
 
-ngx_module_libs=LIBXSLT
+#ngx_module_libs=LIBXSLT
 
 ngx_module_srcs="$ngx_addon_dir/ngx_http_dav_ext_module.c"
 
 ```

NOTE: The final line with only a single whitespace is required for it to be valid patch file.

alex9434, 2019-10-11:
hi @yousong, @cotequeiroz and @paulfertser thank you for all your help! I can confirm that could successfully compile with the patch from @yousong above. Fantastic!

Ansuel, 2019-10-11:
Will do a prendere asap....

Ansuel, 2019-10-11:
@alex9434 @cotequeiroz in your option what is right way to follow about the extra lib?

cotequeiroz, 2019-10-11:
Sorry, I'm not sure I understand your question.  If you're asking me whether to include libxslt as a dependency, then I'd say no.
@yousong's patch will take care of building it with libxml2, without the need for libxslt, so we can just replace expat with libxml2, with no "extra lib".  I'd go with that.

Ansuel, 2019-10-11:
Sorry i read about 2 approach to this and got confused...

cotequeiroz, 2019-10-29:
Fixed by #10204

closed 2019-10-29 by cotequeiroz