openwrt/packages #10216: cifsd-tools:  ERROR: Unable to parse smb configuration file

lucize, 2019-10-13:
Maintainer: @Andy2244 @neheb 
Environment: x86 update to git (2019-10-08)

Description: I'm unable to use this, never been, but today I tried to manual run the cifsd and I'm getting:

```
[cifsd-worker/2788]: ERROR: Unable to parse smb configuration file
[cifsd-worker/2788]: ERROR: Failed to parse configuration files
[cifsd-manager/2787]: ERROR: WARNING: child process exited abnormally: 2788
[cifsd-worker/2789]: ERROR: Unable to parse smb configuration file
[cifsd-worker/2789]: ERROR: Failed to parse configuration files
[cifsd-manager/2787]: ERROR: WARNING: child process exited abnormally: 2789

```

first it complained about cifsdpwd.db but after creating the file it will still complain about unable to parse

my /etc/cifs/smb.conf (as symlink generated from template)

```
[global]
        netbios name = iSCSI
        server string = Cifsd on OpenWrt
        workgroup = WORKGROUP
        interfaces = eth0
        bind interfaces only = yes
        ipc timeout = 8
        deadtime = 15
        map to guest = Bad User

config share
        option name 'testshare'
        option path '/tmp'
        option guest_ok 'yes'
        option create_mask '0666'
        option dir_mask '0777'
        option writeable 'yes'
        option force_root '1'
 
```
I also tried the `/usr/sbin/cifsd --c=/etc/cifs/cifsd.config.example --n`

Please advise

ysc3839, 2019-10-13:
`/etc/cifs/cifsd.config.example` is a uci format config file. However cifsd requires ini format. You can't mix them into one file.

lucize, 2019-10-13:
@ysc3839 the smb.conf is the one presented in quotes, is that a wrong format?
Or can you paste an example?

ysc3839, 2019-10-13:
@lucize Yes, It's wrong. You should modify /etc/config/cifsd.

lucize, 2019-10-13:
@ysc3839 thank you for the hint, it worked adding the entries in /etc/config/cifsd

closed 2019-10-13 by lucize