openwrt/packages #10235: 502 Bad Gateway of nginx

panda-mute, 2019-10-14:
Maintainer: @Ansuel 
Environment: (mvebu, OpenWrt latest master version with nginx/uwsgi/cgi-io fix)

Description:

```
Mon Oct 14 15:06:38 2019 daemon.info uwsgi-emperor: workers have been inactive for more than 360 seconds (1571036798-1571036437)
Mon Oct 14 15:06:38 2019 daemon.info uwsgi-emperor: Mon Oct 14 15:06:38 2019 - [emperor] bringing back instance /etc/uwsgi/vassals/luci-webui.ini to on-demand mode
Mon Oct 14 15:06:38 2019 daemon.info uwsgi-emperor: Mon Oct 14 15:06:38 2019 - received message 0 from emperor
Mon Oct 14 15:06:38 2019 daemon.info uwsgi-emperor: SIGINT/SIGQUIT received...killing workers...
Mon Oct 14 15:06:38 2019 daemon.info uwsgi-emperor: Mon Oct 14 15:06:38 2019 - [emperor] curse the uwsgi instance /etc/uwsgi/vassals/luci-webui.ini (pid: 19560)
Mon Oct 14 15:06:39 2019 daemon.info uwsgi-emperor: worker 1 buried after 1 seconds
Mon Oct 14 15:06:39 2019 daemon.info uwsgi-emperor: goodbye to uWSGI.
Mon Oct 14 15:06:39 2019 daemon.info uwsgi-emperor: VACUUM: unix socket /var/run/luci-webui.socket removed.
Mon Oct 14 15:06:39 2019 daemon.info uwsgi-emperor: unlink(): No such file or directory [core/uwsgi.c line 1660]
Mon Oct 14 15:06:40 2019 daemon.info uwsgi-emperor: [uwsgi-emperor] /etc/uwsgi/vassals/luci-webui.ini -> back to "on demand" mode, waiting for connections on socket "/var/run//luci-webui.socket" ...

```

Then surf the router web page, report 502 Bad Gateway

root@OpenWrt:~# ls /var/run/*socket
/var/run/luci-cgi_io.socket

Ansuel, 2019-10-14:
Post your nginx.conf

Is the firmware from latest commit?

panda-mute, 2019-10-14:
'''
root@OpenWrt:~# cat /etc/nginx/nginx.conf

user nobody nogroup;
worker_processes  2;

#error_log  logs/error.log;
#error_log  logs/error.log  notice;
#error_log  logs/error.log  info;

pid        /var/run/nginx.pid;

events {
    worker_connections  1024;
}

http {
    include       mime.types;
    default_type  application/octet-stream;

    sendfile on;
    keepalive_timeout 0;

    client_body_buffer_size 10K;
    client_header_buffer_size 1k;
    client_max_body_size 1G;
    large_client_header_buffers 2 1k;

    gzip on;
    gzip_http_version 1.1;
    gzip_vary on;
    gzip_comp_level 1;
    gzip_proxied any;
    gzip_types text/plain text/css application/json application/x-javascript text/xml application/xml application/xml+rss text/javascript application/javascript image/svg+xml;

    root /www;

    server {
        listen 80 default_server;
        listen [::]:80 default_server;
        server_name  localhost;
		include ariang.conf;

        location ~* .(jpg|jpeg|png|gif|ico|css|js)$ {
            expires 365d;
        }

        include luci_uwsgi.conf;

    }

    include /etc/nginx/conf.d/*.conf;
}

'''

Ansuel, 2019-10-14:
What about the other questions?

panda-mute, 2019-10-14:
It is the  commit.

commit aeb198bb82252e937a122edb2ce47b56a28be7e2
Author: Nikos Mavrogiannopoulos <n.mavrogiannopoulos@gmail.com>
Date:   Sun Oct 13 08:06:10 2019 +0200

    Merge pull request #10208 from dengqf6/oc
    
    openconnect: update to 8.05

Ansuel, 2019-10-14:
> It is the commit.
>
> commit aeb198b
> Author: Nikos Mavrogiannopoulos n.mavrogiannopoulos@gmail.com
>
> Merge pull request #10208 from dengqf6/oc
>
> openconnect: update to 8.05

As I thought your latest is not latest ahahah
Just update the firmware / package
Master to that commit lack of a crucial change to nginx (included in later
commit)

panda-mute, 2019-10-14:
I think it include all your fix about nginx/uwsgi/cgi-io, please help to check

Ansuel, 2019-10-14:
Again if you build your firmware just update your repo...

Anyway this is missing

https://git.openwrt.org/?p=feed/packages.git;a=commit;h=b9c7a570a2342c65ad049e4e77e038b3f8f3b90a

that got merged AFTER commit that updates openconnect

panda-mute, 2019-10-14:
@Ansuel 

xw@xw-ubuntu:~/workspace/github/openwrt-master/package$ cat feeds/packages/nginx/files-luci-support/luci_uwsgi.conf
location /cgi-bin/luci {
                index  index.html;
                include uwsgi_params;
                uwsgi_param SERVER_ADDR $server_addr;
                uwsgi_modifier1 9;
                uwsgi_pass unix:////var/run/luci-webui.socket;
}
location ~ /cgi-bin/cgi-(backup|download|upload) {
                include uwsgi_params;
                uwsgi_param SERVER_ADDR $server_addr;
                uwsgi_modifier1 9;
                uwsgi_pass unix:////var/run/luci-cgi_io.socket;
}

location /luci-static {
}

panda-mute, 2019-10-14:
This bug is when idle timeout, the /var/run/luci-webui.socket been removed

Ansuel, 2019-10-14:
So a uwsgi restart fix it?
I'm using it from 2 days and never experienced this problem

panda-mute, 2019-10-14:
yes, you are right, restart uwsgi  
make it work  again.

panda-mute, 2019-10-14:
Can you check the log I pasted, do you have same log of socket remove releated?

leonardpitzu, 2019-10-15:
Sort of the same happens to me but I do not see the socket removed in the log. Restarting uwsgi brings everything back online for a while. I am on openwrt master running on mvebu

Ansuel, 2019-10-15:
@peter-stadler any hint on this?

peter-stadler, 2019-10-15:
I cannot test it right now, try removing the `vacuum` line in the uwsgi configurations /etc/uwsgi/vassals/* (For testing it, a smaller timeout could be useful, too. Reset it to a higher value after debugging)

leonardpitzu, 2019-10-15:
@peter-stadler same thing, it does not fix it

peter-stadler, 2019-10-15:
Reading https://uwsgi-docs.readthedocs.io/en/latest/OnDemandVassals.html I saw that the vassals config should not set the socket. Try to remove that ...
@Ansuel: If that works, a cleaner solution would be if the emperor removes the socket option of the vassals that are activated by socket. I don't know if this is possible.

peter-stadler, 2019-10-15:
A possible problem could be that the socket is not chmod (see `ls -al /var/run/*.socket`). And we cannot use the vacuum, can we?

peter-stadler, 2019-10-15:
If it is not working, is the uwsgi log changing? You could enable more verbose logging by uncommenting the `logger = syslog:uwsgi-luci` line in /etc/uwsgi/vassals/luci-webui.ini, too ...

Ansuel, 2019-10-15:
Consider that we must keep the socket or we can't call the app to be
started...

peter-stadler, 2019-10-15:
Yes we have to keep the socket, but it must not be created by the vassals (that is how I understand the OnDemandVassals docs). To make the ini files useable with and without emperor some [if statement]( https://uwsgi-docs.readthedocs.io/en/latest/ConfigLogic.html) could be useful ...

Ansuel, 2019-10-15:
Think the problem here is vassal socket and vacuum

panda-mute, 2019-10-16:
@Ansuel @peter-stadler 

Refer to https://stackoverflow.com/questions/45437713/uwsgi-not-restarting-after-being-idle

I try to remove the socket option in /etc/uwsgi/vassals/luci-webui.ini
`root@OpenWrt:~# cat /etc/uwsgi/vassals/luci-webui.ini 
[uwsgi]
strict = true
#socket = /var/run/luci-webui.socket
cgi-mode = true
cgi = /www/
chdir = /usr/lib/lua/luci/
buffer-size = 10000
reload-mercy = 8
max-requests = 2000
limit-as = 200
reload-on-as = 256
reload-on-rss = 192
no-orphans = true
vacuum = true
enable-threads = true
post-buffering = 8192
socket-timeout = 120
thunder-lock = true
logger = syslog:uwsgi-luci
disable-logging = true
log-format = %(addr) %(method) %(uri) => generated %(rsize) bytes in %(msecs) msecs
threads = 3
processes = 3
chmod-socket = 666
cheap = true
cheaper-algo = spare
cheaper = 1
cheaper-initial = 2
cheaper-step = 1
master = true
`

leonardpitzu, 2019-10-16:
i managed only now to get my hand on a computer
removing the socket from the vassals solves the issue

commit 4d9358bb (2019-10-17): uwsgi-cgi: fix #10235
commit 5bc14d1f (2019-10-17): Merge pull request #10263 from Ansuel/fixuwsgi

uwsgi-cgi: fix #10235
closed 2019-10-17 by hnyman