openwrt/packages #10247: perl-inline compile error

DittelHome, 2019-10-15:
Maintainer: @pprindeville, @neheb,
https://downloads.openwrt.org/releases/faillogs-18.06/mips_24kc/packages/perl-inline/check-compile.txt

neheb, 2019-10-15:
Real failure is here: https://downloads.openwrt.org/releases/faillogs-18.06/mips_24kc/packages/perl-inline/host-compile.txt

No idea how to fix.

lucize, 2019-10-15:
the problem with perl is that it's a link to the host system instead of host tools
try this in perl-inline Makefile (or other perl modules)

```
HOST_CONFIGURE_VARS+= \
        PATH="$(STAGING_DIR_HOSTPKG)/bin:$(STAGING_DIR_HOSTPKG)/usr/bin:$(PATH)"

```

DittelHome, 2019-10-15:
```
the problem with perl is that it's a link to the host system instead of host tools
try this in perl-inline Makefile (or other perl modules)
```
Who can change this ? I only want to use the compiled modules with the imagebuilder.

lucize, 2019-10-15:
this is only a proposal, I didn't test it on 18.06 branch

BKPepe, 2019-11-07:
I don't see any git reference that it was fixed. It appears that the issue with perl-inline is still there on OpenWrt 18.06.

https://downloads.openwrt.org/releases/faillogs-18.06/aarch64_cortex-a53/packages/perl-inline/check-compile.txt

```
 File '.compile' does not exist.
  Considering target file '/builder/shared-workdir/build/sdk/bin/packages/aarch64_cortex-a53/packages/perl-inline_0.80+perl5.28-1_aarch64_cortex-a53.ipk'.
   File '/builder/shared-workdir/build/sdk/bin/packages/aarch64_cortex-a53/packages/perl-inline_0.80+perl5.28-1_aarch64_cortex-a53.ipk' does not exist.
    Considering target file '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/perl/Inline-0.80/.built'.
     File '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/perl/Inline-0.80/.built' does not exist.
...
...
make[3]: *** No rule to make target '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/perl/Inline-0.80/.prepared_d9f18a609b9082bdebcb607943fa3320_18f1e190c5d53547fed41a3eaa76e9e9_check', needed by '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/perl/Inline-0.80/.prepared_d9f18a609b9082bdebcb607943fa3320_18f1e190c5d53547fed41a3eaa76e9e9'.  Stop.
```

https://downloads.openwrt.org/releases/faillogs-18.06/aarch64_cortex-a53/packages/perl-inline/host-compile.txt

```
bash: /builder/shared-workdir/build/sdk/staging_dir/hostpkg/usr/bin/perl5.28.0: Text file busy
Makefile:65: recipe for target '/builder/shared-workdir/build/sdk/build_dir/hostpkg/perl/Inline-0.80/.configured' failed
make[3]: *** [/builder/shared-workdir/build/sdk/build_dir/hostpkg/perl/Inline-0.80/.configured] Error 126
time: package/feeds/packages/perl-inline/host-compile#0.24#0.07#0.52
```

cotequeiroz, 2021-08-11:
This is an old issue, but it appears that it can be fixed with #16326.

closed 2021-08-14 by neheb