openwrt/packages #10249: modemmanger: compilation error

feckert, 2019-10-15:
Maintainer: @nickberry17 
Environment: x86_64, APU3, current OpenWrt master

Description:
I wanted to evaluate the modemmanger to replace my current setup uqmi proto with the modemmanager. On my development machine (debian 9) compilation works.
But On my compilation server machine (also debian 9) with minimal setup the compilation of modemmanger failes. I think that hast todo with the modemmanager i am using on my development machine for network connectivity so that the compilation is not independent. On the compilation server no networkmanager ist installed.

Error output of the compilation:

```

/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/bin/glib-mkenums \
        --fhead "#include <ModemManager.h>\n#ifndef __MM_ENUMS_TYPES_H__\n#define __MM_ENUMS_TYPES_H__\n" \
        --template ../../build-aux/mm-enums-template.h \
        --ftail "#endif /* __MM_ENUMS_TYPES_H__ */\n" \
        ../../include/ModemManager-enums.h > mm-enums-types.h
/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/bin/glib-mkenums \
        --fhead "#ifndef __MM_ERRORS_TYPES_H__\n#define __MM_ERRORS_TYPES_H__\n" \
        --template ../../build-aux/mm-errors-template.h \
        --ftail "#endif /* __MM_ERRORS_TYPES_H__ */\n" \
        ../../include/ModemManager-errors.h > mm-errors-types.h
/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/bin/gdbus-codegen \
        --interface-prefix org.freedesktop.ModemManager1. \
        --c-namespace=MmGdbus \
        --generate-docbook mm-gdbus-doc \
        --generate-c-code mm-gdbus-manager \
        ../../introspection/org.freedesktop.ModemManager1.xml \

Traceback (most recent call last):
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/bin/gdbus-codegen", line 53, in <module>
    from codegen import codegen_main
ImportError: No module named codegen
Makefile:1149: recipe for target 'mm-gdbus-manager.c' failed
make[6]: *** [mm-gdbus-manager.c] Error 1
make[6]: Leaving directory '/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/ModemManager-1.10.6/libmm-glib/generated'
Makefile:1250: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/ModemManager-1.10.6/libmm-glib'
Makefile:510: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/ModemManager-1.10.6'
Makefile:442: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/ModemManager-1.10.6'
Makefile:130: recipe for target '/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/ModemManager-1.10.6/.built' failed
make[2]: *** [/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/ModemManager-1.10.6/.built] Error 2
make[2]: Leaving directory '/home/bbworker/bbworker/owrt_master_x86_64/System6/build/feeds/packages/net/modemmanager'
time: package/feeds/packages/modemmanager/compile#4.80#2.28#10.70
package/Makefile:111: recipe for target 'package/feeds/packages/modemmanager/compile' failed
make[1]: *** [package/feeds/packages/modemmanager/compile] Error 2
make[1]: Leaving directory '/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt'
/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/include/toplevel.mk:225: recipe for target 'package/feeds/packages/modemmanager/compile' failed
make: *** [package/feeds/packages/modemmanager/compile] Error 2

```

aleksander0m, 2019-10-16:
> ImportError: No module named codegen

ModemManager requires the `gdbus-codegen` tool that is built/installed along with GLib's GIO. This should be a dependency on the host toolchain, e.g. same thing as for `glib-mkenums`, which is another tool required during build. In Debian I believe these tools are provided by the `libglib2.0-dev-bin` package.

feckert, 2019-10-16:
Thanks for reply. But should this not come from the buildsystem?
I want to have as few dependencies in the hostsystem as possible.
So could we add a hostbuild dependency into openwrt so that this tool get build in the buildsystem and use them?

aleksander0m, 2019-10-16:
@feckert really don't know what's best w.r.t. the actual openwrt hostsystem/buildsystem really. The tool is required to build MM, not sure exactly what's the best way to provide it. Is `glib-mkenums` built and included in the buildsystem instead of required in the host system?

If it helps, we detect the `gdbus-codegen` program during configure using `GDBUS_CODEGEN=$PKG_CONFIG --variable=gdbus_codegen gio-2.0`, in the same way as we detect the `glib-mkenums` program using `GLIB_MKENUMS=$PKG_CONFIG --variable=glib_mkenums glib-2.0`

feckert, 2019-10-18:
@aleksander0m thanks for feedback.
The problem is not that the buildsystem does not find the **glib-mkenums** and **gdbus-codegen**.
This package are correct installed and are found by the buildsystem. I think the problem is that **gdbus-codegen** wants to load **codegen_main.py** and this is not found!!

> 
> File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/bin/gdbus-codegen", line 53, in <module>
>     from codegen import codegen_main

Because its not installed during the build for the host package only the binaries are installed.

nickberry17, 2019-10-22:
Have you tried installing it via `pip` or `pip3`?  I believe I installed codegen via pip3 on my dev machine.

feckert, 2019-10-22:
No I did not and I do not want that. The codegen source comes with the package. I think I only have to install that on the right position in the staging_dir. If that is done then it should get found by the bin

aleksander0m, 2019-10-22:
@feckert Looks like the missing bits are in `/usr/share/glib-2.0/codegen` so I assume that directory must be get installed in staging_dir in the glib2 package, or something like that, otherwise gdbus-codegen will break.

nickberry17, 2019-10-23:
Have a look at this commit:
https://github.com/dankamongmen/sprezzos-world/commit/2714ecc1ede27415a40875b37fc368f338db80ba

`usr/share/glib-2.0/codegen/*py* usr/share/glib-2.0/`

BKPepe, 2019-10-25:
Same issue is present on buildbots.
https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/modemmanager/compile.txt

feckert, 2019-10-25:
A workaround could be install the dependency on the the build host.
But I do not like this dependency I think this is not an option.
The buildsystem should compile the necessary packages itself.
So we have less build host dependencies.

I think we need to install the missing python lib which are provided by glib2 either in the
Build/Installdev directive 
https://github.com/openwrt/packages/blob/00eda4bb0d9c4784bee8bf0ba87c02f954f5936e/libs/glib2/Makefile#L77-L80

or we need a new

Host/install directive in https://github.com/openwrt/packages/blob/master/libs/glib2/Makefile

I don't know which way is the right one. Is this a Host/Install or a Build/InstallDev problem.

PR #10362 (2019-10-29): modemmanager: fix compilation
MM has a problem with cross-compiling. It needs to run some tools during
compilation (glib-mkenums & gdbus-codegen) but uses pkg-config to detect
them. But like this it finds the wrong tools, the ones in
$(STAGING_DIR). The correct tools are in $(STAGING_DIR_HOSTPKG)/bin.

As a workaround this commit patches configure.ac so the correct tools
are used, the ones from glib2/host. The latter is also added to
PKG_BUILD_DEPENDS to reflect this.

libxslt/host is also added to PKG_BUILD_DEPENDS and the related
RequireCommand call is removed. This is OK to do since we have [... cut]
commit fa8ad6e6 (2019-10-29): modemmanager: fix compilation

MM has a problem with cross-compiling. It needs to run some
tools during compilation (glib-mkenums & gdbus-codegen) but uses
pkg-config to detect them. But like this it finds the wrong tools, the
ones in $(STAGING_DIR). The correct tools are in
$(STAGING_DIR_HOSTPKG)/bin.

As a workaround this commit patches configure.ac so the correct tools
are used, the ones from glib2/host. The latter is also added to
PKG_BUILD_DEPENDS to reflect this.

libxslt/host is also added to PKG_BUILD_DEPENDS and the related
RequireCommand call is removed. This is OK to do since we hav [... cut]
closed 2019-10-29 by neheb