openwrt/packages #10254: mjpg_streamer memory leak and modules folder

mgondium, 2019-10-15:
Running trunk on ath9k, the mjpg_streamer modules are placed on /usr/lib/mjpg-streamer but it searches for them on /usr/lib

```
MJPG Streamer Version.: 2.0
ERROR: could not find output plugin output_http.so
       Perhaps you want to adjust the search path with:
       # export LD_LIBRARY_PATH=/path/to/plugin/folder
       dlopen: Error loading shared library output_http.so: No such file or directory
```

I chose as a remedy to place symlinks on /usr/lib for the modules that I use.

Also, it looks like there is some memory leakage as the system goes oom and closes the program after a few hours.
Perhaps it's related to this unclean exit I'm getting when executing on the command line:

```
root@OpenWrt:~# mjpg_streamer -i "input_uvc.so -f 5" -o "output_http.so -p 8080"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 5
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: FPS coerced ......: from 5 to 15
UVCIOC_CTRL_ADD - Error at Pan (relative): Not a tty (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Not a tty (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Not a tty (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Not a tty (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Not a tty (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Not a tty (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Not a tty (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Not a tty (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Not a tty (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Not a tty (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Not a tty (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Not a tty (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Not a tty (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Not a tty (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Not a tty (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Not a tty (25)
 o: www-folder-path......: disabled
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
^Csetting signal to stop
 i: cleaning up resources allocated by input thread
 o: cleaning up resources allocated by server thread #00
force cancellation of threads and cleanup resources
Segmentation fault
root@OpenWrt:~#
```

I'll post the oom when it happens again.

mgondium, 2019-10-17:
```
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: starting application
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: MJPG Streamer Version.: 2.0
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: Using V4L2 device.: /dev/video0
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: Desired Resolution: 640 x 480
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: Frames Per Second.: 3
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: Format............: JPEG
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: TV-Norm...........: DEFAULT
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: FPS coerced ......: from 3 to 15
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: www-folder-path......: disabled
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: HTTP TCP port........: 8080
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: HTTP Listen Address..: (null)
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: username:password....: disabled
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: commands.............: enabled
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: starting input plugin input_uvc.so
Thu Oct 17 01:21:18 2019 user.info MJPG-streamer [2192]: starting output plugin: output_http.so (ID: 00)

Thu Oct 17 12:07:15 2019 kern.warn kernel: [86548.139875] netifd invoked oom-killer: gfp_mask=0x14201ca(GFP_HIGHUSER_MOVABLE|__GFP_COLD), nodemask=(null),  order=0, oom_score_adj=0

Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.151073] CPU: 0 PID: 949 Comm: netifd Not tainted 4.14.149 #0
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.156776] Stack : 804d0000 8049d780 00000000 00000000 80479560 8088fb7c 818a0ebc 804ad3c7
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.165194]         804755e0 000003b5 80523800 00000100 00000040 00000001 8088fb30 29eaeb4a
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.173512]         00000000 00000000 80520000 80528008 00000000 00000000 00000008 00000000
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.181834]         000000a9 21c00000 00000000 00024e21 80000000 819b2f00 8088fd10 8047deec
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.190086]         0000008a 00000100 00000040 014201ca 00000003 00000010 00000000 80520000
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.198461]         ...
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.200767] Call Trace:
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.203062] [<8006c560>] show_stack+0x58/0x100
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.207666] [<8010c478>] dump_header+0x98/0x200
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.212188] [<8010b5cc>] oom_kill_process+0xd0/0x494
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.217143] [<8010c314>] out_of_memory+0x2c0/0x310
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.221912] [<8011080c>] __alloc_pages_nodemask+0x834/0xc9c
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.227402] [<80108ccc>] filemap_fault+0x27c/0x710
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.232267] [<80133bc0>] __do_fault+0x11c/0x190
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.236750] [<80138308>] handle_mm_fault+0x8b0/0xdb4
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.241695] [<80073bb4>] __do_page_fault+0x164/0x480
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.246596] [<8007953c>] tlb_do_page_fault_0+0x10c/0x114
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.251914] Mem-Info:
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.254123] active_anon:336 inactive_anon:6 isolated_anon:0
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.254123]  active_file:69 inactive_file:97 isolated_file:12
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.254123]  unevictable:0 dirty:0 writeback:0 unstable:0
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.254123]  slab_reclaimable:154 slab_unreclaimable:1004
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.254123]  mapped:903 shmem:16 pagetables:60 bounce:0
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.254123]  free:431 free_pcp:0 free_cma:0

Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286042] netifd: page allocation failure: order:0, mode:0x1081020(GFP_ATOMIC|__GFP_NORETRY), nodemask=(null)

Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286083] CPU: 0 PID: 949 Comm: netifd Not tainted 4.14.149 #0
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286088] Stack : 804d0000 8049d780 00000000 00000000 80479560 81807c2c 818a0ebc 804ad3c7
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286122]         804755e0 000003b5 80523800 00000001 00000030 00000001 81807be0 29eaeb4a
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286153]         00000000 00000000 80520000 b6072a64 00000004 803dc74c 00004eb5 00000000
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286184]         002c3a3c e0e00000 80550000 804c0000 80000000 00000000 8047e1d0 00000000
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286216]         00000000 00000001 00000030 01081020 00000000 00000000 00000000 80520000
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286246]         ...
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286254] Call Trace:
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286274] [<8006c560>] show_stack+0x58/0x100
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286290] [<8010ff40>] warn_alloc+0xc8/0x160
Thu Oct 17 12:07:19 2019 kern.warn kernel: [86548.286304] [<80110a60>] __alloc_pages_nodemask+0xa88/0xc9c
[... cut ...]
Thu Oct 17 12:07:19 2019 kern.err kernel: [86550.391085] Out of memory: Kill process 2192 (mjpg_streamer) score 138 or sacrifice child

Thu Oct 17 12:07:19 2019 kern.err kernel: [86550.399649] ath: skbuff alloc of size 1974 failed

Thu Oct 17 12:07:19 2019 kern.err kernel: [86550.404071] Killed process 2192 (mjpg_streamer) total-vm:7276kB, anon-rss:256kB, file-rss:3604kB, shmem-rss:0kB
Thu Oct 17 12:07:19 2019 kern.info kernel: [86559.425086] oom_reaper: reaped process 2192 (mjpg_streamer), now anon-rss:0kB, file-rss:3600kB, shmem-rss:0kB

```

mgondium, 2019-10-17:
The previous build on which I dind't have the OOM problem was "OpenWrt SNAPSHOT, r7066-060e1ecefa". The segmentation fault on the mjpg_streamer executable is there, and it's version 2.0 too, but not md5 equivalent, so i'm starting to question if this is a library or driver leak.

neheb, 2019-10-17:
@ja-pa updated to the latest version at the time. he might have some insight.

mgondium, 2019-10-18:
There it goes again. I'm not sure about the OOM criteria but could mjpg_streamer be the one killed for being the largest but not the one with the leak? If I relaunch it without rebooting, the system will OOM much faster, so something is left behind hogging memory.

```
Fri Oct 18 00:30:30 2019 user.info MJPG-streamer [1095]: starting application
Fri Oct 18 00:30:30 2019 user.info MJPG-streamer [1095]: MJPG Streamer Version.: 2.0
Fri Oct 18 00:30:30 2019 user.info MJPG-streamer [1095]: Using V4L2 device.: /dev/video0
Fri Oct 18 00:30:30 2019 user.info MJPG-streamer [1095]: Desired Resolution: 640 x 480
Fri Oct 18 00:30:30 2019 user.info MJPG-streamer [1095]: Frames Per Second.: 3
Fri Oct 18 00:30:30 2019 user.info MJPG-streamer [1095]: Format............: JPEG
Fri Oct 18 00:30:30 2019 user.info MJPG-streamer [1095]: TV-Norm...........: DEFAULT
Fri Oct 18 00:30:30 2019 user.info MJPG-streamer [1095]: FPS coerced ......: from 3 to 15
Fri Oct 18 00:30:31 2019 user.info MJPG-streamer [1095]: www-folder-path......: disabled
Fri Oct 18 00:30:31 2019 user.info MJPG-streamer [1095]: HTTP TCP port........: 8080
Fri Oct 18 00:30:31 2019 user.info MJPG-streamer [1095]: HTTP Listen Address..: (null)
Fri Oct 18 00:30:31 2019 user.info MJPG-streamer [1095]: username:password....: disabled
Fri Oct 18 00:30:31 2019 user.info MJPG-streamer [1095]: commands.............: enabled
Fri Oct 18 00:30:31 2019 user.info MJPG-streamer [1095]: starting input plugin input_uvc.so
Fri Oct 18 00:30:31 2019 user.info MJPG-streamer [1095]: starting output plugin: output_http.so (ID: 00)

Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.092332] procd invoked oom-killer: gfp_mask=0x14201ca(GFP_HIGHUSER_MOVABLE|__GFP_COLD), nodemask=(null),  order=0, oom_score_adj=0
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.103327] CPU: 0 PID: 1 Comm: procd Not tainted 4.14.149 #0
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.108787] Stack : 804d0000 8049d780 00000000 00000000 80479560 8181fb7c 818202fc 804ad3c7
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.117142]         804755e0 00000001 80523800 00000100 00000040 00000001 8181fb30 d0a26e9f
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.125550]         00000000 00000000 80520000 80528008 00000000 00000000 00000008 00000000
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.133870]         000000ab 8ea00000 00000000 0001939f 80000000 818a0bc0 8181fd10 8047deec
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.142280]         00000092 00000100 00000040 014201ca 00000000 00000010 00000000 80520000
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.150755]         ...
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.152717] Call Trace:
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.155299] [<8006c560>] show_stack+0x58/0x100
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.159809] [<8010c478>] dump_header+0x98/0x200
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.164287] [<8010b5cc>] oom_kill_process+0xd0/0x494
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.169271] [<8010c314>] out_of_memory+0x2c0/0x310
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.174037] [<8011080c>] __alloc_pages_nodemask+0x834/0xc9c
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.179711] [<80108ccc>] filemap_fault+0x27c/0x710
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.184350] [<80133bc0>] __do_fault+0x11c/0x190
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.188893] [<80138308>] handle_mm_fault+0x8b0/0xdb4
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.193773] [<80073bb4>] __do_page_fault+0x164/0x480
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.198858] [<8007953c>] tlb_do_page_fault_0+0x10c/0x114
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.203962] Mem-Info:
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.206132] active_anon:435 inactive_anon:6 isolated_anon:0
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.206132]  active_file:147 inactive_file:224 isolated_file:0
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.206132]  unevictable:0 dirty:0 writeback:0 unstable:0
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.206132]  slab_reclaimable:157 slab_unreclaimable:990
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.206132]  mapped:904 shmem:16 pagetables:62 bounce:0
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.206132]  free:248 free_pcp:0 free_cma:0
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.238123] Node 0 active_anon:1740kB inactive_anon:24kB active_file:588kB inactive_file:896kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:3616kB dirty:0kB writeback:0kB shmem:64kB writeback_tmp:0kB unstable:0kB all_unreclaimable? yes
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.260332] Normal free:988kB min:1024kB low:1280kB high:1536kB active_anon:1740kB inactive_anon:24kB active_file:588kB inactive_file:896kB unevictable:0kB writepending:0kB present:32768kB managed:27228kB mlocked:0kB kernel_stack:320kB pagetables:248kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.287146] lowmem_reserve[]: 0 0
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.289871] Normal: 1*4kB (U) 1*8kB (U) 39*16kB (U) 11*32kB (UE) 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 988kB
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.301963] 387 total pagecache pages
Fri Oct 18 14:28:27 2019 kern.warn kernel: [50270.305284] 0 pages in swap cache
[... cut ...]
Fri Oct 18 14:28:27 2019 kern.info kernel: [50270.442473] [ 2202]     0  2202      301       10       3       0        0             0 sh
Fri Oct 18 14:28:27 2019 kern.info kernel: [50270.450779] [ 2203]     0  2203      312       21       4       0        0             0 sh
Fri Oct 18 14:28:27 2019 kern.info kernel: [50270.459237] [ 2205]     0  2205      319       27       4       0        0             0 sh
Fri Oct 18 14:28:27 2019 kern.err kernel: [50270.467462] Out of memory: Kill process 1095 (mjpg_streamer) score 146 or sacrifice child
```

mgondium, 2019-10-19:
I've enabled kmemleak but it didn't find anything. Free memory is slowly decreasing and slab usage is growing, mjpg_streamer still running and will probably OOM in a few hours.

```
MemTotal:          27212 kB
MemFree:            1424 kB
MemAvailable:       2004 kB
Buffers:            1232 kB
Cached:             1924 kB
SwapCached:            0 kB
Active:             3476 kB
Inactive:           1144 kB
Active(anon):       1480 kB
Inactive(anon):      100 kB
Active(file):       1996 kB
Inactive(file):     1044 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:          1472 kB
Mapped:             5112 kB
Shmem:               116 kB
Slab:               7888 kB
SReclaimable:        712 kB
SUnreclaim:         7176 kB
KernelStack:         312 kB
PageTables:          232 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:       13604 kB
Committed_AS:       5988 kB
VmallocTotal:    1048372 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB

PID USER       VSZ STAT COMMAND
    1 root      1572 S    /sbin/procd
    2 root         0 SW   [kthreadd]
    3 root         0 IW   [kworker/0:0]
    4 root         0 IW<  [kworker/0:0H]
    5 root         0 IW   [kworker/u2:0]
    6 root         0 IW<  [mm_percpu_wq]
    7 root         0 SW   [ksoftirqd/0]
    8 root         0 IW<  [netns]
   65 root         0 SW   [oom_reaper]
   66 root         0 IW   [kworker/u2:2]
   80 root         0 IW<  [writeback]
   82 root         0 SW   [kcompactd0]
   83 root         0 IW<  [crypto]
   85 root         0 IW<  [kblockd]
  118 root         0 SW   [kswapd0]
  199 root         0 SW   [spi0]
  313 root         0 IW<  [dsa_ordered]
  323 root         0 SWN  [kmemleak]
  325 root         0 IW<  [kworker/0:1H]
  348 root         0 IW   [kworker/0:2]
  466 root      1224 S    /sbin/ubusd
  471 root       920 S    /sbin/askfirst /usr/libexec/login.sh
  513 root      1024 S    /sbin/urngd
  606 root         0 IW<  [cfg80211]
  868 root      1288 S    /sbin/logd -S 64
  894 root      1944 S    /sbin/rpcd -s /var/run/ubus.sock -t 30
  996 root      1076 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 1070 root      1768 S    /sbin/netifd
 1330 root      1204 S    udhcpc -p /var/run/udhcpc-br-lan.pid -s /lib/netifd/dhcp.script -f -t 0 -i br-lan -x hostname:OpenWrt -C -O 121
 1443 root         0 SWN  [jffs2_gcd_mtd3]
 1480 root      1208 S<   /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.org -p 2.openwrt.pool.ntp.org -p 3.openwrt.poo
 1672 root      7292 S    /usr/bin/mjpg_streamer --input input_uvc.so --device /dev/video0 --fps 3 --resolution 640x480 --output output_http.so --port 8080
 2461 root      1144 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 2462 root      1212 S    -ash
 2518 root         0 IW   [kworker/0:1]
 2519 root         0 IW   [kworker/0:3]
 2629 root      1736 S    /usr/sbin/hostapd -s -P /var/run/wifi-phy0.pid -B /var/run/hostapd-phy0.conf
 2661 root      1204 R    ps

Name:	mjpg_streamer
Umask:	0022
State:	S (sleeping)
Tgid:	1672
Ngid:	0
Pid:	1672
PPid:	1
TracerPid:	0
Uid:	0	0	0	0
Gid:	0	0	0	0
FDSize:	32
Groups:	 
NStgid:	1672
NSpid:	1672
NSpgid:	1
NSsid:	1
VmPeak:	    7388 kB
VmSize:	    7292 kB
VmLck:	       0 kB
VmPin:	       0 kB
VmHWM:	    4416 kB
VmRSS:	    4016 kB
RssAnon:	     288 kB
RssFile:	    3728 kB
RssShmem:	       0 kB
VmData:	    2364 kB
VmStk:	     132 kB
VmExe:	      12 kB
VmLib:	    1156 kB
VmPTE:	      20 kB
VmPMD:	       0 kB
VmSwap:	       0 kB
Threads:	4
SigQ:	0/199
SigPnd:	00000000000000000000000000000000
ShdPnd:	00000000000000000000000000000000
SigBlk:	00000000000000000000000000000000
SigIgn:	00000000000000000000000000001000
SigCgt:	00000000000000000000000000000002
CapInh:	0000000000000000
CapPrm:	0000003fffffffff
CapEff:	0000003fffffffff
CapBnd:	0000003fffffffff
CapAmb:	0000000000000000
NoNewPrivs:	0
Seccomp:	0
Speculation_Store_Bypass:	unknown
Cpus_allowed:	1
Cpus_allowed_list:	0
voluntary_ctxt_switches:	75
nonvoluntary_ctxt_switches:	21
```

neheb, 2019-10-19:
kmemleak I'm guessing is for the kernel. So problem is most likely in userspace. Maybe valgrind would help...

I'm also very curious who and what broke this. Could be me (protobuf(-c) changes), @ja-pa (version update) or @dangowrt (modularize commit).

Might want to try bisecting this.

mgondium, 2019-10-19:
kmemleak got something this time, not sure what. Sorry for all the spam.

```
unreferenced object 0x80c93c80 (size 128):
  comm "kmodloader", pid 326, jiffies 4294937906 (age 20379.570s)
  hex dump (first 32 bytes):
    00 c9 3d 80 c0 09 a0 00 00 00 20 00 00 00 00 02  ..=....... .....
    80 c9 3d 80 00 00 00 01 00 00 00 00 80 07 0f 14  ..=.............
  backtrace:
    [<801526bc>] __kmalloc_track_caller+0x204/0x25c
    [<801275c4>] memdup_user+0x30/0xc8
    [<8012769c>] strndup_user+0x40/0x68
    [<800dcf10>] load_module+0x18c8/0x2390
    [<800ddb18>] SyS_init_module+0x140/0x18c
    [<80071d6c>] syscall_common+0x34/0x58
unreferenced object 0x80c93d80 (size 128):
  comm "kmodloader", pid 326, jiffies 4294937906 (age 20379.570s)
  hex dump (first 32 bytes):
    6e 6c 73 5f 62 61 73 65 00 00 00 00 00 00 00 00  nls_base........
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  backtrace:
    [<801526bc>] __kmalloc_track_caller+0x204/0x25c
    [<80127118>] kstrdup+0x40/0x84
    [<803ce0e0>] kobject_set_name_vargs+0x38/0xc4
    [<803ce338>] kobject_init_and_add+0x3c/0x84
    [<800dd12c>] load_module+0x1ae4/0x2390
    [<800ddb18>] SyS_init_module+0x140/0x18c
    [<80071d6c>] syscall_common+0x34/0x58
unreferenced object 0x80c9d700 (size 128):
  comm "kmodloader", pid 326, jiffies 4294937975 (age 20378.880s)
  hex dump (first 32 bytes):
    00 c9 d8 00 c0 0c 90 00 00 00 50 00 00 00 00 02  ..........P.....
    80 c9 d8 00 00 00 00 04 00 00 00 00 80 07 0f 14  ................
  backtrace:
    [<801526bc>] __kmalloc_track_caller+0x204/0x25c
    [<801275c4>] memdup_user+0x30/0xc8
    [<8012769c>] strndup_user+0x40/0x68
    [<800dcf10>] load_module+0x18c8/0x2390
    [<800ddb18>] SyS_init_module+0x140/0x18c
    [<80071d6c>] syscall_common+0x34/0x58
unreferenced object 0x80c92380 (size 128):
  comm "kmodloader", pid 326, jiffies 4294938029 (age 20378.340s)
  hex dump (first 32 bytes):
    00 c9 24 80 c0 0d 20 00 00 00 20 00 00 00 00 02  ..$... ... .....
    80 c9 24 80 00 00 00 01 00 00 00 00 80 07 0f 14  ..$.............
  backtrace:
    [<801526bc>] __kmalloc_track_caller+0x204/0x25c
    [<801275c4>] memdup_user+0x30/0xc8
    [<8012769c>] strndup_user+0x40/0x68
    [<800dcf10>] load_module+0x18c8/0x2390
    [<800ddb18>] SyS_init_module+0x140/0x18c
    [<80071d6c>] syscall_common+0x34/0x58
unreferenced object 0x80c92480 (size 128):
  comm "kmodloader", pid 326, jiffies 4294938029 (age 20378.340s)
  hex dump (first 32 bytes):
    67 70 69 6f 5f 62 75 74 74 6f 6e 5f 68 6f 74 70  gpio_button_hotp
    6c 75 67 00 00 00 00 00 00 00 00 00 00 00 00 00  lug.............
  backtrace:
    [<801526bc>] __kmalloc_track_caller+0x204/0x25c
    [<80127118>] kstrdup+0x40/0x84
    [<803ce0e0>] kobject_set_name_vargs+0x38/0xc4
    [<803ce338>] kobject_init_and_add+0x3c/0x84
    [<800dd12c>] load_module+0x1ae4/0x2390
    [<800ddb18>] SyS_init_module+0x140/0x18c
    [<80071d6c>] syscall_common+0x34/0x58
unreferenced object 0x80cc1300 (size 128):
  comm "kmodloader", pid 326, jiffies 4294938088 (age 20377.750s)
  hex dump (first 32 bytes):
    00 cc 15 80 c0 0e 00 00 00 00 20 00 00 00 00 02  .......... .....
    80 cc 15 80 00 00 00 01 00 00 00 00 80 07 0f 14  ................
[... 1145 lines cut ...]
    [<803ce0e0>] kobject_set_name_vargs+0x38/0xc4
    [<803ce338>] kobject_init_and_add+0x3c/0x84
    [<800dd12c>] load_module+0x1ae4/0x2390
    [<800ddb18>] SyS_init_module+0x140/0x18c
    [<80071d6c>] syscall_common+0x34/0x58
unreferenced object 0x80ab2880 (size 128):
  comm "kmodloader", pid 561, jiffies 4294941833 (age 20340.420s)
  hex dump (first 32 bytes):
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02  ................
    00 00 00 00 55 53 00 00 00 00 00 01 00 01 00 00  ....US..........
  backtrace:
    [<801507fc>] kmem_cache_alloc+0x158/0x1a8
    [<80847844>] regulatory_hint+0x94/0x5c0 [cfg80211]
    [<80a62a20>] ath9k_init_device+0x9a0/0xa38 [ath9k]
    [<80a6eab0>] ath_pci_exit+0x264/0x314 [ath9k]
unreferenced object 0x8083b000 (size 1024):
  comm "kworker/0:1", pid 110, jiffies 4294941836 (age 20340.390s)
  hex dump (first 32 bytes):
    00 00 00 00 00 00 00 00 00 00 00 06 55 53 00 00  ............US..
    00 00 00 01 00 24 a6 d0 00 25 b8 40 00 00 9c 40  .....$...%.@...@
  backtrace:
    [<80151074>] __kmalloc+0x204/0x25c
    [<80847934>] regulatory_hint+0x184/0x5c0 [cfg80211]
    [<800a8da4>] check_preempt_curr+0x2c/0xac
```

mgondium, 2019-10-23:
Went back to "OpenWrt 18.06.4, r7808-ef686b7292" and in 2 days uptime hasn't OOM'd. Trunk remains problematic.

zokl, 2020-01-07:
Program modules folder is resolved by this patch: https://github.com/openwrt/packages/pull/10954

mgondium, 2020-01-19:
I can confirm that the forders are fixed.
Regarding the memory leak, mjpg_streamer triggers the problem but I don't understand how a terminated program can continue hogging memory.
I'm biased to believe that this is a driver/kernel problem, but I can't get the memory back by unloading modules.
Unvariably, within every 48 hours the device restarts because it runs out of memory.

neheb, 2020-01-19:
Does it fork to a separate process?

mgondium, 2020-01-19:
I'm not sure.

```
  PID USER       VSZ STAT COMMAND
    1 root      1588 S    /sbin/procd
    2 root         0 SW   [kthreadd]
    4 root         0 IW<  [kworker/0:0H-kb]
    6 root         0 IW<  [mm_percpu_wq]
    7 root         0 SW   [ksoftirqd/0]
    8 root         0 IW<  [netns]
    9 root         0 SW   [oom_reaper]
   10 root         0 IW   [kworker/u2:1-ev]
   96 root         0 IW<  [writeback]
   98 root         0 SW   [kcompactd0]
   99 root         0 IW<  [crypto]
  101 root         0 IW<  [kblockd]
  102 root         0 IW   [kworker/0:1-eve]
  103 root         0 SW   [watchdogd]
  136 root         0 SW   [kswapd0]
  145 root         0 IW   [kworker/u2:2-ph]
  212 root         0 SW   [spi0]
  366 root         0 IW   [kworker/0:2-eve]
  372 root         0 IW<  [kworker/0:1H-kb]
  384 root         0 SW   [irq/12-keys]
  482 root         0 SWN  [jffs2_gcd_mtd4]
  549 root      1220 S    /sbin/ubusd
  563 root       920 S    /sbin/askfirst /usr/libexec/login.sh
  580 root      1024 S    /sbin/urngd
  639 root         0 IW<  [cfg80211]
  756 root      4524 S    /usr/sbin/hostapd -s -n phy0 -g /var/run/hostapd-phy0/global
  812 root      1244 S    /sbin/logd -S 64
  852 root      1104 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
  925 root      1704 S    /sbin/netifd
  960 root      1208 S    /usr/sbin/crond -f -c /etc/crontabs -l 9
  988 root      7260 S    /usr/bin/mjpg_streamer --input input_uvc.so --device /dev/video0 --fps 5 --resolution 640x480 --output output_http.so --port 8080
 1112 root      1204 S    udhcpc -p /var/run/udhcpc-br-lan.pid -s /lib/netifd/dhcp.script -f -t 0 -i br-lan -x hostname:ap_g -C -R -O 121
 1369 root      1204 S<   /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.org -p 2.openwrt.pool.ntp.org -p 3.op
 3685 root      1172 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 3686 root      1212 S    -ash
 3705 root         0 IW   [kworker/u2:0-ev]
 3717 root      1204 R    ps

Name:	mjpg_streamer
Umask:	0022
State:	S (sleeping)
Tgid:	988
Ngid:	0
Pid:	988
PPid:	1
TracerPid:	0
Uid:	0	0	0	0
Gid:	0	0	0	0
FDSize:	32
Groups:	 
NStgid:	988
NSpid:	988
NSpgid:	1
NSsid:	1
VmPeak:	    7540 kB
VmSize:	    7260 kB
VmLck:	       0 kB
VmPin:	       0 kB
VmHWM:	    4980 kB
VmRSS:	    4848 kB
RssAnon:	     792 kB
RssFile:	    4056 kB
RssShmem:	       0 kB
VmData:	    2332 kB
VmStk:	     132 kB
VmExe:	      12 kB
VmLib:	    1156 kB
VmPTE:	      24 kB
VmSwap:	       0 kB
CoreDumping:	0
Threads:	4
SigQ:	0/198
SigPnd:	00000000000000000000000000000000
ShdPnd:	00000000000000000000000000000000
SigBlk:	00000000000000000000000000000000
SigIgn:	00000000000000000000000000001000
SigCgt:	00000000000000000000000000000002
CapInh:	0000000000000000
CapPrm:	0000003fffffffff
CapEff:	0000003fffffffff
CapBnd:	0000003fffffffff
CapAmb:	0000000000000000
NoNewPrivs:	0
Seccomp:	0
Speculation_Store_Bypass:	unknown
Cpus_allowed:	1
Cpus_allowed_list:	0
voluntary_ctxt_switches:	104
nonvoluntary_ctxt_switches:	20

```

neheb, 2020-01-19:
Try using GNU top. Last I remember, it shows transmission as 4 separate processes as opposed to busybox ps/top.

mgondium, 2020-01-19:
For the life of me, I couldn't find a package to install GNU top on running openwrt (?!), here's some htop:
```

  CPU[###*************                                                                                                                                    10.3%]   Tasks: 14, 3 thr; 1 running
  Mem[|||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||####***********21.6M/26.1M]   Load average: 0.25 0.46 0.26 
  Swp[                                                                                                                                                    0K/0K]   Uptime: 00:06:13

  PID USER      PRI  NI  VIRT   RES   SHR S CPU% MEM%   TIME+  Command                                                                                                                                                                                                                                                                
  755 root       20   0  4524   944   684 S  0.0  3.5  0:05.86 /usr/sbin/hostapd -s -n phy0 -g /var/run/hostapd-phy0/global
 1254 root       20   0  1172   452   352 S  0.0  1.7  0:02.79 /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 1595 root       20   0  7204  4132  3996 S  0.6 15.5  0:02.31 /usr/bin/mjpg_streamer --input input_uvc.so --device /dev/video0 --fps 5 --resolution 640x480 --output output_http.so --port 8080
 1137 root        5 -15  1208   540   504 S  0.0  2.0  0:02.29 /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.org -p 2.openwrt.pool.ntp.org -p 3.openwrt.pool.ntp.org
    1 root       20   0  1588   424   296 S  0.0  1.6  0:02.26 /sbin/procd
 1642 root       20   0  3288  2856   796 R  3.2 10.7  0:02.14 htop -t -C
 1614 root       20   0  7204  4132  3996 S  0.6 15.5  0:01.20 /usr/bin/mjpg_streamer --input input_uvc.so --device /dev/video0 --fps 5 --resolution 640x480 --output output_http.so --port 8080
  925 root       20   0  1704   436   260 S  0.0  1.6  0:00.91 /sbin/netifd
  812 root       20   0  1244   380   272 S  0.0  1.4  0:00.90 /sbin/logd -S 64
  580 root       20   0  1024    48     4 S  0.0  0.2  0:00.86 /sbin/urngd
 1609 root       20   0  7204  4132  3996 S  0.0 15.5  0:00.57 /usr/bin/mjpg_streamer --input input_uvc.so --device /dev/video0 --fps 5 --resolution 640x480 --output output_http.so --port 8080
  549 root       20   0  1220   356   300 S  0.0  1.3  0:00.38 /sbin/ubusd
 1384 root       20   0  1212   536   492 S  0.0  2.0  0:00.25 -ash
 1610 root       20   0  7204  4132  3996 S  0.0 15.5  0:00.22 /usr/bin/mjpg_streamer --input input_uvc.so --device /dev/video0 --fps 5 --resolution 640x480 --output output_http.so --port 8080
  852 root       20   0  1104    44     4 S  0.0  0.2  0:00.11 /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
  563 root       20   0   920    36     4 S  0.0  0.1  0:00.09 /sbin/askfirst /usr/libexec/login.sh
 1113 root       20   0  1204    40     4 S  0.0  0.1  0:00.01 udhcpc -p /var/run/udhcpc-br-lan.pid -s /lib/netifd/dhcp.script -f -t 0 -i br-lan -x hostname:ap_g -C -R -O 

```

neheb, 2020-01-20:
Yeah I meant htop. There are three processes with 3 different PIDs.

mgondium, 2020-01-20:
Four PIDs in the tree view. Don't take the values as a status progress, it rebooted in the meantime.

```
  CPU[###*********                                                                                        9.7%]   Tasks: 15, 3 thr; 1 running
  Mem[||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||###********20.5M/26.1M]   Load average: 0.01 0.00 0.00 
  Swp[                                                                                                   0K/0K]   Uptime: 00:25:17

  PID USER      PRI  NI  VIRT   RES   SHR S CPU% MEM%   TIME+  Command                                                                                                                                                               
    1 root       20   0  1588   668   544 S  0.0  2.5  0:01.49 /sbin/procd                                                                                                                                                           
 1163 root        5 -15  1208   728   688 S  0.0  2.7  0:00.42 `- /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.org -p 2.openwrt.pool.ntp.org -p 3.openwrt.pool.ntp.org
 1001 root       20   0  7224  4296  4064 S  1.9 16.1  0:32.04 `- /usr/bin/mjpg_streamer --input input_uvc.so --device /dev/video0 --fps 5 --resolution 640x480 --output output_http.so --port 8080
 1373 root       20   0  7224  4296  4064 S  1.3 16.1  0:20.44 |  `- /usr/bin/mjpg_streamer --input input_uvc.so --device /dev/video0 --fps 5 --resolution 640x480 --output output_http.so --port 8080
 1051 root       20   0  7224  4296  4064 S  0.0 16.1  0:00.00 |  `- /usr/bin/mjpg_streamer --input input_uvc.so --device /dev/video0 --fps 5 --resolution 640x480 --output output_http.so --port 8080
 1050 root       20   0  7224  4296  4064 S  1.3 16.1  0:11.14 |  `- /usr/bin/mjpg_streamer --input input_uvc.so --device /dev/video0 --fps 5 --resolution 640x480 --output output_http.so --port 8080
  973 root       20   0  1208   616   576 S  0.0  2.3  0:00.05 `- /usr/sbin/crond -f -c /etc/crontabs -l 9
  938 root       20   0  1704   764   588 S  0.0  2.9  0:00.52 `- /sbin/netifd
 1126 root       20   0  1204   700   664 S  0.0  2.6  0:00.02 |  `- udhcpc -p /var/run/udhcpc-br-lan.pid -s /lib/netifd/dhcp.script -f -t 0 -i br-lan -x hostname:ap_g -C -R -O 121
  865 root       20   0  1104   616   576 S  0.0  2.3  0:00.12 `- /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 1671 root       20   0  1172   780   680 S  0.6  2.9  0:02.36 |  `- /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 1672 root       20   0  1212   844   800 S  0.0  3.2  0:00.03 |     `- -ash
 1680 root       20   0  3320  2816   852 R  2.6 10.6  0:03.37 |        `- htop
  825 root       20   0  1244   560   476 S  0.0  2.1  0:00.12 `- /sbin/logd -S 64
  765 root       20   0  4524  1340  1080 S  0.0  5.0  0:02.92 `- /usr/sbin/hostapd -s -n phy0 -g /var/run/hostapd-phy0/global
  593 root       20   0  1024   500   456 S  0.0  1.9  0:00.87 `- /sbin/urngd
  562 root       20   0   920   396   368 S  0.0  1.5  0:00.09 `- /sbin/askfirst /usr/libexec/login.sh
  549 root       20   0  1220   540   484 S  0.0  2.0  0:00.25 `- /sbin/ubusd

```

mgondium, 2020-02-11:
The webcam I used (Logitech C170) was damaged and I had to replace it with a different one (HP 1300). So far, the problem has not manifested itself, so maybe it has something to do with bad data from the camera.

neheb, 2020-02-23:
I take it this can be closed?

mgondium, 2020-02-23:
I've been trying to get hold of another same model webcam since the old one was thrashed, so that I can clarify if it was THAT camera in partucular or this model altogether.

Meanwhile and as it stands, I believe this can be closed.

closed 2020-02-23 by mgondium