openwrt/packages #10255: gcc: fails to build with static library

flyn-org, 2019-10-16:
Maintainer: @noblepepper
Environment: x86_64 master

Description:
The new gcc build with static-library support failed to build for me:

```
x86_64-openwrt-linux-musl-gcc   -g -O2 -g -Os -O2  -g -O2  -g -Os -DIN_GCC    -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include   -fpic -mlong-double-80 -DUSE_ELF_SYMVER -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector   -fpic -mlong-double-80 -DUSE_ELF_SYMVER -I. -I. -I../../host-x86_64-openwrt-linux-musl/gcc -I../.././libgcc -I../.././libgcc/. -I../.././libgcc/../gcc -I../.././libgcc/../include  -DHAVE_CC_TLS  -DUSE_TLS -o unwind-dw2.o -MT unwind-dw2.o -MD -MP -MF unwind-dw2.dep -fexceptions -c ../.././libgcc/unwind-dw2.c -fvisibility=hidden -DHIDE_EXPORTS
cc1: note: someone does not honour COPTS correctly, passed 0 times
../.././libgcc/unwind-dw2.c:41:10: fatal error: sys/sdt.h: No such file or directory
 #include <sys/sdt.h>
          ^~~~~~~~~~~
compilation terminated.
make[7]: *** [../.././libgcc/shared-object.mk:14: unwind-dw2.o] Error 1
make[7]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0/x86_64-openwrt-linux-musl/libgcc'
make[6]: *** [Makefile:12807: all-target-libgcc] Error 2
make[6]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0'
make[5]: *** [Makefile:901: all] Error 2
make[5]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0'
make[4]: *** [Makefile:225: /home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0/.built] Error 2
make[4]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user/feeds/packages/devel/gcc'
time: package/feeds/packages/gcc/compile#1028.52#123.62#1230.58
make[3]: *** [package/Makefile:113: package/feeds/packages/gcc/compile] Error 2
make[3]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user'
make[2]: *** [package/Makefile:107: /home/mike/Source/aq/aquinas/openwrt-aquinas-user/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[2]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user'
make[1]: *** [/home/mike/Source/aq/aquinas/openwrt-aquinas-user/include/toplevel.mk:227: world] Error 2
make[1]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user'
make: *** [Makefile:396: aquinas-user-openwrt-x86-64-combined-ext4.img] Error 2
```

Here is my diffconfig:

```
# 1. Run make menuconfig.
# 2. Run ./scripts/diffconfig.sh > THIS-FILE

CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_Generic=y
CONFIG_INCLUDE_STATIC_LIBC=y
CONFIG_INCLUDE_STATIC_LIBPTHREAD=y
CONFIG_PACKAGE_ar=y
CONFIG_PACKAGE_binutils=y
CONFIG_PACKAGE_gcc=y
CONFIG_PACKAGE_libbfd=y
CONFIG_PACKAGE_libopcodes=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_objdump=y
CONFIG_PACKAGE_zlib=y
CONFIG_TARGET_ROOTFS_PARTSIZE=1024
```

noblepepper, 2019-10-19:
I can't replicate here, I believe you have hit an old [bug ](https://github.com/openwrt/packages/issues/296)that only affects certain distros. Would you try compiling with [this](https://raw.githubusercontent.com/bk138/packages/480452dce864d7d0f5c35a951e8229b4e2ef03df/devel/gcc/patches/020-disable-check-for-sys-sdt-h.patch) file in package/feeds/packages/gcc/patches? 

If that fixes the problem I'll add it to github if it doesn't cause problems elsewhere.

BTW-It probably doesn't matter if you add the static library stuff, this error should hit you without.

flyn-org, 2019-10-19:
Yes, I build on Fedora, and it does indeed appear issue https://github.com/openwrt/packages/issues/296 is at play here. The patch you propose above fixes this.

However, the build now fails with the following. My host GCC is Fedora 31 beta's gcc-9.2.1-1.fc30.x86_64.

```
libtool: compile:  x86_64-openwrt-linux-musl-c++ -B/home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0/x86_64-openwrt-linux-musl/libstdc++-v3/src/.libs -B/home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0/x86_64-openwrt-linux-musl/libstdc++-v3/libsupc++/.libs -DHAVE_CONFIG_H -I. -I../.././libitm -I../.././libitm/config/linux/x86 -I../.././libitm/config/linux -I../.././libitm/config/x86 -I../.././libitm/config/posix -I../.././libitm/config/generic -I../.././libitm -mrtm -Wall -pthread -Werror -std=gnu++0x -funwind-tables -fno-exceptions -fno-rtti -fabi-version=4 -g -O2 -D_GLIBCXX_INCLUDE_NEXT_C_HEADERS -D_GNU_SOURCE -g -Os -MT beginend.lo -MD -MP -MF .deps/beginend.Tpo -c ../.././libitm/beginend.cc  -fPIC -DPIC -o .libs/beginend.o
../.././libitm/beginend.cc: In member function 'void GTM::gtm_transaction_cp::save(GTM::gtm_thread*)':
../.././libitm/beginend.cc:434:67: error: 'void* memcpy(void*, const void*, size_t)' writing to an object of non-trivially copyable type 'class GTM::aa_tree<long unsigned int, GTM::gtm_alloc_action>'; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
   memcpy(&alloc_actions, &tx->alloc_actions, sizeof(alloc_actions));
                                                                   ^
In file included from ../.././libitm/libitm_i.h:76,
                 from ../.././libitm/beginend.cc:25:
../.././libitm/aatree.h:127:7: note: 'class GTM::aa_tree<long unsigned int, GTM::gtm_alloc_action>' declared here
 class aa_tree : public aa_tree_key<KEY>
       ^~~~~~~
../.././libitm/beginend.cc: In member function 'void GTM::gtm_transaction_cp::commit(GTM::gtm_thread*)':
../.././libitm/beginend.cc:452:67: error: 'void* memcpy(void*, const void*, size_t)' writing to an object of non-trivially copyable type 'class GTM::aa_tree<long unsigned int, GTM::gtm_alloc_action>'; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
   memcpy(&tx->alloc_actions, &alloc_actions, sizeof(alloc_actions));
                                                                   ^
In file included from ../.././libitm/libitm_i.h:76,
                 from ../.././libitm/beginend.cc:25:
../.././libitm/aatree.h:127:7: note: 'class GTM::aa_tree<long unsigned int, GTM::gtm_alloc_action>' declared here
 class aa_tree : public aa_tree_key<KEY>
       ^~~~~~~
../.././libitm/beginend.cc: In member function 'void GTM::gtm_thread::rollback(GTM::gtm_transaction_cp*, bool)':
../.././libitm/beginend.cc:488:71: error: 'void* memcpy(void*, const void*, size_t)' writing to an object of non-trivially copyable type 'class GTM::aa_tree<long unsigned int, GTM::gtm_alloc_action>'; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
       memcpy(&alloc_actions, &cp->alloc_actions, sizeof(alloc_actions));
                                                                       ^
In file included from ../.././libitm/libitm_i.h:76,
                 from ../.././libitm/beginend.cc:25:
../.././libitm/aatree.h:127:7: note: 'class GTM::aa_tree<long unsigned int, GTM::gtm_alloc_action>' declared here
 class aa_tree : public aa_tree_key<KEY>
       ^~~~~~~
cc1plus: all warnings being treated as errors
make[9]: *** [Makefile:546: beginend.lo] Error 1
make[9]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0/x86_64-openwrt-linux-musl/libitm'
make[8]: *** [Makefile:725: all-recursive] Error 1
make[8]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0/x86_64-openwrt-linux-musl/libitm'
make[7]: *** [Makefile:390: all] Error 2
make[7]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0/x86_64-openwrt-linux-musl/libitm'
make[6]: *** [Makefile:14296: all-target-libitm] Error 2
make[6]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0'
make[5]: *** [Makefile:901: all] Error 2
make[5]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0'
make[4]: *** [Makefile:225: /home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/target-x86_64_musl/gcc-7.4.0/.built] Error 2
make[4]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user/feeds/packages/devel/gcc'
time: package/feeds/packages/gcc/compile#1079.30#146.14#1261.05
make[3]: *** [package/Makefile:113: package/feeds/packages/gcc/compile] Error 2
make[3]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user'
make[2]: *** [package/Makefile:107: /home/mike/Source/aq/aquinas/openwrt-aquinas-user/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[2]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user'
make[1]: *** [/home/mike/Source/aq/aquinas/openwrt-aquinas-user/include/toplevel.mk:227: world] Error 2
make[1]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user'
make: *** [Makefile:396: aquinas-user-openwrt-x86-64-combined-ext4.img] Error 2
```

neheb, 2019-10-19:
Identical to this: https://github.com/openwrt/packages/issues/10198

flyn-org, 2019-10-19:
Nice catch. Recommend we close this issue report (sdt.h fixed) and address the latter issue in #10198.

closed 2019-11-09 by flyn-org