openwrt/packages #10270: xl2tpd : compile error with glibc

DeepDarkShadow, 2019-10-18:
Maintainer:  @yousong
Environment:  arm64, mediatek mt7622, OpenWrt trunk
xl2tpd version : 1.3.15
glibc version : 2.27
libpcap version : 1.9.1

Description:
  After OpenWRT trunk upgrade libpcap to 1.9.1 , we find out if build xl2tpd with glibc then
it will result a compile error (undefined reference to 'pcap_strlcpy') .
```
make[4]: Entering directory '/home4/mark/mtk_quark/1014/openwrt/lede/build_dir/target-aarch64_cortex-a53_glibc/xl2tpd-1.3.13'
aarch64-openwrt-linux-gnu-gcc -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home4/mark/mtk_quark/1014/openwrt/lede/build_dir/target-aarch64_cortex-a53_glibc/xl2tpd-1.3.13:xl2tpd-1.3.13 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home4/mark/mtk_quark/1014/openwrt/lede/staging_dir/target-aarch64_cortex-a53_glibc/usr/include -I/home4/mark/mtk_quark/1014/openwrt/lede/staging_dir/target-aarch64_cortex-a53_glibc/include -I/home4/mark/mtk_quark/1014/openwrt/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.4.0_glibc/usr/include -I/home4/mark/mtk_quark/1014/openwrt/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.4.0_glibc/include  -DDEBUG_PPPD -DTRUST_PPPD_TO_DIE -Os -Wall -Wextra -DSANITY -DLINUX -I./linux/include/ -DUSE_KERNEL -DIP_ALLOCATION -c contrib/pfc.c
aarch64-openwrt-linux-gnu-gcc -L/home4/mark/mtk_quark/1014/openwrt/lede/staging_dir/target-aarch64_cortex-a53_glibc/usr/lib -L/home4/mark/mtk_quark/1014/openwrt/lede/staging_dir/target-aarch64_cortex-a53_glibc/lib -L/home4/mark/mtk_quark/1014/openwrt/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.4.0_glibc/usr/lib -L/home4/mark/mtk_quark/1014/openwrt/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.4.0_glibc/lib -znow -zrelro  -o pfc pfc.o -lpcap
/home4/mark/mtk_quark/1014/openwrt/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.4.0_glibc/lib/gcc/aarch64-openwrt-linux-gnu/7.4.0/../../../../aarch64-openwrt-linux-gnu/bin/ld: /home4/mark/mtk_quark/1014/openwrt/lede/staging_dir/target-aarch64_cortex-a53_glibc/usr/lib/libpcap.so: undefined reference to `pcap_strlcpy'
collect2: error: ld returned 1 exit status
```

yousong, 2019-10-18:
For xl2tpd, we can work around the issue by skipping the build of pfc, as has been done in the latest commit.

However, the real issue should be still there in libpcap, where glibc's lacking of strlcpy causing the link failure.  Ping @dengqf6 for suggestions ;)

yousong, 2019-10-18:
Likely caused by the libpcap patch package/libs/libpcap/patches/100-debian_shared_lib.patch

yousong, 2019-10-18:
Working on converting libpcap to using cmake

yousong, 2019-10-18:
Sent a patch [libpcap: build with cmake](http://patchwork.ozlabs.org/patch/1179360/)

DeepDarkShadow, 2019-10-21:
Hi Yousong :
  I have update xl2tp and libpcap , the compile error was gone , thanks a lot!

closed 2019-10-21 by DeepDarkShadow