openwrt/packages #10273: pdns-recursor missing dependency

neheb, 2019-10-18:
https://downloads.openwrt.org/snapshots/faillogs/mipsel_24kc_24kf/packages/pdns-recursor/compile.txt

ping @James-TR 

I'm curious about this dependency. I downloaded the source code and saw that it is not being used at all. It's only used in the configure script.

The failure is due to https://github.com/openwrt/packages/commit/260d4a346afeffec5f7e4818d1179ea8db70420e#diff-eda03f2256b0bcbeec4978a233231a89

And similarly, it's not used here either.

James-TR, 2019-10-24:
That's a good point. I'll open an issue in the upstream project to see what their thoughts are.

James-TR, 2019-10-25:
But if you look in the actual library code you'll see there has been a check for `<mutex>` since 1.61.0 to use that instead of Boost::Thread where it is available.

https://github.com/boostorg/context/blob/boost-1.71.0/build/Jamfile.v2#L783-L789

hnyman, 2019-11-10:
Any forthcoming fix for this error?  (as right now the package does not compile at all)

Even a temporary fix?
Add the (actually unnecessary?) dependency to the package?

James-TR, 2019-11-10:
I don't know why I didn't think of that as a stop gap. I'll whip up a PR tonight.

PR #10516 (2019-11-13): pdns-recursor: add temporary depends to fix build
Maintainer: me
Compile tested: N/A (Dependency is not used)
Run tested: N/A (Dependency is not used)

Description:

There is an issue with the included build scripts which improperly check for a
requirement on libboost-thread where another implementation is available.

This only affects the bundled boost.m4 from https://github.com/tsuna/boost.m4 which is distributed
along with pdns-recursor. I will be working on a separate fix for the M4 macro later.

Fixes #10273.
commit 36ed7562 (2019-11-13): pdns-recursor: add temporary depends to fix build

There is an issue with the included build scripts which improperly check for a
requirement on libboost-thread where another implementation is available.

Fixes #10273.
closed 2019-11-13 by neheb