openwrt/packages #10277: MWAN3 - rules based on source MAC? or a helpful alternative

jake5253, 2019-10-19:
Maintainer: @feckert (Florian Eckert)
Environment: 
   > Arch: ARM 
   > Model: RPi_2B
   > Version: 18.06 (or any)

Description:
I see from documentation that MWAN3 allows rules based on src_ip and src_port, along with dest_ip and dest_port, however, no rules based on MAC address.  I'd like to use 2 connections for my ever-expanding bandwidth needs while being stuck with only ADSL and USB Tethering from my phone, but I'm at a point where I'd like to force my XBOX to only use one specific connection (as long as it's available). 
I have, for now, set a static IP on my XBOX and set a rule for that IP but I travel for work and generally take my XBOX with me if I'm going to be overnight, so I have to go in and manually set my manual IP address up again every time I come back to my home network which becomes tedious after you do it once or twice a week.
I have a list of ports Microsoft recommends opening for XBOX, but that list includes port 80 so I can't include all the ports with rules or my http access from other devices will end up all on one connection.
Microsoft lists the following ports:
>Port 88 (UDP)
>Port 3074 (UDP and TCP)
>Port 53 (UDP and TCP)
>**Port 80** (TCP)
>Port 500 (UDP)
>Port 3544 (UDP)
>Port 4500 (UDP)

I cannot make a rule based on destination IP because XBOX uses far too many different IPs depending on what it's doing (signing in, updating, store, each game -- sometimes even multiple IPs per game) -- that list of destination IPs would become huge, not to mention that I'd have to log and keep track of that list of IPs.

I've concluded that, if at all possible, setting a rule based on src_mac would be a straight one-shot -- no more manual IP address configuration, no more port overlapping, and no more chasing down destination IPs.

I would certainly be open to other options or suggestions, if a mac-based rule is out of question.

Thank you

feckert, 2019-10-20:
Thank you for this feature request. I'll think about it.

feckert, 2019-11-01:
@aport thanks for the file will evaluate this in the next days

feckert, 2019-11-07:
@aport thanks fore sharing your code.
@jake5253 Thanks for your suggestion for this feature
I have staged a branch with the changes https://github.com/TDT-AG/packages/tree/pr/20191107-mwan3 please try. If it works for you I will open a pullrequest for merging this feature.

feckert, 2019-11-14:
As I can see this is not simple possible.
This must be investigated deeper.
See test results https://github.com/TDT-AG/packages/tree/pr/20191107-mwan3

feckert, 2020-07-30:
From my point of view this does not fit into the mwan3.
This propsed change works in layer2.
But mwan3 works on layer3.
I will close this now, because nothing has happened for a long time and from my point of view it does not fit into mwan3.

closed 2020-07-30 by feckert