openwrt/packages #10307: ntpd: compilation fails due to missing libevent_pthread dependency

KloudJack, 2019-10-20:
Maintainer: @tripolar 
Environment: OpenWRT snapshot r11266-34939711a5 - packages @ fa790f0

Description:
Building master of openwrt and all feeds with any ntpd package selected fails due to missing libevent_pthreads dependency during linking.

I'm building with libstdc++ support instead of uClibc/++.
This could make a difference if uClibc pthreads support evaluates differently in ntpd's automake definitions vs. those of GNU's.

closed 2019-10-22 by tripolar

karlp, 2019-11-07:
@tripolar  This is also broken in 19.07 branch

karlp, 2019-11-07:
SNTP is the only user of libevent, are you sure this is the right fix?

karlp, 2019-11-07:
and by that, I meant the sntp binary, not sntp support in the toolset itself

KloudJack, 2019-11-07:
@karlp libevent was previously defined as a dependency for ntp-keygen when I created the PR to fix this.

Checking the ntp 4.2.8p13 Makefile and running a build without sntp, I agree with you that libevent is only required for the sntp binary.

I've created a new PR which includes the sntp binary as a part of the ntp-utils package while removing it and its libevent dependencies from the other packages.