openwrt/packages #10313: luajit: compilation fails except on x86_64

feckert, 2019-10-21:
Maintainer: @milani @neheb 
Environment: x86_64 lantiq_xrx200, APU3, OpenWrt master

Description:
The compilation fails on lantiq_xrx200 and x86_geode.
I didn't try the other targets. 
Only x86_64 does build luajit as expectd.

Problem:
I update my config and now I have a dependency to luajit.
On my build I am using snort. So I think this was commit https://github.com/openwrt/packages/commit/102e67cc15eaa7b917e4d8f33ada1fa8d0b745c1#diff-a0db22578994a62ff285551444ef788e
which added luajit to the dependency tree and now my build needs luajit.
But I get the following compilation error.

Compilation error output:
```
make -j1 -C /home/feckert/workspace/openwrt/build_dir/target-mips_24kc_musl/LuaJIT-2.1.0-beta3 HOST_CC="gcc -O2 -I/home/feckert/workspace/openwrt/staging_dir/host/include -I/home/feckert/workspace/openwrt/staging_dir/hostpkg/include -I/home/feckert/workspace/openwrt/staging_dir/target-mips_24kc_musl/host/include -m32" CROSS="mips-openwrt-linux-musl-" DPREFIX=/home/feckert/workspace/openwrt/build_dir/target-mips_24kc_musl/LuaJIT-2.1.0-beta3/ipkg-install/usr PREFIX=/usr TARGET_SYS=Linux TARGET_CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -ffile-prefix-map=/home/feckert/workspace/openwrt/build_dir/target-mips_24kc_musl/LuaJIT-2.1.0-beta3=LuaJIT-2.1.0-beta3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro
make[3]: Entering directory '/home/feckert/workspace/openwrt/build_dir/target-mips_24kc_musl/LuaJIT-2.1.0-beta3'
==== Building LuaJIT 2.1.0-beta3 ====
make -C src
make[4]: Entering directory '/home/feckert/workspace/openwrt/build_dir/target-mips_24kc_musl/LuaJIT-2.1.0-beta3/src'
HOSTCC    host/minilua.o
In file included from /usr/include/limits.h:25:0,
                 from /usr/lib/gcc/x86_64-linux-gnu/6/include-fixed/limits.h:168,
                 from /usr/lib/gcc/x86_64-linux-gnu/6/include-fixed/syslimits.h:7,
                 from /usr/lib/gcc/x86_64-linux-gnu/6/include-fixed/limits.h:34,
                 from host/minilua.c:33:
/usr/include/features.h:364:25: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
                         ^
compilation terminated.
Makefile:691: recipe for target 'host/minilua.o' failed
make[4]: *** [host/minilua.o] Error 1
```

feckert, 2019-10-21:
I think we have a hostbuild dependency because x86_64 builds thats my build arch.

neheb, 2019-10-21:
You need GCC multilib to build on other platforms.

feckert, 2019-10-22:
> You need GCC multilib to build on other platforms.

@neheb can you please explain more exactly what I should do / can do

feckert, 2019-10-22:
Ok now it compiles. I installed gcc-multilib on my debian stretch host with the command

```aptitude install gcc-multilib```

Should we not add this as a build dependency to the luajit package?

feckert, 2020-01-20:
@neheb On every re installation of my build system, I run into this error!
I know that I could fix this by installing `aptitude install gcc-multilib`.
But could we not add a check, that we need gcc-multilib before compilation?

I'm not sure if this host build dependency is intended. Should this files not be part of openwrt toolchain?

This is the error output.
`
==== Building LuaJIT 2.1.0-beta3 ====
make -C src
make[4]: Entering directory '/home/feckert/workspace/openwrt/LDM-master-xrx200/build/openwrt/build_dir/target-mips_24kc_musl/LuaJIT-2.1.0-beta3/src'
HOSTCC    host/minilua.o
In file included from /usr/lib/gcc/x86_64-linux-gnu/8/include-fixed/limits.h:194,
                 from /usr/lib/gcc/x86_64-linux-gnu/8/include-fixed/syslimits.h:7,
                 from /usr/lib/gcc/x86_64-linux-gnu/8/include-fixed/limits.h:34,
                 from host/minilua.c:33:
/usr/include/limits.h:26:10: fatal error: bits/libc-header-start.h: No such file or directory
 #include <bits/libc-header-start.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated
`

neheb, 2020-01-28:
I don't think a full host GCC should be included.

neheb, 2020-07-23:
@feckert the check would need to be done in include/prereq-build.mk in base, not here.

Upstream has not had any desire to do this.