openwrt/packages #10316: mwan3: _disconnected_ event not fired when an ifdown event occurs

thebaron06, 2019-10-21:
tested with mwan3 version: 2.6.18
on a Dragino MIPS Board with inet via WAN and an USB Modem using pppd.

Maintainer: @feckert @damadmai 
Environment: mips, dragino, OpenWRT 18.06.2

Description:
With mwan3 tracking the WAN i/f and another pppd based i/f the _disconnected_ hotplugd event is never fired.
I do think this is caused by setting the _score_ value to zero in case of an detected ifdown event, see  [here](https://github.com/openwrt/packages/blob/openwrt-18.06/net/mwan3/files/usr/sbin/mwan3track#L19) and [here](https://github.com/openwrt/packages/blob/openwrt-18.06/net/mwan3/files/usr/sbin/mwan3track#L240). This causes [this condition](https://github.com/openwrt/packages/blob/openwrt-18.06/net/mwan3/files/usr/sbin/mwan3track#L198) to never be true.

thebaron06, 2019-10-21:
Oh, just realized this issue is already fixed in the master. But it remains open for mwan3 2.6.18 that is the default for OpenWRT, isn't it?

Temikus, 2019-10-25:
I'm running into the same issue it looks like. Can I just pluck a newer package out of 19.07 or trunk? Or were there any changes that will make it incompatible with 18.06?

feckert, 2019-10-25:
Difficult to say, but I figured it would work.
Nothing has changed in the confiugration.
If then only new things have been added.
To be on the safe side, you should first make a backup of the configuration.
After that you should uninstall the 18.06 mwan3 package and install the new 19.07 one.

feckert, 2020-05-29:
@thebaron06 @Temikus I haven't heard from you. I assume you've solved the problem. If not, you can reopen the ticket or tell us how you solved the problem.

closed 2020-05-29 by feckert

thebaron06, 2020-06-01:
As I already stated in a former comment, it was fixed in newer versions already.

As we pinned OS and App version, we solved this by backporting the missing signal to the version we are stuck with.