openwrt/packages #10328: luci-app-acme: Challenge error: {"type":"urn:acme:error:malformed","detail":"Expired authorization","status": 404}

NomadThanatos, 2019-10-22:
Maintainer: @tohojo 
Environment: Linksys WRT3200ACM, ARMv7 Processor rev 1 (v7l), OpenWrt 18.06.4 r7808-ef686b7292 / LuCI openwrt-18.06 branch (git-19.170.32094-4d6d8bc)

Description:

The cert will be expired in 20 days, but luci-app-acme fails to renew.
The error shows ```Challenge error: {"type":"urn:acme:error:malformed","detail":"Expired authorization","status": 404}```
I see a similar issue(https://github.com/Neilpang/acme.sh/issues/2479) on ACME, but it is closed, so I think this might be a luci-app-acme issue.

Here is my debug log.

```
Mon Oct 21 17:22:11 2019 daemon.info acme: Running ACME for router.yx.ht
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: acme: Running ACME for router.yx.ht
Mon Oct 21 17:22:11 2019 daemon.info acme: Found previous cert config. Issuing renew.
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: acme: Found previous cert config. Issuing renew.
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: Lets find script dir.
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: _SCRIPT_='/usr/lib/acme/acme.sh'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: _script='/usr/lib/acme/acme.sh'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: _script_home='/usr/lib/acme'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: Using config home:/etc/acme
Mon Oct 21 17:22:11 2019 daemon.info run-acme[22226]: https://github.com/Neilpang/acme.sh
Mon Oct 21 17:22:11 2019 daemon.info run-acme[22226]: v2.7.8
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: Using config home:/etc/acme
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: ACME_DIRECTORY='https://acme-v01.api.letsencrypt.org/directory'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: DOMAIN_PATH='/etc/acme/router.yx.ht'
Mon Oct 21 17:22:11 2019 daemon.info run-acme[22226]: Renew: 'router.yx.ht'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: Le_API='https://acme-v01.api.letsencrypt.org/directory'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: Using config home:/etc/acme
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: ACME_DIRECTORY='https://acme-v01.api.letsencrypt.org/directory'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: _main_domain='router.yx.ht'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: _alt_domains='no'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: Using ACME_DIRECTORY: https://acme-v01.api.letsencrypt.org/directory
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: _init api for server: https://acme-v01.api.letsencrypt.org/directory
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: GET
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: url='https://acme-v01.api.letsencrypt.org/directory'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: timeout=
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: _CURL='curl -L --silent --dump-header /etc/acme/http.header  -g '
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: ret='0'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: ACME_KEY_CHANGE='https://acme-v01.api.letsencrypt.org/acme/key-change'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: ACME_NEW_AUTHZ='https://acme-v01.api.letsencrypt.org/acme/new-authz'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: ACME_NEW_ORDER='https://acme-v01.api.letsencrypt.org/acme/new-cert'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: ACME_NEW_ACCOUNT='https://acme-v01.api.letsencrypt.org/acme/new-reg'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: ACME_REVOKE_CERT='https://acme-v01.api.letsencrypt.org/acme/revoke-cert'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: ACME_AGREEMENT='https://letsencrypt.org/documents/LE-SA-v1.2-November-15-2017.pdf'
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: ACME_NEW_NONCE
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: ACME_VERSION
Mon Oct 21 17:22:11 2019 daemon.err run-acme[22226]: Le_NextRenewTime='1570683740'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _on_before_issue
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _chk_main_domain='router.yx.ht'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _chk_alt_domains
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: Le_LocalAddress
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: d='router.yx.ht'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: Check for domain='router.yx.ht'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _currentRoot='dns_cf'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: d
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _saved_account_key_hash is not changed, skip register account.
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: Read key length:2048
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _createcsr
Mon Oct 21 17:22:12 2019 daemon.info run-acme[22226]: Single domain='router.yx.ht'
Mon Oct 21 17:22:12 2019 daemon.info run-acme[22226]: Getting domain auth token for each domain
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: ok, let's start to verify
Mon Oct 21 17:22:12 2019 daemon.info run-acme[22226]: Verifying:router.yx.ht
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: d='router.yx.ht'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: keyauthorization='tlw_ia3aEkzWBhsTynIEjPgBFi5RWTVfNFasAyizPfU.TOrUuKUFN1TphQMVvGj3Lb73ClZU_clck1hIE2R9OzQ'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: uri='https://acme-v01.api.letsencrypt.org/acme/chall-v3/711930408/ggfzHQ'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _currentRoot='dns_cf'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: url='https://acme-v01.api.letsencrypt.org/acme/chall-v3/711930408/ggfzHQ'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: payload='{"resource": "challenge", "keyAuthorization": "tlw_ia3aEkzWBhsTynIEjPgBFi5RWTVfNFasAyizPfU.TOrUuKUFN1TphQMVvGj3Lb73ClZU_clck1hIE2R9OzQ"}'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: RSA key
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: GET
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: url='https://acme-v01.api.letsencrypt.org/directory'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: timeout=
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _CURL='curl -L --silent --dump-header /etc/acme/http.header  -g '
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: ret='0'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: POST
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _post_url='https://acme-v01.api.letsencrypt.org/acme/chall-v3/711930408/ggfzHQ'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _CURL='curl -L --silent --dump-header /etc/acme/http.header  -g '
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _ret='0'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: code='404'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: router.yx.ht:Challenge error: {"type":"urn:acme:error:malformed","detail":"Expired authorization","status": 404}
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: Skip for removelevel:
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: pid
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: No need to restore nginx, skip.
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _clearupdns
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: skip dns.
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _on_issue_err
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: Please add '--debug' or '--log' to check more details.
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: See: https://github.com/Neilpang/acme.sh/wiki/How-to-debug-acme.sh
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: url='https://acme-v01.api.letsencrypt.org/acme/chall-v3/711930408/ggfzHQ'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: payload='{"resource": "challenge", "keyAuthorization": "tlw_ia3aEkzWBhsTynIEjPgBFi5RWTVfNFasAyizPfU.TOrUuKUFN1TphQMVvGj3Lb73ClZU_clck1hIE2R9OzQ"}'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: POST
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _post_url='https://acme-v01.api.letsencrypt.org/acme/chall-v3/711930408/ggfzHQ'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _CURL='curl -L --silent --dump-header /etc/acme/http.header  -g '
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: _ret='0'
Mon Oct 21 17:22:12 2019 daemon.err run-acme[22226]: code='404'
Mon Oct 21 17:22:13 2019 daemon.err run-acme[22226]: socat doesn't exists.
Mon Oct 21 17:22:13 2019 daemon.err run-acme[22226]: Diagnosis versions:
Mon Oct 21 17:22:13 2019 daemon.err run-acme[22226]: openssl:openssl
Mon Oct 21 17:22:13 2019 daemon.err run-acme[22226]: OpenSSL 1.0.2t  10 Sep 2019
Mon Oct 21 17:22:13 2019 daemon.err run-acme[22226]: apache:
Mon Oct 21 17:22:13 2019 daemon.err run-acme[22226]: apache doesn't exists.
Mon Oct 21 17:22:13 2019 daemon.err run-acme[22226]: nginx:
Mon Oct 21 17:22:13 2019 daemon.err run-acme[22226]: nginx doesn't exists.
Mon Oct 21 17:22:13 2019 daemon.err run-acme[22226]: socat:
```

tohojo, 2019-10-22:
Could you post the contents of /etc/config/acme, please?

NomadThanatos, 2019-10-22:
```
root@WRT3200ACM:~# cat /etc/config/acme

config acme
        option state_dir '/etc/acme'
        option account_email '***REDACTED***'
        option debug '0'

config cert 'YXHT'
        option update_uhttpd '1'
        option webroot '/www/'
        list domains 'router.yx.ht'
        option dns 'dns_cf'
        list credentials 'CF_Key=***REDACTED***'
        list credentials 'CF_Email=***REDACTED***'
        option use_staging '0'
        option enabled '1'
        option keylength '4096'
```
The keylength used to be 2048, but since I cannot auto renew the cert, I would like to get a 4096 one.

tohojo, 2019-10-23:
NomadThanatos <notifications@github.com> writes:

> ```
> root@WRT3200ACM:~# cat /etc/config/acme
>
> config acme
>         option state_dir '/etc/acme'
>         option account_email '***REDACTED***'
>         option debug '0'
>
> config cert 'YXHT' [...]

Right, but the error you posted you got before you changed the key
length, right?

tohojo, 2019-10-23:
It seems you are using an old-ish version of acme.sh; I just pushed an
update to the openwrt package to update acme.sh to 2.8.3. It seems that
this version has also switched to the v2 ACME API, so perhaps that can
resolve this issue?

NomadThanatos, 2019-10-23:
Yeah, the error occurred before I changed the key length.
My package comes from ```src/gz openwrt_packages http://downloads.openwrt.org/releases/18.06.4/packages/arm_cortex-a9_vfpv3/packages```
The luci-app-acme/acme version is still 2.7.8-4.

tohojo, 2019-10-26:
Could you try manually downloading and installing acme v2.8.3 from here,
please:
https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/acme_2.8.3-1_all.ipk

Assuming it fixes the problem, I can push an update to the 18.06 branch...

NomadThanatos, 2019-10-28:
It fixes the problem and now my cert is renewed, thanks for your help.
```
Mon Oct 28 12:32:44 2019 daemon.info run-acme[13404]: Cert success.
Mon Oct 28 12:32:44 2019 daemon.err run-acme[13404]: _on_issue_success
```

tohojo, 2019-10-29:
NomadThanatos <notifications@github.com> writes:

> It fixes the problem and now my cert is renewed, thanks for your help.

Great, thanks for testing! I'll push an update to the package for 18.06...

closed 2019-10-29 by tohojo